alexa-pi/AlexaPi #2: [metabug] New codebase

renekliment, 2016-09-09:
- [x] New common repo
- [ ] Cleaner directory structure
- [ ] Platforms
    - [ ] Device platforms
    - [ ] OS platforms
    - [ ] Merge AlexaCHIP & AlexaOPi (and redirect those repos here)
    - [ ] Make it possible to select multiple triggering methods when they are available
    - [ ] Refactor and merge @maso27's snowboy implementation
- [ ] Common configuration in the config file (no config in `.py` files), maybe YAML

renekliment, 2016-09-18:
@illperipherals Let's move the discussion here so we don't spam that actual issue.

The instructions for Mason were for the older PR that didn't solve the path things so I'm not surprised it didn't work.

CR and testing is exactly what's needed! So if you have any time, please look at #8 and #10. They both received quite some attention already and I think there are no more problems, but I need someone with a final word. Then we can move to the stuff that rocks :-)
Thank you!

illperipherals, 2016-09-18:
Sounds good.

renekliment, 2017-11-04:
The original idea of this is done basically. Closing.

closed 2017-11-04 by renekliment