alexa-pi/AlexaPi #3: Continuous Integration (CI)

renekliment, 2016-09-09:
Probably Travis CI? Do you guys have any experience with it? I can give it a shot.

renekliment, 2016-09-20:
@sammachin I've wanted to register Travis CI for this, but it seems that it needs your permission as the organization owner. It should have sent you an e-mail or something. Could you allow please?

Also, could you add @illperipherals to the organization? 
Thank you.

sammachin, 2016-09-20:
Done and done!

renekliment, 2016-09-20:
Cool, thanks. I see however at least the setup of Travis CI as a part of this issue (the tests can go in a different one), therefore I'm keeping this open for now.

renekliment, 2016-09-25:
Forgot to tag this issue in the commit (damn), so mentioning here and now. 
#16 contains a super basic setup of Travis CI. Keeping this open until we get something more sane and the tests don't fail by default.

renekliment, 2016-10-17:
Silly me again! #36 contains fixes, so this repo builds successfully by default now. The build pass is mandatory now for new PRs. When we do #21, we'll enable other pylint stuff to check more than just errors and then we can close this.

For tests, there should be a separate issue.

commit 82a97187 (2016-11-10): Code cleanup (#67), fixes #21, fixes #3

* Cleaned up code (almost PEP8)

* Added pylint rc file

* Better requirements file(s)
closed 2016-11-10 by renekliment