alexa-pi/AlexaPi #6: Merge work in forks

renekliment, 2016-09-10:
There are various forks of the original project. Moreover, there are forks of those forks. The goal of this is that when a common code base is provided, we can unify the work of so many people, so we can work together to be effective, do not duplicate work and not to confuse end users.

Known forks that we would like to merge eventually (actually rather soon):
- [ ] @flooie https://github.com/flooie/AlexaPi/
- [ ] @sammachin https://github.com/sammachin/AlexaCHIP
    - [ ] @jitto https://github.com/jitto/AlexaCHIP
    - [ ] @johnwskinner https://github.com/johnwskinner/AlexaCHIP
    - [ ] @billzajac https://github.com/billzajac/AlexaCHIP
- [ ] @dony71 https://github.com/dony71/AlexaOPi
    - [ ] @GadgetReactor https://github.com/GadgetReactor/AlexaOPi

If you are one of the devs mentioned above and seeing this, please consider joining forces with us. The new code base that is gonna support multiple devices and OS platforms is gonna be ready in a week or two for testing and implementing new things. This is a note so you might consider this. If you have any questions, do not hesitate to ask in comments. Thank you!

flooie, 2016-09-10:
@renekliment happy to help.  just let me know what you need me to do.  I'm not a professional dev.

jitto, 2016-09-10:
@renekliment My changes were to run in python3 and use USB audio and a few months old.  If these are not already in upstream, I can definitely create a PR once the new code base is ready.

renekliment, 2016-09-10:
Awesome, glad to hear that :-) I'll let you know once the new code base is ready.

PhyberApex, 2016-09-11:
Reporting in. I don't know what the "current" code base is but I did the voice activation by a third party thing called "Snowboy" so I am ready to help out once the codebase is up.

renekliment, 2016-09-12:
Also, if any of you has some time and wants to at least review Pull Requests in this project, please contact @sammachin and ask him to add you to the organization. Thanks!

renekliment, 2016-09-18:
So before we're ready to merge the changes in your forks, we have to first stabilize the new code base. However, it seems that there's not so many people with much free time to help out with that. Do you guys (@flooie, @PhyberApex, @jitto, others mentioned) have any free time to do code review / testing of the PRs I'm pushing? 

Sorry for spamming so much and thanks!

flooie, 2016-09-18:
@renekliment - Unfortunately, I am starting a new job tomorrow - so my availability will be significantly limited for the foreseeable future.

GadgetReactor, 2016-09-19:
Sorry for the late reply, I will be happy to help but availability (and
capability) is limited.

My code is running on an OrangePi. I have a RaspberryPi 3 but I don't have
a USB mic.

renekliment, 2016-10-26:
@bdatdo0601 I somehow missed your fork, sorry! If you have any amount of free time, please join us here :)

bdatdo0601, 2016-10-27:
@renekliment this sounds really amazing. For my fork I worked on combining AlexaPi with Snowboy and ensure only one audio input stream is active. I really want to switch from Snowboy and use Python Speech Recognition as a hotword detection for this project.

Unfortunately, I don't have a lot free time. I think I will mostly available during the weekend so I can help out then. Just let me know when you need me :).

Oh and I'm not a professional dev neither. Just doing this out of curiosity.

renekliment, 2016-12-11:
I've noticed that also @jaws-ug-kobe and @rhee have been doing some work in their forks. Do you guys want to join us to make one great repo / version?

renekliment, 2016-12-11:
Also tagging @tomh05, @Robertsl2 and @the-raspberry-pi-guy.

renekliment, 2016-12-25:
And tagging @rhee

pedrominatel, 2016-12-29:
@renekliment My fork works on the Intel Edison boards with the MRAA. I'm considering this!

Thanks!

renekliment, 2016-12-29:
@pedrominatel Great to hear that! From a brief look, it should be enough to create a new device_platform (there's a plenty to take an inspiration from).

renekliment, 2017-02-19:
@jaws-ug-kobe @haruharuharuby
I see you're doing various work on the old AlexaPi fork. How about joining forces here?

haruharuharuby, 2017-02-19:
@renekliment sure! i will soon join here.