alexa-pi/AlexaPi #7: Python 3 support

renekliment, 2016-09-10:
There are two sides of this:
- [ ] a support to run in under py3
- [ ] make py3 default

We need to be sure that there is no major drop in performance there to use it as default.

Also, we need to be sure that all the dependencies support this to do that. If the don't - find another solution if possible. The base should however support py3 without a problem - achieve this first.

sammachin, 2016-09-20:
I think this may be a requirement for the new API, IIRC the python HTTP/2 lib that I was looking at only runs under Py3.

renekliment, 2016-09-20:
If you mean [hyper](https://hyper.readthedocs.io/en/latest/) it should run under both py2 and py3.

> `hyper` supports Python 3.4 and Python 2.7.9, and can speak HTTP/2 and HTTP/1.1.

renekliment, 2016-10-19:
Don't forget to enable Python 3 on Travis CI when this is done.

closed 2018-12-22 by renekliment