alexa-pi/AlexaPi #9: Refactor init scripts / monitor

renekliment, 2016-09-14:
- [ ] when issuing `/etc/init.d/AlexaPi stop`, the monitor should stop too
- [ ] merge `initd_alexa.sh` and `initd_alexa_monitored.sh`, because they're doing almost the same thing - use some configuration handling in the init script to find out if the monitor should be started too; where do we store the config option? setup script needs to reflect this too
- [ ] use a `.pid` file which is unique; store in AlexaPi dir so we don't need root for that - possibly `/opt/AlexaPi/tmpcontent/AlexaPi.pid` eventually
    - gets rid of `/tmp/dont_start` since its not application unique
    - gets rid of `pgrep python -a | grep main.py` which could potentially hit another app
    - [ ] when the app starts a `.pid` file is written
    - [ ] the monitor checks that a process with that PID exists, if not -> start an instance

renekliment, 2016-09-17:
@maso27 So if you want to take some of these things up, in #11, you can find a hint how the config should be handled. I'm not sure yet what the config variable (and section) should be called. 

I've got a question - why is the monitor actually there? Did you encounter crashes of AlexaPi? What was the reason? Or is it there _just in case_?

illperipherals, 2016-09-18:
Pulling a clean build still requires "root" to be edited to "opt" in two places. Both errors show up in the log. Verified at an Alexa Hackathon yesterday. Should this be rolled into an existing refactor issue or a new PR?

renekliment, 2016-09-18:
@illperipherals Are you pulling the `master` branch? I'm sorry for the misguided `README.md` in `master`, I don't know why I edited it before it the code was there.

There is currently a pending PR (#10) that gets rid of `/root` (or `/opt` for that matter) in the python code. You can clone it wherever you want. If you want AlexaPi to start on boot however, you have to clone it into `/opt`.

If you want to give it a try:
```
git clone -b universal-paths https://github.com/renekliment/AlexaPiNG.git AlexaPi
sudo ./AlexaPi/src/scripts/setup.sh
```  

Also, if you're a developer, familiar with the project and want to make it better, you're more than welcome to join in on the project. I've got a lot of ideas (and code) for it, but I'm in a desperate need for someone to review my PRs.

illperipherals, 2016-09-18:
I have it up in a battery-run unit using snowboy, but I like this initiative to bring everything together. I develop, and I would like to put more in, but I do not want to interfere with the whirlwind refactor going on. If you need someone for CR, however, I am very available.

illperipherals, 2016-09-18:
To answer your question, master was pulled yesterday, but previously I was pulling in via the instructions given to Mason. I had two bad paths on that as well, but not related to what I am currently discussing.

Not sure why I am getting different results. (but got it working fine)

renekliment, 2016-09-19:
Notes to myself: 
- Move the silent option to the config file once #11 is merged.
- We'll lose Alexa starting in silent mode when the service gets restarted by systemd, but whatever.

commit 87016768 (2016-10-05): Init scripts refactor - fixes #9

* added systemd unit
* use systemd as default
* runs as unprivileged user now (alexapi)
commit cf01c41e (2016-10-05): Merge pull request #28 from renekliment/inits-refactor-2

Init scripts refactor - fixes #9
closed 2016-10-05 by renekliment