alexa-pi/AlexaPi #12: [request] Benchmarks for different hardware/config builds.

illperipherals, 2016-09-18:
I belive it would be of benefit to all parties to correleate benchmarks for different HW platforms. This extends from SDs to what processes are being run on the device.

Thoughts?

renekliment, 2016-09-18:
Sounds cool. However, we have to postpone this to the future, when we can actually run AlexaPi on multiple platforms. 
I personally don't have any experience with benchmarks though.

sammachin, 2016-09-20:
We could certainly start out by testing the code on all the versions of the RPi (pretty sure i'be got one of each apart from the A.

In terms of benchmarking as a start we could just look at CPU utilisation when Idle. I don't think most of this is the sort of thing that will run 'faster' on better hw most of the time is in recording/playing and network connections

illperipherals, 2016-10-22:
I think it would also be valid to outline the importance of speed-testing your SD cards in the docs.
Case in point:

These are are "technically" labeled as the same speed cards: (and they are all pretty major brands)

Orange: Kingston 32GB: Mistserver Arch Linux - Nginx streamer
Test finished without errors.
Writing speed: 11.6 MByte/s
Reading speed: 78.3 MByte/s

Purple: Kingmax 16GB : Kali
Test finished without errors.
Writing speed: 7.50 MByte/s
Reading speed: 20.2 MByte/s

Pink Samsung: 16GB, fresh
Test finished without errors.
Writing speed: 7.70 MByte/s
Reading speed: 16.1 MByte/s

Red/Grey SanDisk 32GB Pi3 main backup (raspberrypi)
Test finished without errors.
Writing speed: 22.5 MByte/s
Reading speed: 77.7 MByte/s

Red/Grey SanDisk 32GB MTM Pi Zero button (zeropi)
Test finished without errors.
Writing speed: 20.5 MByte/s
Reading speed: 82.5 MByte/s

White and Orange Samsung 32GB pi@pizero
Test finished without errors.
Writing speed: 20.0 MByte/s
Reading speed: 41.6 MByte/s

Kingston 16GB Pi Zero
Test finished without errors.
Writing speed: 4.80 MByte/s
Reading speed: 20.6 MByte/s

renekliment, 2016-10-22:
Regarding that, here's T. Kaiser's comprehensive forum thread about that over at Armbian: http://forum.armbian.com/index.php/topic/954-sd-card-performance/