alexa-pi/AlexaPi #13: [metabug] The future

renekliment, 2016-09-19:
Things we'd like to do:
- [ ] Switch over to the new API!
- [ ] Make sure the time is synced (NTP). We need this for timers/alarms to work properly.
- distribution packages
    - [ ] Linux packages
    - [ ] Windows packages
    - [ ] OS X packages

sammachin, 2016-09-20:
I have some code that plays the responses back via SONOS on the local network which I'd like to feed into the project, will submit a PR at some point with it.

sammachin, 2016-09-20:
In terms of distribution packages I've been wondering about providing a RasPi image ready to go that you can just burn on your laptop and drop into the Pi.
As well as this I think I would then set it up with a set of dedicated developer credentials so the user just has to go through the sign in to amazon part and doesn't need to register as a developer as thats been a bit of a barrier to entry with the project so far.
I need to look into the security side though and what the implications of putting the developer creds into a package are, it might need a custom auth server to be run on the web somewhere.

illperipherals, 2016-09-20:
I like this idea, but what are your thoughts on going Arch Linux?

sammachin, 2016-09-20:
@illperipherals

While I don't have a strong bias towards any particular distro there are a couple of reasons I would currently stick with Raspbian.

First and foremost we are a Raspberry Pi project Raspbainnis the default distro and for new people getting started with this as their first hack using a different distro is another complication.

Raspbian is the most common and therefore I would imagine we would have the best range of support for add ons like sonos and airplay.

Having said that if we can build a project that can run on both then we should try and enable it but for now on this code base I think we need to concentrate on getting a clean and stable project with support for as much of the AVS API as possible, other things are a distraction

illperipherals, 2016-09-20:
I tend to agree, but I think running a headless *minimal* install makes more sense on Arch (esp when dealing with the Zeros). Maybe this is an area where I could be of use. I would like to get onto API v2 sooner than later, so I don't want to make distractions. Maybe just keep it in the back of your mind.

renekliment, 2016-09-20:
@sammachin That's gonna be awesome! The sound handling is definitely gonna be modular, since I want to play music stuff through my local mpd server.

I gave it some more thought and I guess you're right - it might be beneficial for quite a number of users. We have to be smart about it though. 
For example we should make a `.deb` package out of AlexaPi and include that package installed in the image, not just some copied files who knows where. The default config with credential would be in a separate package. 
I'm for digitally signing that package and providing file hashes and strongly encouraging users to check both. The How-To is gonna be two lines longer, but security should not be taken lightly.
Regarding distributing credentials to users I know literally nothing, so that's up to you or someone to find out.
I guess we'll have a proper discussion when we get there :-) Let's have this as some notes for us.

And one more thing that I'd like to mention is that I'd like this image thing to be a secondary distribution choice. I believe installing AlexaPi on an existing systems, where users already have things they want to run is a way to go. For this we have to make the installation (and instructions) as easy as possible. I'll have some commits on that when we get there :-)

renekliment, 2016-09-20:
@sammachin @illperipherals 

Well, the AlexaPi itself will run on any Linux distro on any device just fine (I actually have it running on my Arch Linux laptop already with this `device-platforms` thing). So if you @illperipherals want to do an Arch Linux version of that I say why not. Again, we'd have to be smart about that as mentioned above and we'd probably put a note there like "for advanced users only" and keep the Raspbian variant the default and recommended. This way beginners wouldn't be confused and advanced users would be given a choice.

EmerickH, 2017-05-31:
@renekliment 
Hello, I'll be happy to help you, I can help you to compile packages for Rasperry and Windows and manage the issues on Github. I love helping!
Is it possible to add me as collaborator?

Thank you,
Emerick

renekliment, 2017-05-31:
@EmerickH Awesome! I've sent you an invite to the org. and I'll add you some permissions later. Please note the project rules on code-related stuff (review, etc - in the repo).

Also, note that the development happens in the `dev` branch. We've already switched to PyAudio from pyalsaaudio there.

EmerickH, 2017-05-31:
@renekliment Thank you very much!
For PyAudio, it's just perfect! I am actually working on a Windows adaptation with Cygwin and the only problem was alsa!

Emerick

EmerickH, 2017-06-11:
Hello @renekliment,
I'm happy to announce that AlexaPi-Windows is now working!
There's still some little problems but it's working!

Emerick

renekliment, 2017-06-11:
@EmerickH Great! Can you rebase it on top this repo's `dev`? Maintaining separate projects is way too much effort. If you could do that, all of the people would benefit from future changes.