alexa-pi/AlexaPi #14: AVS legal stuff

renekliment, 2016-09-19:
So I've been reading through [Alexa Voice Service Functional Design Guide
](https://developer.amazon.com/public/solutions/alexa/alexa-voice-service/content/alexa-voice-service-functional-design-guide) and [Alexa Voice Service Terms and Agreements](https://developer.amazon.com/public/solutions/alexa/alexa-voice-service/support/terms-and-agreements) and there's a lot of things that kinda suck that we don't have (yet) like:
- you can't change the wake word,
- you have to indicate recording start and **stop**,
- the user needs to be able to mute the mic and this needs to be indicated,
- etc.

So I was thinking if any of you know if we have to also conform with those two documents even when we are just messing with it, not doing anything commercial. We can probably implement those things that we don't have yet, but it's gonna take some time. Is there an issue that we're doing the development "out in open" and people can use it now even if it doesn't comply with those two documents?

sammachin, 2016-09-20:
Ok SO IANAL but here's my view on things, (@flooie is a Lawyer I think so he might be well placed to comment)

Firstly we should remember that we are an open source project not a finished Alexa Device, right now we don't even _need_ to have an Alexa Developer Account ourselves (in theory) When a user installs AlexaPi they signup and setup their own set of developer credentials and they are agreeing to the terms then, its up to them what they do with the code if they make changes they should ensure they are still within the terms of the agreement. 

I hadn't properly looked at the T&C's since they removed the blanket ban on using a wakeword at all, interesting that they now say you can't change it
`p) you will not use or allow the use of any wake word or other trigger word as part of accessing the Alexa Service in Your Products other than the wake or trigger words authorized by Amazon (such as “Alexa”) without Amazon’s prior written approval;`
They don't specify there what the authorized wake words are exactly, I would take it to read that its "Alexa, Amazon, & Echo" like in the Echo so having the ability to change that within our code is fine, again if people use something totally different thats their lookout.

Indicating recording start/stop is already something that was in the project from day 1, The LED shows when its recording. If people choose to not wire this up thats their choice.

The ability to mute the mic is more interesting and I think has only been introduced with the allowing of wake words, In the original push to talk version I would argue that the mic was permanently muted until the user pressed the button to talk and then engaged it.
In a new wakeword setup thins might need some more work, options that I see are;
- The dumb solution is to say "Its a USB Mic, if you want to mute it just unplug the mic"
- Some of the USB dongles actually have a mic mute button built in, we recommend using this one.
- We could re-use the button as a mute on/off switch in our code. How would we actually mute the mic? Would it be sufficient to disable wakeword when 'muted'
- One school of thought is that seeing as we are open software then you _can_ disable the mic by SSH'ing onto the box and running alsa mixer!

The naming is interesting, so far there are far more commercial projects that are using variants of the Alexa name and haven't had an issue, Amazon are very much aware of this project so I'm happy to continue as we are, 
Again I don't really see this as something we need to be too worried about, I've got a pretty good rapport with Amazon I'm sure if there are any problems they will reach out in a friendly way first and we'll find a way to comply then. The whole no wakeword thing was something that I raised with them early on and they were really good at finding ways around it.

I do think some of these points above are worth @nascentobjects thinking about though as their project is far more of a commercial appliance and they don't have either a visual indicator or a button on the device.

renekliment, 2017-01-04:
No one from Amazon tried to talk to us (AFAIK). Closing ... Can be reopened when needed.

closed 2017-01-04 by renekliment