alexa-pi/AlexaPi #22: Audio output issue (solved)

tinker3433, 2016-09-23:
Hi,
Solved my own issue as i was writing it up

I had no audiooutput, but i could tell from the LED that it was recording and playing back. 
Green and Red (I miss the Yellow for "transmitting)
It also would give a Green LED if i just said "Alexa"

The issue was that i had a HDMI monitor connected to set up the WI-FI. which i found out, as i'm sure  most of you guys know already, that that kills the Audio Jack output.

Might be worth a mention in the instructions for others that are like me and just starting out using Pi.

renekliment, 2016-09-24:
Hey. Thanks for the notice!
We'll make a separate config options for playback and recording device and document it, so people can actually use sound over jack even when HDMI is connected.

sammachin, 2016-09-25:
@tinker3433 yep I miss the transmitting or `processing` indication too, its top of my list to bring back in

renekliment, 2016-09-25:
Depends of #26.

closed 2016-10-30 by renekliment

renekliment, 2016-10-30:
@tinker3433 So now you can also set the output device in `config.yaml`. Please test if you can set audio output to go through the analog, with HDMI connected.

maso27, 2016-10-30:
As a side note: I can make an external sound card work by changing output_device to "hw:2,0" just fine, so this is a good advancement.  But I'm stumped on getting the jack to work when HDMI is connected.

ShawnHarte, 2016-10-30:
I can only get analog audio to play with HDMI connected by using force_audio 0.  mpg123 can use the -o flag to output audio to HDMI or analog with 1 or 0 respectively.  Someone a bit more familiar with the code may be able to make this work programatically, as it does seem possible to play analog sound with HDMI connected.

commit 012d532f (2016-10-30): Merge pull request #66 from renekliment/audio-output

Configurable audio output, fixes #22
commit 2d1ffcec (2016-10-30): configurable audio output, fixes #22