alexa-pi/AlexaPi #24: Refactor: triggering

renekliment, 2016-09-24:
Currently, it's kinda messy. We need to make it a little more general / modular.

We definitely need to support:
- platform triggering
- pocketsphinx / snowboy triggering
- keypress triggering when not running in daemon mode

sammachin, 2016-09-25:
what do you mean by `device_platform` triggering.

I'm keen to bring back the option of 'press to talk' only mode like in my original, I was giving a talk the other eve with a Pi running this codebase on the desk and the amount of times it triggered when I said "Alexa" was crazy!

Been thinking about adding support for Keyboard/HID triggers for a while, I've got one of the RasPi SenseHats which has a grid of neopixels and a joystick which mimicks keyboard inputs would be nice to have that as a hardware option for the Pi as thats then soldering free for users.

renekliment, 2016-09-25:
@sammachin I'm sorry I keep mentioning the _device_platforms_ thing when there's no public code. It's exactly what you mean by #26. I have some code ready on my local machine, but needed to merge the config first. I'll rebase and publish it ASAP.

Yes, we'll definitely make the voice activation configurable.

We can then rework the the things you mention once we have #26.

sammachin, 2016-09-25:
Ok would be good to see what you're doing ASAP I've got a little time today to do some coding (finally!) and I was gonna work on the indications as I'm dying to bring back the processing state LED its really starting to bug me not having that.

renekliment, 2017-01-07:
Damn, forgot to close by commit message - again! Fixed by #119.

PR #119 (2017-01-07): Triggering refactor
This fixes #24.

It is a very important PR project-wise for these two reasons:
- All the triggers can be enabled / disabled, therefore the voice activation is now optional and once this is released (in _v1.4_), we can recommend **all** the users of the original Sam's repos AlexaPi and AlexaCHIP to use the version from our repo. This is the one _feature_ that we've been missing so far and I believe there are people who don't want voice activation / the recognition engines are not available for their platform.
- Since all the triggers are now modules, we can very, **very** ea [... cut]
closed 2017-01-07 by renekliment