alexa-pi/AlexaPi #26: Refactor: Indications

sammachin, 2016-09-25:
I'd like to abstract the notifications out from the main code base into their own functions that then call whatever hardware we might be using, so I propose a number of functions like `ind_record_on` `ind_record_off` `ind_ready` etc.

renekliment, 2016-09-25:
We also need to figure out if and how we can do GPIO stuff on other SBCs (OrangePi and CHIP at least) under an unprivileged user. If it's not possible, we should do some GPIO proxy or something. I'd hate to have it running under root.

sammachin, 2016-09-25:
I'm going to setup a chip later as a test machine for this project. The one I used to start it all seems to be dead :(

@Renekliment do you want a CHIP? I've got a couple of extras the guys have me for the original project I could mail you one?

renekliment, 2016-09-25:
@sammachin I suppose I could use one for testing, even though I don't have any particular interest in CHIP. If the mailing (to Czech Republic) doesn't bother you, I'd be happy to see how it runs on this SBC. Please write me an e-mail (it's public on my profile) so we can discuss further.

commit 3e8bdd69 (2016-10-21): Device platforms - fixes #26 (#30)

* device platforms!

* enhanced long_press feature

* temporarily disabled voice confirmation of long_press

* made platform confirmation configurable

* bcolors in a separate module
closed 2016-10-21 by renekliment