alexa-pi/AlexaPi #27: Multicast DNS for Setup

sammachin, 2016-09-25:
mDNS https://en.wikipedia.org/wiki/Multicast_DNS Makes things a lot easier when creating the Amazon oAuth security profile.

Instead of using the IP address you can use the hostname of the device in the .local domain eg http://raspberrypi.local This way it doesn't matter if the IP address changes and there is less scope for confusion.

I suggest we change the setup script and docs to tell users to add raspberrypi.local and localhost as the 2 hostnames.

This will need some testing to ensure it works with the operating systems that users will be using on their desktops to browse from.
OS X is OK 
Windows 10 ?
Windows 8?
Windows 7/Vista/XP (do we care?)

Ubuntu 16:04 ??
Ubuntu 15:10 ??
Ubuntu 15:04 ??
Ubuntu 14:04 ??

ArchLinux ??

Fedora ??

Any others??

There's also a question around network support, I don't think that mDNS needs any special support in the router other than broadcast addresses not being blocked, this might not work on corporate LANs or other more managed networks but so long as its suported on the typical domestic setup I think we should go with it. Localhost is there as a fallback and using IP address for advanced users in the docs?.

renekliment, 2016-09-25:
If it's possible, we could independently advertise `alexapi.local` hostname.

sammachin, 2016-09-25:
I did a quick google and it didn't look like its possible to have mDNS respond to multiple hostnames, (not sure why)

I can see that in some setups eg where people are installing AlexaPi on an existing system we wouldn't want to change the hostname, but I think 99% of the users so far especially the less tech savvy have been using a brand new Pi.

How about we print the hostname to the screen at the start of the setup script and say 'goto developer.amazon.com to setup with this hostname` then once we've got the rest of the creds and start auth_web we can use that hostname.

I'm keen to offer a ready built card image anyway once we have a stable codebase for basic users to get started quick so in that I can use alexapi.local

renekliment, 2016-09-25:
Also, should we really care about the old Ubuntus? I mean 16.04 is an LTS, 16.10 is soon gonna be there. I'd personally give a damn from 16.04 onward.

sammachin, 2016-09-25:
fine by me I wasn't sure how far back its sensible to go with Ubuntu, looking at https://wiki.ubuntu.com/Releases 12.04.5 and 14.04 are both LTS and good for at least another 6 months otherwise its just 16.04. Although we are talking Desktop OS here so I would guess there's less to hold someone back from updating anyway.