alexa-pi/AlexaPi #29: "Can't reach Tune-In right now"

jaimbo, 2016-09-25:
There seems to be a problem with Tune-In support currently, I know this repo is in a state of flux currently but I figured I would raise an issue anyway.

Here is what happens when I run with the --debug flag:

`Checking Internet Connection...
Connection OK
Play_Audio Request for: /opt/AlexaPi/src/resources/hello.mp3
[01b19658] pulse audio output error: PulseAudio server connection failure: Connection refused
[01aae110] pulse audio output error: PulseAudio server connection failure: Connection refused
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
0Play_Audio Request for: /opt/AlexaPi/src/resources/alexayes.mp3
[01aac8a8] pulse audio output error: PulseAudio server connection failure: Connection refused
[01a65e90] pulse audio output error: PulseAudio server connection failure: Connection refused
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
detected the edge, setting up audio
Starting to listen...
Debug: End recording
Debug: Sending audio to be processed
Sending Speech Request...
Processing Request Response...
JSON String Returned: {"messageBody": {"directives": [{"namespace": "Speaker", "name": "SetMute", "payload": {"mute": false}}, {"namespace": "SpeechSynthesizer", "name": "speak", "payload": {"contentIdentifier": "amzn1.as-ct.v1.Domain:Application:Music#ACRI#DeviceTTSRendererV4_b0983f7b-66be-4845-aef4-262924e93cd2", "audioContent": "cid:DeviceTTSRendererV4_b0983f7b-66be-4845-aef4-262924e93cd2_923030554"}}]}, "messageHeader": {}}
Play_Audio Request for: /tmp/AlexaPi-runtime-0jPrP8/DeviceTTSRendererV4_b0983f7b-66be-4845-aef4-262924e93cd2_923030554.mp3
[01c6b720] pulse audio output error: PulseAudio server connection failure: Connection refused
[01c4fbf8] pulse audio output error: PulseAudio server connection failure: Connection refused
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
`

renekliment, 2016-09-25:
Does normal Alexa sound playback / recording work for you?
It looks like not working system audio setup.

We'll be hopefully adding some documentation / tools regarding setting up sound, because it's one of the trickiest things when setting up AlexaPi.

To you specific messages:
> [01c6b720] pulse audio output error: PulseAudio server connection failure: Connection refused

1. Are you using PulseAudio intentionally? 
2. Are you running AlexaPi under ordinary user (pi?), or are you trying to run it as a daemon (system service)? as root?
3. If as daemon, are you running PA in system mode?

sammachin, 2016-09-25:
There seems to be some wider problems with Tune in and Alexa at the moment I get the errors on my TAP too.

I'm guessing things are being moved around in preparation for he UK and DE launch of the echo this week so I think we should park any TuneIn issues until it's working reliably on the Amazon hardware

jaimbo, 2016-09-25:
@renekliment I have no issue with audio playback or recording, just specifically Tune-In.

1. I am not intentionally trying to use PulseAudio, ~~in fact, I even uninstalled it and still get that error and since It doesn't seem to actually cause any problems, I have just been ignoring it.~~ I had to reboot after uninstalling and that message is gone now.
2. I am just running under the user pi, but launching main.py with sudo.
3. As above, not daemon.

@sammachin That makes a lot of sense!

As a side note, I also get a lot of this device does not support *insert action here* is this due to using the older API?

jaimbo, 2016-09-25:
Update, Tune-In wasn't working because I changed my address on alexa.amazon.com to my UK address, changing back to a random US address fixes Tune-In!

renekliment, 2016-09-25:
Yes, that's correct. Mainly the alarms / timers are a things that we need the new API for. For Amazon music I think we need to have a certified device and I'm not sure that we can do that - just a guess though.

renekliment, 2016-09-25:
@jaimbo If it works for you then, please close this.

closed 2016-09-26 by jaimbo