alexa-pi/AlexaPi #32: Microphone Sensitivity

willsmithdev, 2016-09-28:
Hi,
I've finally got AlexaPi working after a few separate attempts due to various reasons.
It all _works_ but I literally have to shout at my microphone to get the keyphrase to be detected. How exactly do I change the sensitivity of my microphone? It's a decent microphone and I don't normally have any issues like this.

I realise this isn't directly related to AlexaPi but I have no idea what AlexaPi uses and how I'd change the settings for it...

Interestingly it seems much more responsive when I run it directly and not the always on service, but maybe thats just coincidence.

renekliment, 2016-09-30:
> Interestingly it seems much more responsive when I run it directly and not the always on service, but maybe thats just coincidence.

I don't see a reason for this to happen.

You can adjust the microphone sensitivity with the `alsamixer` program. If you're experiencing the issue only with the wake word, it's something we have to solve. We'll be adding some more configuration options regarding audio and support for _snowboy_, which should work better than _pocketsphinx_. We'll see how it goes.

chekeogh, 2016-10-05:
Thanks for all the work on this... installed and still tweaking it myself.
On a side I'm working with led strips to simulate the Echo activation light at the moment.

But in regards to microphone settings, I am using the PS3 Eye and trying to get it working with AlexaPi.
I seen it in Mason Stone's build... so I bought one on eBay for £5... works ok but am struggling to get it working with the array in .asoundrc adding up the 4 mics. Am sure that this will be something it will need to get working so it'll be better with far field voice recognition.

CEOGavin, 2016-10-22:
@willsmithdev Try changing the hotword in `/etc/opt/AlexaPi/config.yaml`. I had temporarily changed it from "Alexa" to "Max" and it had stopped dropping the hotword. Interesting to know that Alexa has 3 syllables ([https://www.howmanysyllables.com/words/alexa](url)) while Max has only 1 ([https://www.howmanysyllables.com/words/max](url)). Hence 3 times more error prone? ;)

renekliment, 2016-10-22:
Keep in mind that changing the trigger word might violate Amazon's ToS (#14).

illperipherals, 2016-10-22:
The most recent changes to AVS terms are as follows:

Changes posted August 23, 2016.
"We’ve updated the Alexa Agreement to remove the requirement that you obtain Amazon’s written approval prior to implementing far-field voice recognition or use of a spoken word to trigger the activation of the Alexa Service in Your Products, and to make other updates. Please review the full text of the updated Alexa Agreement carefully."

from https://developer.amazon.com/public/solutions/alexa/alexa-voice-service/support/terms-and-agreements#changes

renekliment, 2016-10-22:
@illperipherals That's voice activation. This is the wake word used for it.

> (p) you will not use or allow the use of any wake word or other trigger word as part of accessing the Alexa Service in Your Products other than the wake or trigger words authorized by Amazon (such as “Alexa”) without Amazon’s prior written approval; 

Read the thing.

illperipherals, 2016-10-22:
No need to be rude. I was just outlining the most recent changes. The guys over at Kitt were very careful with their release as well. Some of that conversation was on flooie's branch, I believe.

renekliment, 2016-10-22:
Sorry if that came out too strong. I merely wanted to suggest reading the whole document.

CEOGavin, 2016-10-22:
I think the interesting word here is "Your Product". It's not really a Product, is it (esp when it's unfinished)?

illperipherals, 2016-10-22:
De nada, man.

renekliment, 2016-10-22:
I'm actually not sure at all as I have suggested in #14. We left it at "we're developers, tinkers and we're just playing and if Amazon didn't like it, they would probably tell us".

illperipherals, 2016-10-22:
Also, Amazon helped fund Kitt, so those guys probably have some insight here.

Joe-houghton, 2016-11-10:
I am actually having the same issue as the original poster. 
The wake word hardly ever works, however in the very odd occasion that it does, my speech is recognised perfectly. 
I think it may be too do with accents. I am also from the UK. I was watching a video of someone  with an American accent demonstrate alexa and every time they said alexa on the video it triggered.

Coaxial, 2016-12-01:
Same issue here, wake word "Alexa" or "Computer" rarely works. I changed the keyword to "Bob" and it seems to work much better.

joe248, 2016-12-22:
I'm having trigger word issues running this on a CHIP using a mic in the 3.5mm jack.  If I start talking and speak a random sentence or two and then say "Alexa", the trigger word registers.  However, if I start with silence and then say "Alexa" or "Hey Alexa", it doesn't register.  It's like there has to be some amount of speech before the trigger word in order for it to trigger.  Has anyone else experienced this?  Any ideas how I can debug this further?

renekliment, 2017-01-09:
This issue should be addressed soon (tweaking pocketsphinx's threshold & adding snowboy support). Stay tuned.

the-winemaker, 2017-02-12:
hi!
this worked for me - voice recognition is now better, also with the Playstation Eye: At the website from kitt.ai - the makers of a wakeword-detector for alexa (also for ok google) I found an tutorial for the microphone-issue. Gave that a try - and: Now its really better. 

See information on that page: [http://docs.kitt.ai/snowboy/#set-up-audio](http://docs.kitt.ai/snowboy/#set-up-audio) 

I changed the ~/.asoundrc file as they described and then the rec-command worked out fine.

renekliment, 2017-03-01:
Version 1.4 has been released today, where PocketSphinx's threshold is configurable.

mano1979, 2017-03-02:
How can we set the pocketspinx threshold? Is there a config file?

maso27, 2017-03-02:
There is.  Look in _/etc/opt/AlexaPi/config.yaml_

EmerickH, 2017-06-15:
Like now threshold is configurable, I think I can close this issue.

Emerick

closed 2017-06-15 by EmerickH

rsanjeev525, 2019-01-24:
after pressing T on keyboard and enter button simultaneously then alexa works.
but not working when I tell alexa only