alexa-pi/AlexaPi #33: Feature List

willsmithdev, 2016-09-29:
Is there any possibility of creating a feature list that AlexaPi currently supports and in some cases to what extent it supports a particular features? I know the Alexa Voice Service provides a series of different API calls such as music and alerts. I realise that most of them work in a similar way but its difficult to know what is actually working at the moment. That way I know if I have a setup issue or if the particular feature isn't currently implemented.

Perhaps we could have a skills list that we define as working or not as the case may be.

renekliment, 2016-09-30:
Not in the near future. We haven't touched the actual Alexa stuff during the refactoring yet. When we get there and separate it into a module, we're gonna start work on supporting the new API, which will enable us to use alarms/timers and stuff like that. But that's quite the future. AFAIK, no one has started work on the new API.

renekliment, 2016-09-30:
And I think that when we have the new API support, we'll have it all.

closed 2016-09-30 by renekliment