alexa-pi/AlexaPi #34: filenames with : don't play with VLC

asmagill, 2016-10-01:
When requesting my flash briefing, I kept getting errors like this:

~~~
[57716ee8] core input error: open of `/tmp/AlexaPi-runtime-etC4np/DailyBriefingPrompt.51ed008e-1138-4784-a983-912fcc310e3c:Say:DAILYBRIEFING:DailyBriefingIntroduction_1235550225.mp3' failed
[57716ee8] core input error: Your input can't be opened
[57716ee8] core input error: VLC is unable to open the MRL '/tmp/AlexaPi-runtime-etC4np/DailyBriefingPrompt.51ed008e-1138-4784-a983-912fcc310e3c:Say:DAILYBRIEFING:DailyBriefingIntroduction_1235550225.mp3'. Che
ck the log for details.
~~~

On a hunch, since it referred to the path as an MRL, I wondered about the ":" characters in the filename and stuck "file://" in front of `tmp_path` at the following places in `main.py` and it seems to have fixed it:

~~~sh
# grep -n file: main.py 
267:                                            play_audio("file://" + tmp_path + directive['payload']['audioContent'].lstrip("cid:")+".mp3")
287:                                                            content = "file://" + tmp_path + stream['streamUrl'].lstrip("cid:")+".mp3"
315:                                    content = "file://" + tmp_path + stream['streamUrl'].lstrip("cid:")+".mp3"
~~~

I can submit this as a pull request if you prefer, but since I am running this on a Raspberry Pi 3 with Arch Linux, I wanted to first check and see if anyone else has seen this behavior (or is it something specific to my setup?) and get some input as to whether this is the proper fix... my experience with Python is extremely limited and new.

renekliment, 2016-10-01:
I can confirm this on my Arch Linux laptop.
Your fix seems appropriate. Feel free to open a PR.

Thank you!

renekliment, 2016-10-02:
Closed via #35.

And I should remember for the next time to edit commit messages so the closing happens automatically.

closed 2016-10-02 by renekliment