alexa-pi/AlexaPi #38: No Audio RPi3

eugenelai, 2016-10-09:
I've installed the package exactly as outlined however I'm getting the following error message:

Oct 09 23:25:11 raspberrypi python[576]: alsaaudio.ALSAAudioError: Input/output error [plughw:1]
Oct 09 23:25:11 raspberrypi systemd[1]: AlexaPi.service: main process exited, code=exited, status=1/FAILURE
Oct 09 23:25:11 raspberrypi systemd[1]: Unit AlexaPi.service entered failed state.
Oct 09 23:25:12 raspberrypi systemd[1]: AlexaPi.service holdoff time over, scheduling restart.
Oct 09 23:25:12 raspberrypi systemd[1]: Stopping Alexa client for all your devices...
Oct 09 23:25:12 raspberrypi systemd[1]: Starting Alexa client for all your devices...
Oct 09 23:25:12 raspberrypi systemd[1]: Started Alexa client for all your devices.
Oct 09 23:25:17 raspberrypi python[1321]: Checking Internet Connection...
Oct 09 23:25:17 raspberrypi python[1321]: Connection OK
Oct 09 23:25:17 raspberrypi python[1321]: Home directory not accessible: Permission denied
Oct 09 23:25:17 raspberrypi python[1321]: W: [pulseaudio] core-util.c: Failed to open configuration file '/home/alexapi/.config/pulse//daemon.conf': Permission denied
Oct 09 23:25:17 raspberrypi python[1321]: W: [pulseaudio] daemon-conf.c: Failed to open configuration file: Permission denied
Oct 09 23:25:17 raspberrypi python[1321]: Home directory not accessible: Permission denied
Oct 09 23:25:17 raspberrypi python[1321]: W: [pulseaudio] core-util.c: Failed to open configuration file '/home/alexapi/.config/pulse//daemon.conf': Permission denied
Oct 09 23:25:17 raspberrypi python[1321]: W: [pulseaudio] daemon-conf.c: Failed to open configuration file: Permission denied
Oct 09 23:25:18 raspberrypi python[1321]: Home directory not accessible: Permission denied
Oct 09 23:25:18 raspberrypi python[1321]: W: [pulseaudio] core-util.c: Failed to open configuration file '/home/alexapi/.config/pulse//daemon.conf': Permission denied
Oct 09 23:25:18 raspberrypi python[1321]: W: [pulseaudio] daemon-conf.c: Failed to open configuration file: Permission denied

Anyone have any ideas?

renekliment, 2016-10-10:
Hey, thanks for reporting. I'll work on a fix regarding PulseAudio when I find some time.

Does it work though? You can just ignore the messages for now if it does. If it doesn't, try tweaking the audio device in the configuration file. You can give _default_ a shot.

lsumner4, 2016-10-19:
I'm having a lot of issues setting up my audio as well. My setup is outputting audio over the analog jack and using a USB microphone. I've tried changing the audio device within AlexaPi's configuration and I've changed the default devices of alsa to be my usb microphone. My setup changes have either stopped AlexaPi from working entirely or have only allowed me to use it after starting in manually on boot.

renekliment, 2016-10-19:
@lsumner4 Are you using PulseAudio? We have yet to fix the support for PA. Please provide logs.

lsumner4, 2016-10-19:
Yes, I'm using pulse audio. Is there any other option on Raspberry Pi?
My log on startup now:
Oct 19 13:13:59 raspberrypi systemd[1]: Starting Alexa client for all your devices...
Oct 19 13:13:59 raspberrypi systemd[1]: Started Alexa client for all your devices.
Oct 19 13:14:22 raspberrypi python[737]: Home directory not accessible: Permission denied
Oct 19 13:14:22 raspberrypi python[737]: W: [pulseaudio] core-util.c: Failed to open configuration file '/home/alexapi/.config/pulse//daemon.conf': Permission denied
Oct 19 13:14:22 raspberrypi python[737]: W: [pulseaudio] daemon-conf.c: Failed to open configuration file: Permission denied
Oct 19 13:14:22 raspberrypi python[737]: Home directory not accessible: Permission denied
Oct 19 13:14:22 raspberrypi python[737]: W: [pulseaudio] core-util.c: Failed to open configuration file '/home/alexapi/.config/pulse//daemon.conf': Permission denied
Oct 19 13:14:22 raspberrypi python[737]: W: [pulseaudio] daemon-conf.c: Failed to open configuration file: Permission denied
Oct 19 13:14:22 raspberrypi python[737]: Home directory not accessible: Permission denied
Oct 19 13:14:22 raspberrypi python[737]: W: [pulseaudio] core-util.c: Failed to open configuration file '/home/alexapi/.config/pulse//daemon.conf': Permission denied
Oct 19 13:14:22 raspberrypi python[737]: W: [pulseaudio] daemon-conf.c: Failed to open configuration file: Permission denied

bchuckles, 2016-10-19:
Ditto and i hear nothing....
Home directory not accessible: Permission denied
Oct 19 21:33:40 raspberrypi python[548]: W: [pulseaudio] core-util.c: Failed to open configuration file '/home/alexapi/.con
Oct 19 21:33:40 raspberrypi python[548]: W: [pulseaudio] daemon-conf.c: Failed to open configuration file: Permission denie
Oct 19 21:33:40 raspberrypi python[548]: Home directory not accessible: Permission denied
Oct 19 21:33:40 raspberrypi python[548]: W: [pulseaudio] core-util.c: Failed to open configuration file '/home/alexapi/.con
Oct 19 21:33:40 raspberrypi python[548]: W: [pulseaudio] daemon-conf.c: Failed to open configuration file: Permission denie
Oct 19 21:33:40 raspberrypi python[548]: Home directory not accessible: Permission denied
Oct 19 21:33:40 raspberrypi python[548]: W: [pulseaudio] core-util.c: Failed to open configuration file '/home/alexapi/.con
Oct 19 21:33:40 raspberrypi python[548]: W: [pulseaudio] daemon-conf.c: Failed to open configuration file: Permission denie
~

I would like to use this in a presentation coming soon, is there a known work around?

renekliment, 2016-10-19:
@bchuckles Haven't tried it yet, but
1. comment out `ProtectHome=true` in _/usr/lib/systemd/system/AlexaPi.service_. You'll probably have to do that as the user _root_.
2. if there is no directory _/home/alexapi_, do the following (under _root_):
```
mkdir /home/alexapi
chown alexapi:alexapi /home/alexapi
```

Then you should be able to run AlexaPi normally as a service under the _alexapi_ user.
But as I said, I haven't tested it, it's just what I would try. I need to debug this and find a sensible solution instead of this hack.

bchuckles, 2016-10-19:
@renekliment yes this might work in theory...maybe ill wait lol. With that said i guess i will have to switch it back to a pi3 since the zero doesnt have an audio out and the usb sound dongle is PulseAudio

renekliment, 2016-10-19:
@bchuckles Check out the updated comment to see if that works.

> usb sound dongle is PulseAudio

What does that mean?

lsumner4, 2016-10-19:
@renekliment I uninstalled Pulse Audio entirely and that seemed to solve my issue. Perhaps recommend this in step in the ReadME if it works for other people?

renekliment, 2016-10-19:
@lsumner4 Well, that's not a solution, that's a workaround. We need to make a fix so people _can_ use PA if they want to.

Could you maybe test that one command with PA to see if it works? I don't have much time these days to try it out myself.

illperipherals, 2016-10-19:
@bchuckles, I use a C-Media usb dongle and a wifi adapter off of a micro-usb hub on the Zeros. You have to make sure that you are using a strong enough power source when you do this, though. Otherwise, you tend to see your wireless drop out on you. I am not using pulseaudio at all at this point, and it is working well.

bchuckles, 2016-10-20:
Sorry I was rushed when i was typing. I want to install on a zero i was on a 3! So maybe if i get some time tonight ill try on a zero and see how it works!

bchuckles, 2016-10-20:
By the way this is the coolest project ever for Alexa and fits my use case perfect

renekliment, 2016-10-21:
So I guess we need to solve these situations:

1. AlexaPi with PA in system mode - this is most probably headless systems
  - I guess here we just add _alexapi_ in the _pulse-access_ group
2. AlexaPi with PA in user session mode
  - we have to solve some _d-bus_ stuff that happens probably because it's run as a service and not as a user session
3. AlexaPi on systems that run GUI with the user _pi_ with PA in user session mode
  - how the fu** do we solve this?
  - we'd have to reconfigure PA to communicate over TCP / UNIX socket 
  - if the user _pi_ isn't logged in, then what?

Please let us know which scenario is yours.
Suggestions welcome.
If you don't use any audio apps on your pi, you can temporarily uninstall PA.

illperipherals, 2016-10-21:
No, I had initial issues on the Zero. Much had to do with how I wanted to handle the sound/mic issue. You are using the same card I am on the Zero, though, so I know it will work.

bchuckles, 2016-10-21:
@illperipherals would u like to share what you did to handle the issues?

illperipherals, 2016-10-21:
I would love to. I want to start another issue, as we are getting off-topic of PulseAudio.

renekliment, 2016-10-23:
So an alternative to uninstalling PA is to disable it's autostart. This will work if you don't run PA in system mode and don't have for example GUI running with the user _pi_ running PA in user-session mode (basically if you don't have PA running by other means):
```
# all under root
mkdir -p /var/lib/AlexaPi/.config/pulse
chown -R alexapi:alexapi /var/lib/AlexaPi/
cp /etc/pulse/client.conf /var/lib/AlexaPi/.config/pulse/
# edit /var/lib/AlexaPi/.config/pulse/client.conf and set "autospawn = no"
```

Any input on this topic is appreciated.

jamesliin, 2016-11-04:
I have been kicking around this Audio issue for a few days, and found this link that did the trick for me.  It involves modifying the the alsa.conf file to work properly as the default audio devices.  

http://raspberrypi.stackexchange.com/questions/39928/unable-to-set-default-input-and-output-audio-device-on-raspberry-jessie?rq=1

renekliment, 2016-11-04:
@jamesliin If you just need to set the input / output audio device, you can do it in AlexaPi config now, you don't need to modify system config anymore. #22 solved this.

jamesliin, 2016-11-04:
@renekliment No matter what config I tried in the AlexaPi config,(even wiped the sd card and started over a few times) it would not use the USB sound card.  Using Puredata I could test, and use the mic and speakers properly, However, the Hello from Alexa would only run nonstop from the onboard audio jack.   
However, I am still new to this and am probably missing a some other things.  
Thanks

renekliment, 2016-11-04:
@jamesliin Just to be sure, you were setting the new option **output_device** in the config to some value like `plughw:1`, right?

jamesliin, 2016-11-04:
@renekliment  Correct, every combination of plughw:1, plughw:0, system, default, etc...   

When configured under default, the hello sound would only come from the onboard,  But in Puredata it would have a Resource busy errors

renekliment, 2016-11-09:
So I guess the only sensible solution of the PA situation is to run PA in system-wide mode (apart from uninstalling it). If any one of you has some time, it would be cool, if you could write up a how-to in the wiki. We can somehow add it in the setup script later, when we've got that figured out the best way possible. Someone already posted some commands on Gitter. We shouldn't forget add _alexapi_ in the _pulse-access_ group and the user _pi_ on a Raspberry Pi also.

Any thoughts on this?

Felipe-Rubin, 2016-11-10:
Came from the other post.

I "fixed" a part of mine: 

I followed a tutorial to install PulseAudio and it wasn't just PS itself that was installed, other packages suck as PulseAudio-utils and some bluetooth related needed to be removed. Now this issue is not seen in Journal. 

After that AlexaPi keeps restarting because of another error. 
Right now I don't know if the error is because PulseAudio changed something in Alsa or the reason being I don't have a microphone connected to it right now.

But it is Microphone related.

Later I'll see if connecting a microphone fixes it. 

One question though: Doesn't Alexa have something like
"If Input Device is connected
then Start Alexa
else idle"
?

illperipherals, 2016-11-11:
I think it may help out in debugging and tracking down audio issues if we use this as a standard form of data collection:
Enter these two commands in the terminal:
cd ~/
wget http://www.alsa-project.org/alsa-info.sh -O alsa-info.sh && bash alsa-info.sh

revere521, 2016-11-18:
Hello, been having alot of fun with this!  i just wanted to note that i have been running AlexaPi with pulseaudio for a few days.  I'm doing this primarily because the the PS3 Eye does not play well with Alsa.  here's how i have been doing it: Raspberri Pi 2 model B (also running MagicMirror2, so not headless)
1. run pulseaudio as a service in system-wide mode - i can post the content of my pulseaudio.service file if you guys want it
2. setting pulse as the input_device and output in config.yaml
3. added alexapi and pi to pulse-access group

it works, and i can set the volume sliders with pavucontrol, and the array mic gives me far fewer false positives than a single mic.   i have "Magic Mirror" set as my wake word/phrase currently just for giggles.

I do have some crash restarts occasionally with pulseaudio running that i have not diagnosed - it seems to be in relationship to two things 1. the alexpi service still complains about home directory permissions over and over - but it does not seem to really mean anything.  2. every so often i get a mmap memory error for alexapi.  let me know if you want any of my logs of config files to help - i think having the ability to run pulseaudio gives more choices for mics, as alsa cant read from multiple sources afaik, it has to sum them.

renekliment, 2016-11-18:
@revere521 Great to hear you're _kinda successfully_ running PA!

If you could write a simple and straightforward how-to in the [audio wiki section](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging), that would be most awesome. 

ad 1. The home directory errors should be solved by #82 (clean install only). You can do something like 
```
mkdir -p /var/lib/AlexaPi/
chown -R alexapi:alexapi /var/lib/AlexaPi/
usermod --home /var/lib/AlexaPi alexapi
```
which should get rid of it.

ad 2. That looks like something we should solve. Please file an issue containing the service log (journal log or log from when running AlexaPi interactively in debug mode).

Thanks!

revere521, 2016-11-18:
Ok, before I open a new issue - I tried your solution on my existing install and it appears to have resolved the home drive permission errors in the log.  very cool solution to put a home directory for the AlexaPi in /var/lib - i'm no linux power user or anything, but that did not occur to me.    I will monitor it overnight and see what happens.  I have saved the before and after journal log for reference.

revere521, 2016-11-18:
Er.... its still working fine so.... i went ahead and added what i did to the audio debugging page - please feel free to proofread and fix or delete anything you like,  i can be a little wordy.

revere521, 2016-11-20:
still working, although i realize for pavucontrol to work properly in the GUI i had to add some configs to the asoundrc for the PS3 Eye.  I'm not sure if thats unique to me, but i also set the default devices to pulse. so maybe, if someone tries it without the eye, and has trouble they may have to do that.

illperipherals, 2016-11-20:
I am running through the PA guide using a PS3 Eye. 
Here is an interesting article on the usage a friend sent me:
http://www.labbookpages.co.uk/audio/beamforming/delaySum.html

renekliment, 2016-11-20:
@revere521 Thank you for the HowTo! I've made slight changes to the look and shortened it a bit, because people don't like to read :disappointed: For some reason the _double dashes_ in the unit file were transformed to a single long dash. I've fixed that and I'm running it fine now according to the instructions. I've compared it to my own fine-tuned PA system-wide install and it seems nothing is missing. Great job, thank you!

Also, I've put some other options discussed here over to the wiki and if there are no issues appearing regarding PA when properly following the guide I think we can close this. What do you think?

renekliment, 2016-11-20:
@eugenelai @lsumner4 @bchuckles Can you confirm AlexaPi works fine under [system-wide PA](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#running-pa-in-system-wide-mode-recommended)?

revere521, 2016-11-20:
@renekliment no problem at all, and thanks for the double dashes fix (i completely missed that) and cleanup.   Its still working fine for me as well. 

@illperipherals i am running the echo cancel  module (set `load-module module-echo-cancel aec_method=webrtc`  in `/etc/pulse/system.pa`) primarily to cancel noise from my setup's own speaker.  That module does have configuration for beam forming, although haven't been able to get it to work yet.  The module options are listed here: https://www.freedesktop.org/wiki/Software/PulseAudio/Documentation/User/Modules/#index45h3.

renekliment, 2016-11-20:
@revere521 The PA module looks absolutely dope! It could probably even fix #74.

revere521, 2016-11-20:
@renekliment  just realized that i also needed to install the library for processessing - 'sudo apt install libwebrtc-audio-processing-dev'  I'm still doing research on the best implementation for me, primarily to mitigate the false positives i get with the 'echo' wake word.  using 'magic mirror' or 'mirror mirror'  helps alot, but it also makes it harder to get it to respond.

renekliment, 2016-11-20:
@revere521 Try another one-syllable word. Someone has been successfully using it with the word _Max_. We should get better triggering with snowboy (#53), but we need to solve a couple of other things first.

maso27, 2016-11-27:
I took a shot at using PulseAudio on the CHIP, following the steps in the wiki.  I wanted to see if the default audio card worked better with PA than alsa.

I haven't had much luck, but I wanted to point out that it might be worth changing the order of the steps there where we copy /etc/pulse/client.conf _before_ installing pulseaudio.  Shouldn't those steps be swapped?

Also, in my case the default client.conf did not have `autospawn=no` set.  But since there was a note there it informed me that I should do it.  So thanks!

Anyway, when I try to run main.py after configuring for pulse, I get these errors:
```
Play_Audio Request for: /opt/AlexaPi/src/resources/hello.mp3
Player State: State.Opening
Player State: State.Playing
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Access denied

[00e288a0] alsa audio output error: cannot open ALSA device "default": Connection refused
[00e288a0] core audio output error: Audio output failed
[00e288a0] core audio output error: The audio device "default" could not be used:
Connection refused.
[00e288a0] core audio output error: module not functional
[b4612bd0] core decoder error: failed to create audio output
Player State: State.Ended
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Access denied

Traceback (most recent call last):
  File "/opt/AlexaPi/src/main.py", line 598, in <module>
    loop()
  File "/opt/AlexaPi/src/main.py", line 509, in loop
    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, config['sound']['input_device'])
alsaaudio.ALSAAudioError: Connection refused [pulse]
```

There could easily be some user error here, but thought I'd put this out there in case it covered an edge case, or if you folks who have been playing with PA can just say, "you forgot to do ______" or something.

renekliment, 2016-12-07:
@maso27 Sorry, I totally forgot about this. I'll try a new install when I have some time. It's possible that I've instinctively done an additional step even when following the wiki guide. This is something we need to solve, since other people also have issues with this.

renekliment, 2016-12-07:
Came across the issue today and couldn't figure it out for some time. 
If you are running the root override (like on a CHIP or OPi), you also need to add the user _root_ into the _pulse-access_ group: `sudo adduser root pulse-access`.

maso27, 2016-12-08:
Ha!  Yeah, that makes total sense.  I added _root_ to _pulse-access_ and it starts without permission errors now.
But I'm remote right now so I can't hear whether or not it's different on the default audio output.

I'll try that when I get the chance.

Thanks @renekliment. Oh, and no worries about not looking at it.  I hadn't been thinking about it either.

renekliment, 2016-12-11:
I've put the notes about root override & PA in the wiki and swapped the steps, so it tells the user to install PA first before using its files. I think we're all done here. Closing as fixed.

closed 2016-12-11 by renekliment