alexa-pi/AlexaPi #39: Everything but hotword detection works

coach0742, 2016-10-10:
Running the latest code, and I have everything but the hotword detection working. If I use the button, everything is great, but for some reason, no matter what I do, it wont recognize "Alexa"  
  
I'm not even sure where I can start with troubleshooting.  
  
I am using a Blue Snowball microphone which is having no problems when saying any commands, but nothing happens at all with the hotword.  
  
Any ideas on where I can start?

eugenelai, 2016-10-10:
Have you tried to have a look at the error log?

sudo journalctl -u AlexaPi.service

then hit ESC then > to scroll to the end of the file

renekliment, 2016-10-10:
My personal guess is the microphone sensitivity.

Try running it from the shell in debug mode and provide the output to rule out other causes:
```
systemctl stop AlexaPi.service
/opt/AlexaPi/src/main.py -d
```