alexa-pi/AlexaPi #43: memcached → sqlite

renekliment, 2016-10-15:
From what I see, we use it only to store the `access_token` and it makes no sense to force the user to install it and have it running.

We can easily replace it with a local sqlite database, which we can later use to also store timers, alarms, etc. The main thing here is to carefully think about where to store the DB file.

sammachin, 2016-10-15:
> @sammachin <https://github.com/sammachin> Is there any particular reason
> for why you've chosen *memcached* originally?

Reasons I used memcache were:
)1 I was familiar with it as used it on lots of other projects
2) the access token expires after 1hr so when I set it I set an expiry of
just under (3590s IIRC) so it's a fairly easy way to check for an access
token and if there isn't one or it has expired it requests a new one with
the request token.
I'd rather not have to do our own time calaculations to determine if a
token has expired and making the request to find the token has expired then
generate a new one and repost the request would be a huge increase in delay.

renekliment, 2016-10-15:
I see. So it basically makes no sense to store it anywhere (permanently). There is not much reason in doing this so someone that restarts AlexaPi can keep using the existing token. It might make sense for developers, because they might restart it often, but I personally don't have _memcached_ installed on my dev machine so it probably fetches it every time, but it works without a problem.

Let's just store it in a variable together with the timestamp it was acquired. The comparison of the time is easy and we won't have to do requests with an expired token. We'll just expire it locally if the time has passed and get a new one.

sammachin, 2016-10-15:
Ok makes sense yeah store it locally with the time stamp and check that
it's under 3590s old.

The only scenario where mc helps is if you restart the python app but not
the system it will persist but like you said that's an edge case and
fetching a new access token isn't hugely expensive on its own, although I
expect AVS wouldn't like it if we all did it on every request. Limiting the
amount the refresh token is passed over the wire also helps with security

renekliment, 2016-10-15:
Yes, I'm aware. My main concern is getting rid of _memcached_.
I might have a look at the sqlite stuff then. It's probably super easy.

renekliment, 2016-12-30:
Currently, there is no clear idea how (if) we're gonna use sqlite for storing stuff like timers, etc, so let's not go there now. I don't think it's a problem, getting the token each time AlexaPi restarts. Let's just get rid of memcached and we can do this when we know how we want to do this.

lechat, 2016-12-31:
Why not just keep that in a text file somewhere? OS provides timestamping and we don't need no SQL to find that file.

renekliment, 2016-12-31:
Yeah, I thought about that too and then I thought why bother at all. Do you think there's significant advantage in storing it on the filesystem rather than just in a variable (with a timestamp)? I consider the token run-time.

commit 3cf8f3ae (2017-01-17): Get rid of memcached & store the token in runtime memory (#138) fixes #43

* store the AVS token in memory instead of memcached

* improve error handling

* remove memcached dependency
closed 2017-01-17 by renekliment