alexa-pi/AlexaPi #44: Streaming audio to and from AVS

dly9014, 2016-10-17:
While waiting for Echo push notifications for a year now, I did a small extension of https://github.com/sammachin/AlexaPi to essentially enable push by submitting pre-recorded questions to AVS when triggered by an event. It worked pretty well, but the problem I faced is that response time is slow, typically 5 or even 10 seconds. I believe the approach used in https://github.com/alexa/alexa-avs-sample-app address those performance issues by streaming the audio to AVS (and maybe also streaming the response back, I asked that question, hoping to hear back from them). The approach is explained here - https://forums.developer.amazon.com/questions/10982/avs-takes-too-much-time-to-respond.html. 

Is streaming audio to and from AVS in the plan for the new evolving AlexaPi?

renekliment, 2016-10-17:
Sounds interesting. We're not there yet to do these _future_ things - we need to refactor the AVS code first, but we ain't that far either. Let's keep it here for when we get there. 

Also, I'm not sure that streaming **from** AVS is gonna have significant performance impact, because Alexa's answers are usually very small filesize-wise. Streaming **to** Alexa is probably gonna have noticeable impact, because from my experience, upload speed is often crap.

dly9014, 2016-10-23:
Here is a terrific example of implementing streaming to AVS using the v1 API - https://github.com/respeaker/Alexa/blob/master/alexa.py

renekliment, 2017-01-09:
If anyone implemented this, that would be awesome.

dgonano, 2017-01-09:
I thought someone was working on the new API ?

renekliment, 2017-01-09:
Yep, but we might do this for the old also, since it might not be such a big change (I dunno, just guessing) and the new API integration + testing + setting it as default is gonna take some time.

renekliment, 2017-03-04:
@alexa-pi/team 

Okay, I got this working (streaming to Amazon) with our AlexaPi - basically just copied a small portion of code from @respeaker and holy shit! If you don't have an Echo / Dot or haven't tried ReSpeaker you will not believe how fast it can get! I'll open a PR today and I'd like to release it this weekend as a new version (and bump all those planned versions by one), because this is definitely worth it. Stay tuned!

gemx, 2017-03-04:
Awesome. Count me in if you need someone to test it :-)

commit 1f705d24 (2017-03-06): stream audio to AVS instead of recording the whole thing and then sending at once

* Brings huge speed improvement (response latency).

* This means that when your recording LED (or whatever your device equivalent) is on, data gets sent to Amazon already.

* From https://github.com/respeaker/Alexa/blob/master/alexa.py

* fixes #44
closed 2017-03-06 by renekliment