alexa-pi/AlexaPi #45: pyalsaaudio → pyaudio

renekliment, 2016-10-17:
Let's switch the libs to be multi-platform here. Like in [Mason's snowboy branch](https://github.com/maso27/AlexaPi/blob/snowboy/main.py).

ImEmJay, 2017-01-06:
+renekliment Getting rid of pyalsaaudio is a good idea. if no one else has already started i would have a look at it. but do we need to support pyaudio and sounddevice? As far as i see they both wrap the PortAudio library. Snowboy and pocketspinx can both work with them.

renekliment, 2017-01-06:
@ImEmJay go ahead :-)

Note, that the code for pocketsphinx has been moved to triggers and the code around normal recording has been changed a lot - all in a PR #119. So it may be best to wait after the PR is merged, which should be today.

Probably not, I was just thinking out loud.

Whether to do pyaudio or sounddevice ...
- snowboy uses pyaudio for examples
- this speaks in favor of sounddevice: https://www.reddit.com/r/Python/comments/3k11g5/whats_a_good_sound_recording_library/cuu6gvy/

I would go with sounddevice, but it's up to you.

ImEmJay, 2017-01-06:
@renekliment I also think that pyaudio is more legacy and sounddevice is more under active development. but could have been that you know of some huge differences in platform compatibility or such ^^ but as they both just wrap i think that shouldn't really be an issue.

renekliment, 2017-01-06:
Okay. Let's go with sounddevice then.

renekliment, 2017-01-13:
For the record, the messy PRs (#119 & #110) have been merged, so there should generally be no major conflicts when editing the recording code.

renekliment, 2017-01-18:
Hey, just checking in. Any progress on this?

ImEmJay, 2017-01-18:
@renekliment hey, had no time over the last week. Think I'll get it done this weekend. But after my initial tests I would maybe change my decision and go with pyaudio. Sounddevice brings in a good amount of dependencies and propably unnecessary complexity. But will see when I finally get into it.

ImEmJay, 2017-01-18:
@renekliment Because of your comment on #152. Checking and listing devices can be done in both and was planned to add another PR to improve setting/checking the input device.

renekliment, 2017-01-18:
> unnecessary complexity

What do you mean by that exactly?

> good amount of dependencies

Do you mean NumPy? I thought it's optional and we don't have to install it.

renekliment, 2017-01-18:
@ImEmJay Do you know of a good way to check if a device name is valid in pyalsaaudio? Other than trying to open it.

ImEmJay, 2017-01-18:
@renekliment NumPy is optional. More the Python and system libraries for CFFI and their dependencies. Also seemed like you need to cast the input before pocketsphinx can handle it. But only gave them both a very short test drive to compare them a bit.
Regarding pyalsaaudio: Not atm. Only had a short look at their API till now.

renekliment, 2017-02-06:
All the alsaaudio stuff from `main.py` has been moved to an extra module in #156 so all the recording / alsaaudio stuff is in the `capture` module and the `pocketsphinxtrigger` module. 

If you have any work, please rebase on top of that. This makes more sense and as a bonus, we get no merge conflicts with the _main.py stuff_.

Also, you can consider moving the library-dependent stuff from the pocketsphinx trigger to the capture module as suggested in #156.

commit a0aca4d0 (2017-03-25): switch from pyalsaaudio (ALSA) to pyaudio (PortAudio), fixes #45

* we want to be platform-independent

* removes the option to allow unlisted devices as we don't just pass whatever string name to the underlying library, but we have to select a device from the device list, but there should be no issues with this

* voice triggers don't interact with the audio library directly, but through alexapi.capture
closed 2017-10-08 by renekliment