alexa-pi/AlexaPi #47: Crashing after Alexa's "yes"

ckylape, 2016-10-20:
I'm getting a crash immediatelly after Alexa's "yes" response. Any suggestions?

```
('setwarnings', (False,), {})
('cleanup', (), {})
('setmode', ('BCM',), {})
('setup', (18, 'IN'), {'pull_up_down': 'PUD_UP'})
('setup', ([24, 25], 'OUT'), {})
('output', ([24, 25], 'LOW'), {})
Checking Internet Connection...
Connection OK
('output', (24, 'HIGH'), {})
('output', (24, 'LOW'), {})
('output', (24, 'HIGH'), {})
('output', (24, 'LOW'), {})
('output', (24, 'HIGH'), {})
('output', (24, 'LOW'), {})
('output', (24, 'HIGH'), {})
('output', (24, 'LOW'), {})
('output', (24, 'HIGH'), {})
('output', (24, 'LOW'), {})
Play_Audio Request for: /opt/AlexaPi/src/resources/hello.mp3
('output', (24, 'HIGH'), {})
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
('output', (24, 'LOW'), {})
('add_event_detect', (18, 'FALLING'), {'callback': <function detect_button at 0xb64a5f0c>, 'bouncetime': 100})
Play_Audio Request for: /opt/AlexaPi/src/resources/alexayes.mp3
('output', (24, 'HIGH'), {})
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
('output', (24, 'LOW'), {})
detected the edge, setting up audio
Starting to listen...
Traceback (most recent call last):
  File "main.py", line 603, in <module>
    loop()
  File "main.py", line 558, in loop
    silence_listener(VAD_THROWAWAY_FRAMES)
  File "main.py", line 469, in silence_listener
    isSpeech = vad.is_speech(data, VAD_SAMPLERATE)
  File "/usr/local/lib/python2.7/dist-packages/webrtcvad.py", line 27, in is_speech
    return _webrtcvad.process(self._vad, sample_rate, buf, length)
webrtcvad.Error: Error while processing frame
```

ckylape, 2016-10-20:
It seems that commenting out [Line 469 of main.py](https://github.com/alexa-pi/AlexaPi/blob/master/src/main.py#L469) fixes the issue?

renekliment, 2016-10-20:
Interesting. I haven't experienced any such crashes, nor know why it does crash.
However, that line doesn't seem to be of any use, so I think we can safely remove it.

@maso27 might know a little bit more about it?

renekliment, 2016-10-20:
I see you're using the _fakeRPiGPIO_ package. Out of curiosity, what OS / device are you using?

ckylape, 2016-10-20:
It is an older netbook, running 32bit Ubuntu Server 16.04.01. 

The model is an HP Mini 210-1100TU, and the specs can be found [here](http://h20564.www2.hp.com/hpsc/doc/public/display?docId=emr_na-c02031689).

maso27, 2016-10-20:
Yeah, I don't think that line is doing anything either.  
It's in a loop that's just dumping the first 10 frames, and it's really not needed for that.

That was part of an evolution from someone else's code, so I can't be certain, but we can probably just take it out altogether.

renekliment, 2016-10-20:
@ckylape Do you want to open a pull request?

renekliment, 2016-10-23:
Please confirm that #61 fixes this for you. Beware of the configuration changes. Thanks!

renekliment, 2016-11-01:
@ckylape So the line is removed in `master` now. If you have any other issues, please report.

commit 174eeeb1 (2016-11-01): removed excess line, fixes #47
closed 2016-11-01 by renekliment