alexa-pi/AlexaPi #49: Success Install but errors?

fboulegue, 2016-10-20:
Hi i could install all the stuff via SSH, then it did ask me to authorize, that's what i did. So i did close the script after with "ctrl + c" script ended but nothing happened.
So i did try to run the

**python main.py**
```
Traceback (most recent call last):
  File "main.py", line 84, in <module>
    decoder = Decoder(config)
  File "/usr/local/lib/python2.7/dist-packages/pocketsphinx/pocketsphinx.py", line 228, in __init__
    this = _pocketsphinx.new_Decoder(*args)
RuntimeError: new_Decoder returned -1
```

So i also did try to run

** sudo ./monitorAlexa.sh**
but the log show

```
Starting Alexa in silent mode...
python: can't open file '/root/AlexaPi/main.py': [Errno 2] No such file or directory
```

let me add i use @maso27 fork

renekliment, 2016-10-20:
Are you actually using the code from this repo or are you just reposting your issue here?

fboulegue, 2016-10-20:
i just did a fresh install from that repo :)

renekliment, 2016-10-20:
And?

fboulegue, 2016-10-20:
```
pi@AmayonEchoPI:/opt/AlexaPi $ sudo journalctl -u AlexaPi.service
-- Logs begin at Do 2016-10-20 16:17:01 UTC, end at Do 2016-10-20 17:04:33 UTC. 
Okt 20 16:17:05 AmayonEchoPI systemd[1]: Starting LSB: AlexaPi Service...
Okt 20 16:17:06 AmayonEchoPI systemd[1]: Started LSB: AlexaPi Service.
lines 1-3/3 (END)

```

renekliment, 2016-10-20:
Damn you PulseAudio guys :smiley: I have to fix that - a lot of people want to use PA. It's a known bug #38. Can you try the suggested fix? `sudo usermod --home-dir /run/AlexaPi alexapi` and restart ... provide the logs then please.

fboulegue, 2016-10-20:
mmh `sudo usermod --home-dir /run/AlexaPi alexapi` give me no output let me see ...

renekliment, 2016-10-20:
That's okay. Standard programs don't output anything if everything went fine. Just restart and see if AlexaPi works.

fboulegue, 2016-10-20:
still tuck on same messages as before :/

renekliment, 2016-10-20:
Alright, I'll find some time to fix it.

fboulegue, 2016-10-20:
but if i see right `sudo usermod --home-dir /run/AlexaPi alexapi` did not have any effect, it not changed the home dir

renekliment, 2016-10-20:
Just to know your environment:
- Do you have the GUI up, or just console startup?
- If GUI, are you logged in on the pi in GUI or just ssh?

Send output of `sudo grep alexapi /etc/passwd`

fboulegue, 2016-10-20:
GUI is up via VNC but all commands been sended via SSH from other device ...
`alexapi:x:998:995::/home/alexapi:/bin/bash`

renekliment, 2016-10-20:
Alright, 
```
systemctl stop AlexaPi.service
usermod --home /run/AlexaPi alexapi
systemctl stop AlexaPi.service
```

How about this?

fboulegue, 2016-10-20:
After reboot

```
Okt 20 17:43:04 AmayonEchoPI systemd[1]: Starting Alexa client for all your devices...
Okt 20 17:43:04 AmayonEchoPI systemd[1]: Started Alexa client for all your devices.
Okt 20 17:43:32 AmayonEchoPI pulseaudio[1372]: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch
Okt 20 17:43:32 AmayonEchoPI pulseaudio[1372]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-d
Okt 20 17:43:32 AmayonEchoPI pulseaudio[1372]: [pulseaudio] bluez4-util.c: org.bluez.Manager.GetProperties() failed: org.freedesktop.DBus.Error.UnknownMethod: Meth
Okt 20 17:43:34 AmayonEchoPI python[547]: [0295ac48] alsa audio output error: cannot estimate delay: Input/output error
```

renekliment, 2016-10-21:
Let's mark this as a duplicate of #38 so we can better track it. 

If you don't use other sound applications on the pi I suggest you remove PulseAudio until it's fixed. There is no easy & ideal solution right now. Thanks for understanding.

closed 2016-10-21 by renekliment