alexa-pi/AlexaPi #50: Support for new features included in api v2

amahlaka, 2016-10-20:


amahlaka, 2016-10-20:
Also if there is any help needed, I would be more than happy to help

renekliment, 2016-10-20:
Not in the near future unfortunately. We have a lot of refactoring to do, before we can do such big changes. The current primary goal is to get #30 merged. That requires testing and code review. I believe we are close on that one, if Mason won't have further issues.

amahlaka, 2016-10-21:
Would the api update require massive overhaul on components currently being worked on?

If I were to start updating to the api and somehow succeed, would it be possible to add it as a new branch?

renekliment, 2016-10-21:
Okay, here's the thing. The whole API / processing / platform notifying code is kind of intertwined / it's not clear to me what goes after what and when and how we can separate it right now (the specific code changes).

What we will definitely do is to separate the AVS stuff into a module like _alexapi.somethingLikeServices.avs_. We need this, there is no doubt there.

The (shortest) path to new API I had in mind:
1. help #30 get merged - testing & CR, I believe we're almost there, but if you help with it, we can finally get it out of our hair
2. I'll do #21, which will affect every single python file and I'll need CR and testing in case I accidentally break something; this should be pretty quick, but I need someone to review my changes - this is what takes the most time as people don't have much time to do that
3. in #23 we'll split the code for current API so it's gonna be more clear what the module should look like
4. anyone can go crazy and implement the new API

Basically, the path could be shorter, but there's the thing that right now, you don't know how the main code is gonna interface with the _service_ module, therefore we need to do _3_ and _2_ is gonna help us set up better pylint tests and is gonna make _3_ easier as well as general development. I've been postponing it for a while and I think it's time I finally do it.

Also, there's no guarantee that the new API mechanisms can be the same as with the old one. We have to find out.

So if you want to help out, I (and anyone else) would greatly appreciate it. If you'd help according to the scenario above, that would be the best. Then you could work on the new API in a special module and we'd have no merge conflicts and issues while other people are working on other stuff.

renekliment, 2016-11-01:
So just to keep track, @davidmroth is doing some serious work on this at https://github.com/davidmroth/AlexaPi/tree/avs_v2

We'd have to do some things first to get that merged (#21, #23, change it to use #68), but it looks very promising. We might have version 2.0 sooner than we thought :-)

davidmroth, 2016-11-01:
Thanks @renekliment! Not event close to being ready. My plan is to squash the current branch once I have a stable API. Then I was thinking I would back port some of the awesome things you guys are doing. If all goes well, I wanted to see if you'd be interested in merging this into a AVSv2 branch or something.

renekliment, 2016-11-01:
Yeah, it needs some work, but you're not fooling around!

If we have #23 and #68 and you then rebase your branch on top of that, using those interfaces, I'd be glad to merge that into an official avs2 branch, maybe even throw it into master as a preview (not on by default). I haven't looked at #23 yet at all, but I hope we can have these services like AVS modular so we can support both API versions, hell, maybe even other services if available, so people can just chose in their config.

amahlaka, 2016-11-02:
@davidmroth  how far have you gotten on the v2 branch?

renekliment, 2016-12-09:
There seems to be a library for the new API - https://github.com/nicholas-gh/python-alexa-client as pointed by @illperipherals 

@davidmroth You might be interested in this.

renekliment, 2016-12-09:
@nicholas-gh You might be interested in our project. If not, sorry for the spam.

dgonano, 2016-12-11:
There is so many different python implementations of alexa, all with different features and parts working. 
Is there some documentation on what we're working towards in terms of code structure/modularity, threading and generic playback/device/voice service interfaces?

nicholas-gh, 2016-12-20:
I don't think so. It's just a case of who has time and an itch to scratch :)

I'm not actively working on my now - I have "real" echo devices. They have some benefits (mics, spotify), but also some downsides (no local scripting ability.)

renekliment, 2016-12-20:
Another link for the record: https://github.com/nicholasjconn/python-alexa-voice-service
@davidmroth You might wanna check it out.

methanoid, 2019-03-09:
Is this likely to happen soon? I mean this is great project cos its headless but it was reporting things dont work like timers due to API version but that was 2+ years ago. If I could code it, I would!!!

hunterman2244, 2019-05-19:
Any development on this recently?