alexa-pi/AlexaPi #54: Pi Zero "issues"

illperipherals, 2016-10-21:
Discussion for running on the Pi Zero.
Audio/USB Hub/Power/etc...

bchuckles, 2016-10-21:
awsome! so i hear no sound, but no errors.

Working on pi 3, but with analog.

Switched to pi zero and have no sound with the sound dongle. Have changed the default sound in top bar to USB card.

Its frustrating because i remember its something simple like the config file to change the output but i cant remember for the life of me!

renekliment, 2016-10-21:
Maybe you have audio via HDMI by default?

@illperipherals I'm not sure how much AlexaPi-related this is. If you guys have specific issue - _sound output_ for example, name the issue accordingly. Let's prevent these issues where there's discussion about multiple things.

bchuckles, 2016-10-21:
@renekliment "Have changed the default sound in top bar to USB card." so thats not it. And the script is not working for Zero "out of the box" so this thread will help anyone who is trying to set up a Zero. Maybe we can add to the README or update the code after we are finishing debugging and finding solutions.

illperipherals, 2016-10-21:
Okay, does that topic work better? This is actually a Zero related problem, as he stated, the same card works in a Pi3.

illperipherals, 2016-10-21:
@bchuckles, by top bar, are you running a GUI? Although it can be done, I would recommend against it if possible on the Zero. Can you post the output of those commands that I listed in the other thread? Thanks!

CEOGavin, 2016-10-22:
@illperipherals @bchuckles In order to establish that this is an AlexaPi issue (for PiZero or any other Pi), let us first check if the system sound works. Try the following command -
`aplay /usr/share/sounds/alsa/Front_Center.wav`

1) Did you hear a sound?
2) Paste the result of `aplay -l`
3) Also paste the result of `aplay -L`

@renekliment Firstly, excellent work on alexa-pi. I just got it working on my Pi3 today! I am using an external sound card (did not have a USB microphone so I used a 3.5mm one instead. Hence the need to use an external sound card). I might port AlexaPi to Node at some stage (if possible).

bchuckles, 2016-10-22:
@illperipherals thanks for the help. Yea I'm using a GUI for the time being. I'll switch to headless once it's in a state of working order. I'll run those command tomorrow I'm outta town. 

@CEOgavin 3.5mm has always worked with this project and yes that is working win a pi 3. When u switch to a dongle some stuff needs changed so we are trying to figure out what needs changed to help people. If you write this in node let me know cause I'll help you! That's why I struggle cause I'm a node developer and this is python :/

CEOGavin, 2016-10-22:
@bchuckles I would love to write it in Node (this is my first time using Python). Any suggestion how I should start? Anyone else done something similar?

renekliment, 2016-10-22:
No, don't do that, help us here instead :-D

But seriously, if you're bothered by the audio issues, you're gonna have to solve a similar thing on Node.js too. I think we'll make it work more-less out-of-the-box when we switch to pyaudio and do some kind of detection so users can just choose the I/O devices. No one just bothered to do that with the current pyalsaaudio library when we're going to abandon that.

A part of Amazon's official sample app is written in Node.js so I think that's a place to start: https://github.com/alexa/alexa-avs-sample-app

CEOGavin, 2016-10-24:
Thanks so much @renekliment . Ideally I would have loved to help but I don't know Python at all. Plus the lack of drive to get into Python while I am spending all my free time with Redux and React Native.

The list of things to learn is always disproportional to the time available :)

Frankly, the only reason why I am considering porting it to Node is simply to brush-up my Node skills and learn more in the process.

Could I persuade you to switch to Node instead? ;)

renekliment, 2016-10-24:
Nope, I'm sticking with python :) I don't know Node.js at all, even though I do some JS and I think there's a long way from one to the other. Another reason is that I think python is more multi-platform and I personally want to run AlexaPi on OpenWRT routers eventually and Node.js on mips, mipsel or even PowerPCSPE is not easy / real.

illperipherals, 2016-10-24:
@bchuckles, I know you are out of town, but I just wanted to drop a couple things in here.

Depending on what your aplay shows you:

Change device ordering:
	sudo nano /usr/share/alsa/alsa.conf
	Scroll down and change the following two lines:
		From:
			defaults.ctl.card 0
			defaults.pcm.card 0
		To:
			defaults.ctl.card <mycardnumber>
			defaults.pcm.card <mycardnumber>

You can also edit /boot/config.txt and comment the following line:
		# Enable audio (loads snd_bcm2835)
		dtparam=audio=on
	This will stop the loading of the on-board audio. (if you are using USB for all your sound)

If that does not help, and you cannot default to the USB card, you can override this by stating -D sysdefault:CARD=<mycardnumber>

Also, make sure your username belongs to the "audio" group in /etc/group
		sudo nano /etc/group

Let me know when you get back in. I am setting a new Zero up from scratch to walk through it with the new builds.

renekliment, 2016-10-24:
When you guys find solution to the Zero problem, please add it to the documentation (the wiki here).

illperipherals, 2016-10-24:
10-4

amahlaka, 2016-10-25:
the way i use usb audio is 
```
 i = vlc.Instance('--aout=alsa  --alsa-audio-device=sysdefault:CARD=Device')
```
in main.py at def play_audio

renekliment, 2016-10-25:
@amahlaka 
We'll do #22 so you don't have to do that nasty workaround. Soon hopefully :)

bchuckles, 2016-10-25:
@illperipherals thanks I had a demo i needed to give, so i had to switch back to the 3. Hopefully my schedule will free up next week, and will have a free weekend to hack on the zero. These tips help though ill see if i can get an hour to test these out.

benbois, 2016-10-26:
Hi, I've installed Alexa-pi on a Pi Zero with some success.
The sound card is USB one, and here you have what I did to get it work:
in **/etc/modprobe.d/fbdev-blacklist.conf**
`blacklist snd-bcm2835`

in **/etc/asound.conf** 
```
pcm.!default {
type hw
card 0
}

ctl.!default {
type hw
card 0
}
```

in **/etc/modprobe.d/alsa-base.conf**
`options snd-usb-audio index=0`

and finally, in **/etc/opt/AlexaPi/config.yaml**
`  device: "plughw:0"`

The only issue is with trigger, it doesn't work.
It  did once and that's it. I don't know why actually.

renekliment, 2016-10-29:
@benbois Could you please test if with #66, you can leave all this audio settings and just set everything in `config.yaml`?

benbois, 2016-10-29:
@renekliment I changed the line : `device: "plughw:0"` in the configuration file as described in my previous post.

renekliment, 2016-10-29:
Yes, I know. My point is that with #66 you shouldn't need all the

> in **/etc/modprobe.d/fbdev-blacklist.conf**
> `blacklist snd-bcm2835`
> 
> in **/etc/asound.conf** 
> ```
> pcm.!default {
> type hw
> card 0
> }
> 
> ctl.!default {
> type hw
> card 0
> }
> ```
> 
> in **/etc/modprobe.d/alsa-base.conf**
> `options snd-usb-audio index=0`

beucause you can also set output audio device in `config.yaml`.

benbois, 2016-10-30:
Well, I tried to remove the 2 settings (blacklist.conf + asound.conf) and nothing worked!
Then, I supposed that the Zero needs this.

renekliment, 2016-10-30:
Are you trying out the code from #66? What config options did you try?

benbois, 2016-10-30:
I just tried the patch #66, it doesn't work for me.
I don't even get any sound now, I have to come back to the previous conf./files.

renekliment, 2016-10-30:
Ah, too bad :disappointed: I hoped setting the output device only in the configuration file would work.

renekliment, 2016-12-07:
So are there still issues with the Zero? Is there anything we can do to help?
If there are no issues anymore, please put up the relevant info in the wiki (if any) and close this. Thank you.

renekliment, 2016-12-11:
@illperipherals @benbois Forgot to tag you in the previous comment.

illperipherals, 2016-12-11:
I will load a new Zero test up tonight.

garrettheking, 2016-12-22:
Hopefully a fix comes soon. All I have is a Pi Zero at the moment :/ . It seems like everything is working for me except audio output.

renekliment, 2017-01-06:
Any progress on this?

garrettheking, 2017-01-06:
Yes I did manage to fix it. I don't remember exactly what I did and I will post when I figure out. What I did was blacklist hdmi and analog output then changed a value on a PulseAudio file, rebooted And it works.

renekliment, 2017-01-15:
@garrettheking Pinging you on this so we can hopefully close this. Thank you.

illperipherals, 2017-01-15:
The Pi Zero requires external network and audio (aside from the audio hack outlined here: https://learn.adafruit.com/introducing-the-raspberry-pi-zero/audio-outputs )

This brings much higher power requirements to the table.

closed 2017-01-15 by illperipherals

garrettheking, 2017-01-24:
I managed to make audio work. However, “Hey Alexa” Will not work because the CPU goes to 100% and I do not know how to prevent that. You will have to use a button connected to GPIO. On a pi zero, I’m using a cheap eBay sound card and not HDMI audio, so I blacklisted HDMI and Analog output. 
in:
 `/etc/modprobe.d/fbdev-blacklist.conf`
Go to the bottom of the document and add:
`blacklist snd-bcm2835`
(Again, this *WILL MAKE HDMI AUDIO AND ANALOG NOT SHOW UP*)
After blacklisting, you have to change where the audio goes
Go to:
`/usr/share/alsa/alsa.conf`
Find:
`defaults.ctl.card 0`
and 
`defaults.pcm.card 0` (*Should be right below*)
CHANGE THE 0 TO 1 In both of those.
Reboot and you should hear “hello” when boot is complete
I tried my best to follow my steps. However, this should be all though. And once again, I do not know how to prevent CPU going over 100%. It causes the keyword to stop working. You have to use a button.