alexa-pi/AlexaPi #57: Write CHANGELOG

renekliment, 2016-10-22:
This is actually a continuous thing, but we need to start.
I think doing it according to this: http://keepachangelog.com/en/0.3.0/ is a good idea.

The section _Project history_ from `README.md` should be moved there (rewrite it according to that format) because it belongs there and we don't want to have a mile-long README. Also, we should add the stuff that's been changed so far.

The main reason for this is that people are actually starting to use this and it's a good idea to write at least breaking changes somewhere.

renekliment, 2016-10-22:
I'd like to use http://semver.org/ but we're already at MAJOR = 1 and we're gonna do a lot of breaking changes in the near future so I don't want to increase it every time we release a new version. Going "back" to 0.Y.Z doesn't seem like a good idea either, since it would probably confuse people.

@alexa-pi/coreteam What do you guys think?

illperipherals, 2016-10-22:
I feel that if #30 is not causing these new issues to come in, then that makes for a solid major release.

renekliment, 2016-10-22:
@illperipherals I'm sorry, my English fails here, what do you mean?

illperipherals, 2016-10-22:
If the new issues that have come in are not due to issue #30, then I think that is a good candidate for a major number initial release.

renekliment, 2016-10-22:
I see. It's clear, silly me.
 
What I meant is that I'd like to release new versions more often, but the things that we're gonna be doing for quite some time now will be all breaking in some way (configuration mainly) so we'd soon be at major version like 5. 

On the other hand, do we want to bother with _versions_? I guess it's a good idea, but I think everyone might just use `master` and not care, since we're making sure the code there works.

An alternative might be to do versions 1.X until we do all the crazy things we wanna do and when things get calm, we switch to the semantic versioning starting with version 2.0.0.

I'm not sure what's the best approach. Discussion, discussion :speech_balloon:

illperipherals, 2016-10-22:
I would definitely like to use versions. Because we will be working in branches, lets just increment when pushed to master. Does that cause any kind of problem?

illperipherals, 2016-10-22:
*off-topic*
if we are not going to use irc or slack or something, I feel we need a place for quick offtopic notes to go.

renekliment, 2016-10-22:
Also, if we release new version soon (whatever version that may be) I'd like to 
- include #55, because it's basically done, we just need to write some documentation, because it requires some manual steps for now and I don't think it's that discouraging, it's two things really,
- include #42, because it fixes a bug (#51) that appeared when playing all the files that AVS sends (#34).

maso27, 2016-10-22:
Things have EXPLODED around here in the last couple days! (Or maybe it's just because I finally remembered to "watch" this version of the project and now I'm aware of it all.)

My level of ability/experience has been surpassed, but I'll struggle to keep up as best I can.

My opinion on this:
Version tracking is valuable.  Keeping track of who's installed what when they're having troubles has been pretty difficult.  Makes sense to me to keep thing at 1.X until things are in a decent state, then roll the major number when new things get implemented.

renekliment, 2016-10-22:
@illperipherals 

> Because we will be working in branches, lets just increment when pushed to master. Does that cause any kind of problem?

I'm not sure I get what you mean. We would do versioning only on `master`, because I don't think it does make sense on feature-branches or any other branches for that matter (also, we'd have collisions and stuff). 

Yes, I'm feeling the need for some announcements / talk too. Therefore:
1. I'm starting this change log, where users can see what's new with the project and to be aware of important changes.
2. I've started on the wiki here. It would be great if we could build a decent documentation there - mainly regarding audio. Feel free to add stuff in there. I'm gonna send a small PR in a bit, that removes stuff from README that's moved there and puts links to the wiki so people are more aware of it.
3. I've requested Gitter for this repo, but @sammachin needs to approve this first as the org owner. I'll try to get in touch with him to speed it up.

renekliment, 2016-10-22:
@maso27 That issue-solving makes sense. Good point.
For that to work really, we'll have to probably switch to have dev / master branches so when people pull from master they're on a single version, not somewhere in between versions.

renekliment, 2016-10-26:
So Sam gave me some sweet permissions and now we can chat over at https://gitter.im/alexa-pi/Lobby! I'll be sure to update the docs so people can find it.

PR #59 (2016-10-26): Documentation update & Change Log (fixes #57)
This:
- updates tiny bits of README,
- starts a Change Log,
- deletes stuff that has been moved to the Wiki or the Change Log.

Fixes #57.
commit a02e34bb (2016-10-26): docs update / move to wiki; added Change Log (fixes #57) (#59)
closed 2016-10-26 by renekliment