alexa-pi/AlexaPi #58: Crash: alsaaudio..Device or resource busy [plughw:1]

irthomasthomas, 2016-10-22:
Just tried the latest version on a pc running ubuntu 14.04. It says hello before exiting with the following message:
line 498..    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, config['sound']['device'])
alsaaudio.ALSAAudioError: Device or resource busy [plughw:1]

irthomasthomas, 2016-10-22:
Thanks, I am running this in a terminal already. Am I supposed to configure the audio stream somewhere?

CEOGavin, 2016-10-22:
`sudo nano /etc/opt/AlexaPi/config.yaml`

ridai260, 2016-10-22:
Is it possible to simply use the inbuilt RPi3 audio jack for this? I've uninstalled pulse as it's causing nothing but issues, and when I run arecord -L, I only get 'null' or 'pulse'.

irthomasthomas, 2016-10-22:
ok I set it to "default" and it works nicely now. Cheers.

closed 2016-10-22 by renekliment