alexa-pi/AlexaPi #60: Refactor: Setup script

renekliment, 2016-10-22:
- Write it to be very modular.
- Do it is as platform-independent as possible.
- It will have to implement some checks - or should we ask the user about everything?
- In python? Maybe in a later phase if it makes sense?
- Needs to setup OS platforms (covers requirements install for major systems)
    - [ ] try to do as much we can via `pip`
    - [ ] apt-get based systems
    - [ ] pacman/yaourt based systems
    - [ ] yum (or whatever Fedora uses) based systems
    - Mason wrote a pretty comprehensive manual for Arch on RPi: [Arch setup](https://github.com/alexa-pi/AlexaPi/pull/28#issuecomment-251578831) and [making it work under the alexapi user](https://github.com/alexa-pi/AlexaPi/pull/28#issuecomment-253115465)
- Needs to also setup device platforms.
    - Don't forget to do the install of GPIO libraries for RPi and OrangePi

renekliment, 2016-10-22:
I have to say, this is gonna be one of the less fun refactors. But what the hell, we can probably just put everything into functions, maybe separate files if it makes sense and ask the users what they want (which OS and device platforms) and call the corresponding functions and we're done super quick.

closed 2016-11-27 by renekliment

illperipherals, 2016-11-27:
I vote for keeping this as shell script for the time being. I think it might be beneficial to start working our scripts towards a common skeleton (cli params, colors, logging) for consistency.

PR #82 (2016-11-27): Setup script refactor
Fixes #60, #79.

What's still missing:
-  ~~Fedora support - does anyone use that? Do we care?~~ 
Can be easily added later when there are people to use that on daily basis.
- [x] Arch Linux: _"python" => "python2" everywhere (including all other scripts)_ - setup should be fine with the alias now - we'll add choice for py version later; do a systemd unit override for this - something like `Exec=/usr/bin/python2 ...`
- [x] include frequently used systemd unit overrides (usergroup-root and stuff)
- [x] OS / device detection / selection - let's just ask the user and don't o [... cut]
commit 6f64ae9d (2016-11-27): Setup script refactor (#82), fixes #60, fixes #79

* modularized and generalized, cleanup

* config: renamed ProductID to Device_Type_ID

* support for Arch Linux (instructions by @maso27)

* support for currently supported boards in AlexaPI (GPIO libs install, etc)

* added unit overrides for various things

* set up GPIO permissions (udev)

* use pip instead of system packaging where possible

* disabled pip cache due to pypa/pip#2984

* change default device_platform to dummy

* set device-specific config on install