alexa-pi/AlexaPi #63: Intermittent sound problem.

CEOGavin, 2016-10-24:
Every once a while the audio turns into some sort of white-noise. It makes a terrible hissing sound (sometimes with the _alexayes.mp3_ or with the _response mp3 files_). When I manually try and play these files, I still hear the hissing sound via the `aplay` command. However, `mplayer` always plays them properly.

Here is the output -
```
pi@raspberrypi:~/Downloads $ aplay KnowledgePrompt.193ffa5c-72ab-4525-a563-21e0ff96c895_725789800.mp3 
Playing raw data 'KnowledgePrompt.193ffa5c-72ab-4525-a563-21e0ff96c895_725789800.mp3' : Unsigned 8 bit, Rate 8000 Hz, Mono
pi@raspberrypi:~/Downloads $ mplayer KnowledgePrompt.193ffa5c-72ab-4525-a563-21e0ff96c895_725789800.mp3 
MPlayer2 2.0-728-g2c378c7-4+b1 (C) 2000-2012 MPlayer Team
Cannot open file '/home/pi/.mplayer/input.conf': No such file or directory
Failed to open /home/pi/.mplayer/input.conf.
Cannot open file '/etc/mplayer/input.conf': No such file or directory
Failed to open /etc/mplayer/input.conf.

Playing KnowledgePrompt.193ffa5c-72ab-4525-a563-21e0ff96c895_725789800.mp3.
Detected file format: MP2/3 (MPEG audio layer 2/3) (libavformat)
[mp3 @ 0x75bdb5b8]Estimating duration from bitrate, this may be inaccurate
[lavf] stream 0: audio (mp3), -aid 0
Clip info:
 encoder: Lavf57.41.100
Load subtitles in .
Selected audio codec: MPEG 1.0/2.0/2.5 layers I, II, III [mpg123]
AUDIO: 24000 Hz, 2 ch, s16le, 48.0 kbit/6.25% (ratio: 6000->96000)
AO: [pulse] Init failed: Connection refused
AO: [alsa] 48000Hz 2ch s16le (2 bytes per sample)
[AO_ALSA] Unable to find simple control 'Master',0.
Video: no video
Starting playback...
A:   1.3 (01.2) of 1.4 (01.3)  2.5% 

Exiting... (End of file)
```

Here is the mp3 file in question - https://www.dropbox.com/s/efjyesn9m7z2qh4/KnowledgePrompt.193ffa5c-72ab-4525-a563-21e0ff96c895_725789800.mp3?dl=0

Here is the white noise that I hear when aplay or main.py script executes it -
https://www.dropbox.com/s/ntwv6eud9m5zh27/White%20noise.mp3?dl=0

I am using Raspberry Pi 3 with the following program versions -
```
aplay: version 1.0.28 by Jaroslav Kysela <perex@perex.cz>
MPlayer2 2.0-728-g2c378c7-4+b1 (C) 2000-2012 MPlayer Team
VLC media player 2.2.1 Terry Pratchett (Weatherwax) (revision 2.2.1-0-ga425c42)
VLC version 2.2.1 Terry Pratchett (Weatherwax) (2.2.1-0-ga425c42)
Compiled by buildd on testwandboard (Jan 24 2016 00:03:37)
Compiler: gcc version 4.9.2 (Raspbian 4.9.2-10)
Linux raspberrypi 4.4.26-v7+ #915 SMP Thu Oct 20 17:08:44 BST 2016 armv7l GNU/Linux
python 2.7.9
pip 8.1.2 from /usr/local/lib/python2.7/dist-packages (python 2.7)
alsactl version 1.0.28
(I don't have PulseAudio installed)
```

Any idea what's causing this? 

Thanks in advance @renekliment

renekliment, 2016-10-24:
Damn, more audio issues! Alright, I'm moving #45 to my priority list. I'll do that as the next thing so we can start debugging these issues without worrying about moving to another library later only to find out we wasted time on debugging with the previous one.

If we're not able to fix this with that, another option would be #25 so we can for example use the mplayer to play things, but it will take time to get there as there are more important things to work on now. It will be free to take up though, after the current PR on audio queue is done.

Thanks for the thorough issue report.

illperipherals, 2016-10-24:
What are your thoughts on SoX? http://sox.sourceforge.net/

renekliment, 2016-10-24:
For playing audio? 
I don't have much experience with it - I've used it like twice in my life.

Once we have #25, it's gonna be super easy to play things via various programs. It's gonna take literally 5 minutes to write a support for new audio (maybe we should call it playback) handler. So once we're there I don't see no reason not to have support for sox :)

CEOGavin, 2016-10-26:
Unfortunately the app is unusable for me in it's current state. I just worked out that 37% of all audio ends up in loud hissing white-noise (which is a fair bit). 

Fingers crossed. Either #45 or #25 should fix it.

renekliment, 2016-10-26:
I don't see why it would help, since it should go to alsa anyway, but can you try the thing suggested in sammachin/AlexaPi#89 ?

`vlc.instance('--aout=alsa')` to `vlc.instance()`

CEOGavin, 2016-10-27:
Implemented those changes. The white noise has now dropped to about 20% which is definitely an improvement but it hasn't completely gone.

renekliment, 2016-10-27:
Do you have PulseAudio installed? If so, you can try `--aout=pulse`.

CEOGavin, 2016-10-27:
Let me try. I'll update shortly.

CEOGavin, 2016-10-27:
I followed the instructions here - https://wiki.debian.org/PulseAudio and now I seem to have made it worse somehow. I am going to uninstall pulse again and switch back to the old settings (25% white noise is better than no sound). At this point an easier option would be to get rid of vlc and replace it with mplayer instead :(

illperipherals, 2016-10-27:
A couple features I really like about SoX are:

sox short.wav long.wav longer.wav
(concatenates two audio files, and)

sox −m music.mp3 voice.wav mixed.flac
(mixes together two audio files.)

I can think of some interesting ways to use that second functionality.

renekliment, 2016-11-05:
If you have some time, you can try to write a an mplayer playback_handler based on #68. 
From a quick search, you could possibly use:
- https://github.com/baudm/mplayer.py
- http://pythonhosted.org/PyMPlayer/
- If you decided to go the hardcode way and wanted to run mplayer via command-line, it would be a good idea to write a general _clihandler_ and either set the command in the config or to have it as an abstract class and inherit it into an mplayer handler, setting only the command or something.

renekliment, 2016-11-06:
So I gave sox a very quick look and it really looks like it kicks ass even for playback.
It's also probably the only usable thing for us on OpenWRT and it has python bindings as well: https://github.com/rabitt/pysox

illperipherals, 2016-11-06:
Alright. Let's explore in that direction. I got a new Zero build up today using these build scripts ( https://github.com/TheSin-/rpi-img-builder ), and I am going to do a bit of testing on that this morning. I like the way plugins are handled, and I think it would make it easy to work images into more customizable daily CIs if we wanted.

renekliment, 2016-12-11:
@CEOGavin Wait. I've just now noticed that you were trying to play mp3s with `aplay`. That's a no-go - `aplay` does only wave files. Try playing with `cvlc` since we use VLC library to play stuff.

Please try current `master` (watch out for config changes though). There might be a one thing that might help (but no promises).

Also, the playback handlers are in `master` now, so it is possible to write for example the SoX handler we discussed earlier.

renekliment, 2017-01-16:
@CEOGavin Can you verify that using the SoX playback handler (latest `dev` branch) fixes your issue? Can you also try the VLC handler with our latest code on a latest OS?

Since this is open for more than a month without response and there should be a fix, I'm closing this to clean up. Reopen if needed. Cheers.

closed 2017-01-16 by renekliment