alexa-pi/AlexaPi #70: Repeatedly saying "Hello"

KrunchMuffin, 2016-10-31:
Not sure what happened, but every 7secs or say it says "Hello". Not sure what you need to see, so let me know and I will post. RPi3

renekliment, 2016-10-31:
The app crashes and you've enabled automatic restart on crash. That's why you get _Hello_ multiple times. See `journalctl -u AlexaPi.service` for the log. When debugging, it's always useful to run it from command line in debug mode like `/opt/AlexaPi/src/main.py -d`

KrunchMuffin, 2016-10-31:
Thanks. Never occurred to me that it was crashing. wrong device I had typed in.

On another note, I am trying to use the mic that is in my USB cam but it is not working. Is there a way to test it similar to testing audio? I tried alsamixer, but no luck.

renekliment, 2016-10-31:
Try `arecord` with a proper device parameter. (And `aplay` to play.) Once you get it right, set that device in AlexaPi config.

KrunchMuffin, 2016-10-31:
cool, thanks. All working now. Is there a config somewhere to tell it what city you are in and timezone? I have my timezone in pi set up correctly, but she still gives me PST time. And weather is Seattle when I just ask for weather.

SurfSound, 2016-10-31:
Hi KrunchMuffin
can you tell me what you've entered in the config to get the mic working?
Cause I have the same setting but I don't know what name I have to type in.

KrunchMuffin, 2016-10-31:
I just followed this http://www.g7smy.co.uk/2013/08/recording-sound-on-the-raspberry-pi/