alexa-pi/AlexaPi #71: Internet Connection Failed

SurfSound, 2016-10-31:
Hello
in the log I get the message 
"Checking Internet Connection....
 Connection Failed"
but this only appears when I think I have the right device in the config file selected. When I select a wrong device I have no problem with the Connection.

renekliment, 2016-10-31:
Hey! So this doesn't make any sense. Can you run AlexaPi from a command line in debug mode (see our wiki on how to do that) 5 times with each setting and paste the output?

SurfSound, 2016-11-01:
ok so this is when I use the right device:

Checking Internet Connection...
Connection OK
Play_Audio Request for: /opt/AlexaPi/src/resources/hello.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended

but with "sudo journalctl -u AlexaPi.service" I still get Connection Failed.

when I type something other for the device I get:

Checking Internet Connection...
Connection OK
Play_Audio Request for: /opt/AlexaPi/src/resources/hello.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
ALSA lib pcm_hw.c:1667:(_snd_pcm_hw_open) Invalid value for card
Traceback (most recent call last):
  File "/opt/AlexaPi/src/main.py", line 596, in <module>
    loop()
  File "/opt/AlexaPi/src/main.py", line 511, in loop
    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, config['sound']['input_device'])
alsaaudio.ALSAAudioError: No such device [plughw:yd]

And when the right device is selcted Alexa doesn't crash but then it doesn't respond when I say anything

renekliment, 2016-11-01:
This is right.
When you run `sudo journalctl -u AlexaPi.service`, it display the log of AlexaPi when it's run as a service (see the times of the messages).
So it looks like it works.

I believe you just have to set microphone sensitivity properly or say the word _Alexa_ properly. The recognition is not very accurate - sometimes, you just have to know _how_ to say that. But we'll get around that soon.

SurfSound, 2016-11-01:
ok thank you
then I will play a bit till I get it ;)

closed 2016-11-01 by SurfSound