alexa-pi/AlexaPi #72: Alexa does not respond to voice

Mistercl0ud, 2016-11-02:
Hello!

I'm having a problem after the installation. Alexa doesn't to any voice command. I tried to use the /opt/AlexaPi/src/main.py -d command and that's what the terminal is showing me

`pi@raspberrypi:~ $ /opt/AlexaPi/src/main.py -d
Checking Internet Connection...
Connection OK
Play_Audio Request for: /opt/AlexaPi/src/resources/hello.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
Traceback (most recent call last):
  File "/opt/AlexaPi/src/main.py", line 595, in <module>
    loop()
  File "/opt/AlexaPi/src/main.py", line 510, in loop
    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, config['sound']['input_device'])
alsaaudio.ALSAAudioError: Device or resource busy [plughw:1]
`

I've tried changing the input to default but it still doesn't work. Could you help me fix this issue ?

Thanks a lot !

renekliment, 2016-11-02:
See https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging - paste output of `arecord -l` and `arecord -L`.

The message in the issue title means that either another app is using the audio input or that you have wrong device in the config. Make sure no other app is recording and that AlexaPi isn't running as service when you run it from command line interface. Try `systemctl status AlexaPi.service` to see.

Mistercl0ud, 2016-11-02:
Hello ! Thank your answer ! I only have AlexaPi on this raspberrypi and I'm using Raspbian without the desktop part so I don't think anything else is using the microphone. Here's what arecord -l, arecord -L and the last command are giving me 

> pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

> pi@raspberrypi:~ $ arecord -L
null
    Discard all samples (playback) or generate zero samples (capture)
sysdefault:CARD=Device
    USB Audio Device, USB Audio
    Default Audio Device
front:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Front speakers
surround21:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct sample mixing device
dsnoop:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct sample snooping device
hw:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Direct hardware device without any conversions
plughw:CARD=Device,DEV=0
    USB Audio Device, USB Audio
    Hardware device with all software conversions

> pi@raspberrypi:~ $ systemctl status AlexaPi.service
● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
  Drop-In: /etc/systemd/system/AlexaPi.service.d
           └─restart.conf
   Active: active (running) since Wed 2016-11-02 00:54:48 UTC; 12h ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
 Main PID: 642 (python)
   CGroup: /system.slice/AlexaPi.service
           └─642 /usr/bin/python /opt/AlexaPi/src/main.py

Warning: Unit file changed on disk, 'systemctl daemon-reload' recommended.

renekliment, 2016-11-02:
So you can see that AlexaPi service is running, which if it works properly, occupies the recording device. You have to stop it first if you wanna fiddle with it in command line. `systemctl stop AlexaPi.service`

Mistercl0ud, 2016-11-02:
After terminating Alexa service and using the `/opt/AlexaPi/src/main.py -d` command, Alexa does respond to command. However, if I restart the service and try again to talk to her, nothing happens.

> pi@raspberrypi:~ $ /opt/AlexaPi/src/main.py -d
Checking Internet Connection...
Connection OK
Play_Audio Request for: /opt/AlexaPi/src/resources/hello.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
Play_Audio Request for: /opt/AlexaPi/src/resources/alexayes.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
detected the edge, setting up audio
Starting to listen...
Debug: End recording
Debug: Sending audio to be processed
Sending Speech Request...
Processing Request Response...
JSON String Returned: {"messageBody": {"directives": [{"namespace": "Speaker", "name": "SetMute", "payload": {"mute": false}}, {"namespace": "SpeechSynthesizer", "name": "speak", "payload": {"contentIdentifier": "amzn1.as-ct.v1.Domain:Application:Knowledge#ACRI#KnowledgePrompt.88203464-c956-4121-9a65-6323c94b07a7", "audioContent": "cid:KnowledgePrompt.88203464-c956-4121-9a65-6323c94b07a7_75401775"}}]}, "messageHeader": {}}
Play_Audio Request for: file:///tmp/AlexaPi-runtime-hQ7uGh/KnowledgePrompt.88203464-c956-4121-9a65-6323c94b07a7_75401775.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended

Mistercl0ud, 2016-11-02:
After reinstalling the pi from a fresh new installation, Alexa is finally working. Thank you for your help !

renekliment, 2016-11-02:
Glad to hear that! Were you running the same OS version before?

closed 2016-11-02 by renekliment

Mistercl0ud, 2016-11-02:
I believe it was the exact same one. Many times Alexa will respond to a few commands with no problem and then suddenly, I can say the keyword for a solid minute and she will not respond.