alexa-pi/AlexaPi #74: Alexa triggers itself (herself) when playing audio

renekliment, 2016-11-05:
So this thing ...

My thoughts / observations:
1. Sometimes it happens for an obvious reason like when she's talking about herself.
2. Sometimes it happens even when she doesn't say her name.
3. Snowboy might be more accurate and could solve (2).

Any ideas?

maso27, 2016-11-07:
This is really tough to solve. 
My initial thought is that you would need a predictable hardware envelope and some DSP to cancel out the speaker.  This might be how the Echo works, since they know where their speaker is in relation to their microphone.

But I start to doubt that when I realize the Dot is designed to use a separate speaker.  I don't have an actual echo of any sort, so I'm not sure whether the Dot hears its own trigger word or not.

(At one point I thought it would be funny to say "Alexa, Simon says Alexa Simon says Alexa" to see if she could double-trigger herself, but the trigger response wasn't fast enough yet.)

(Also, I've had a heckuva time this last few months listening to anything news-y because the word "election" would trigger my box and interrupt whatever I was listening to.)

renekliment, 2016-11-07:
_election_ :smile: That's hilarious :smiley: 

But back to the topic - I just got a very silly idea. It would be non-ideal, would eat resources, but is something that is theoretically doable. We could set up a spare hot-word detector and tap into the output of AlexaPi. If it detected the hot-word, it would disable the voice trigger for like a second and then enable it back. We would have to solve some delays - either playback or the trigger so it would catch the false positive in time.

But yeah, however (and if) we decided to do this, it is a thing of a somewhat distant future as we have _many many_ things to take care of first.

maso27, 2016-11-07:
It's crazy, but I like it!

It would add some processor load--and effectiveness would probably depend on the platform (Zero Vs. Pi 3, etc.).  But it's probably a more feasible hack than constantly trying to noise-cancel.

illperipherals, 2016-11-07:
Snowboy supports multiple hotword detection.

dgonano, 2016-12-15:
I thought triggering (listening for hotword) is disabled when she is speaking the standard response. It only works on streams like music or flash briefing. 

I also changed my trigger word to "jarvis" (iron man) which seems to detect better and doesn't come up as much.

pavanagrawal123, 2016-12-28:
I think noise cancelling would be ideal! (though hard to implement). Currently, when I am listening to music, the Alexa-Pi doesn't recognize the wake up at all. I tried it with the Amazon's Sample app, and it is the same thing. While playing music, it practically never hears the wake word.

FarvaTechnology, 2018-11-25:
Alexa Triggering Herself is driving me crazy! :(    I created an app that plays various radio traffic from various agencies - Every time I encounter a situation where alexa triggers herself and sometimes stops playing the audio stream.

Does anyone have a solution for this?

maso27, 2018-11-25:
Headphones.  :)

Realistically it would take some consideration to develop something that would disable that.  Perhaps two instances of the voice recognition running simultaneously, with one on the audio stream and one on the microphone stream, and not engaging when the audio stream is triggered.  Not likely to be added unless someone has time and motivation to do it though.

Beyond that, you could change the trigger word and try to find something more obscure.
I know that iheartradio channels are playing commercials for themselves where they say "Alexa, play KBCO on iheartradio" and stuff like that.  Drives me crazy!