alexa-pi/AlexaPi #75: Do logging properly

renekliment, 2016-11-05:
Probably use https://docs.python.org/2/library/logging.html
As @lechat mentioned in #67.

I'd personally like it if we could also have colors with this new thing.

lechat, 2016-11-05:
The colors are possible. There is a couple of libs that will do that, but it can be done w/o them as well.

renekliment, 2016-11-05:
Dope!

renekliment, 2016-12-07:
Also, this should get rid of the stupid `tunein.log` file.

renekliment, 2016-12-10:
Also, sometimes I've written `config['debug']` which wasn't just a trick to pass the value of debug into modules. There should be a config option (defaults to false, or like a zero logging level) for this and the command line parameter would override it. This could be useful for setting debug level even for when AlexaPi runs as daemon.

lechat, 2016-12-11:
I find it harder to support if default logging is set to absolute silence. Default should be minimal logging stating that all started up and all is fine.

renekliment, 2016-12-11:
Ah, you are right as usual :smiley_cat: 

Another thought: when running under systemd, don't output times as journal should do that.

commit ea9cbe43 (2017-01-09): Use python logging instead of prints (#110); fixes #75

* Replaced all prints with python logging

* Added --daemon option to remove timestamp from logs when running as
daemon

* Read default logging level from config

* Changed how config file is read

* Removed all print_function imports

* Removed alexapi.bcolors

* Removed creating tunein.log

* Updated change log
closed 2017-01-09 by renekliment