alexa-pi/AlexaPi #76: Alexa not listening

mohaggag, 2016-11-05:
I am using a usb sound card an the card has a LED when recording the LED blinks so i used it with the java version the original obe by amaxon and its working but wih alexa pi not i tried every thing with the yaml file but with no luck  i used plughw:1 and 1,0  but nothing happened

mohaggag, 2016-11-07:
i removed this project and using the original one from amazon this one are not working and is eating the ram and cpu its very bad the only good thing on it that it can start the alexa on the startup but unfortunately its not working

renekliment, 2016-11-07:
@mohaggag 
It's unfortunate that AlexaPi doesn't work for you. I'd argue that it has many advantages over the sample java app for a lot of people, but if that suits your needs better, use it of course.

If you'd like to make AlexaPi run, you need to file a proper bug report - what exactly doesn't work, what exactly did you do, post **logs** and outputs, ...

Closing for now. Feel free to reopen.

closed 2016-11-07 by renekliment