alexa-pi/AlexaPi #78: stop lisining on boot

mohaggag, 2016-11-07:
thats what i got in debugging mode 

`pi@raspberrypi:~ $ /opt/AlexaPi/src/main.py -d
Checking Internet Connection...
Connection OK
Play_Audio Request for: /opt/AlexaPi/src/resources/hello.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
Traceback (most recent call last):
  File "/opt/AlexaPi/src/main.py", line 595, in <module>
    loop()
  File "/opt/AlexaPi/src/main.py", line 510, in loop
    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, config['sound']['input_device'])
alsaaudio.ALSAAudioError: Device or resource busy [plughw:1]
`

renekliment, 2016-11-07:
Follow https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging and paste output of those four commands. Tell us what device and what sound card you are using. Are you running any other apps that record audio? Are you sure that AlexaPi service isn't running while you try to run it interactively? `systemctl status AlexaPi.service`

mohaggag, 2016-11-07:
arecord -L give me this 
plughw:CARD=Device,DEV=0

i used plughw:1,0
same issue

renekliment, 2016-11-07:
Is that the actual full output? What are your devices? What is the output of the status check?

renekliment, 2016-11-07:
1. Try using the whole string "plughw:CARD=Device,DEV=0" (without the quotes) as the actual input_device.
2. Just to be sure, please post output of `systemctl status AlexaPi.service` and `journalctl -u AlexaPi.service`.
3. Can you record with `arecord -D trysomedevicehere`?

mohaggag, 2016-11-07:
systemctl status AlexaPi.service
● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
  Drop-In: /etc/systemd/system/AlexaPi.service.d
           └─restart.conf
   Active: active (running) since Mon 2016-11-07 19:17:04 UTC; 1h 57min ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
 Main PID: 531 (python)
   CGroup: /system.slice/AlexaPi.service
           └─531 /usr/bin/python /opt/AlexaPi/src/main.py

Warning: Unit file changed on disk, 'systemctl daemon-reload' recommended.

mohaggag, 2016-11-07:
journalctl -u AlexaPi.service
No journal files were found.

mohaggag, 2016-11-07:
arecord -D trysomedevicehere
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM trysomedevicehere
arecord: main:722: audio open error: No such file or directory

renekliment, 2016-11-07:
> Active: active (running) since Mon 2016-11-07 19:17:04 UTC; 1h 57min ago

So AlexaPi is already running as a service and probably accessing your microphone. If you want to run interactively to debug, you have to stop if first: `systemctl stop AlexaPi`

> arecord -D trysomedevicehere
You have to replace _trysomedevicehere_ with something real if you want to try that ;)

mohaggag, 2016-11-07:
arecord -D hw:1,0
Recording WAVE 'stdin' : Unsigned 8 bit, Rate 8000 Hz, Mono
arecord: set_params:1233: Sample format non available
Available formats:

mohaggag, 2016-11-07:
sorry i did 
and yes i stoped alexa pi before

mohaggag, 2016-11-07:
systemctl status AlexaPi.service
● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
  Drop-In: /etc/systemd/system/AlexaPi.service.d
           └─restart.conf
   Active: inactive (dead) since Mon 2016-11-07 21:17:31 UTC; 4min 18s ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
  Process: 531 ExecStart=/usr/bin/python /opt/AlexaPi/src/main.py (code=exited, status=0/SUCCESS)
 Main PID: 531 (code=exited, status=0/SUCCESS)

Warning: Unit file changed on disk, 'systemctl daemon-reload' recommended.

mohaggag, 2016-11-07:
/opt/AlexaPi/src/main.py -d
Checking Internet Connection...
Connection OK
Play_Audio Request for: /opt/AlexaPi/src/resources/hello.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended

and still not listening

renekliment, 2016-11-07:
Alright. Do `arecord -D yourdevicehere test.wav`
And `aplay test.wav`
Can you hear your voice in the recording fine?

mohaggag, 2016-11-07:
`arecord -D hw:1,0 test.wav
Recording WAVE 'test.wav' : Unsigned 8 bit, Rate 8000 Hz, Mono
arecord: set_params:1233: Sample format non available
Available formats:
- S16_LE
`

`arecord -D plughw:CARD=Device,DEV=0 test.wav
Recording WAVE 'test.wav' : Unsigned 8 bit, Rate 8000 Hz, Mono
`

ctrl c 
`CAborted by signal Interrupt...
arecord: pcm_read:2031: read error: Interrupted system call
`

mohaggag, 2016-11-07:
arecord -D plughw:1 test.wav
Recording WAVE 'test.wav' : Unsigned 8 bit, Rate 8000 Hz, Mono
^CAborted by signal Interrupt...
arecord: pcm_read:2031: read error: Interrupted system call

mohaggag, 2016-11-07:
and after successful test yes i can hear myself

mohaggag, 2016-11-07:
i still don't know why alexa doesn't hear me  ..  i think she shy from me or something

renekliment, 2016-11-07:
So the configuration is okay. Try tweaking the microphone intensity in `alsamixer`. Sometimes, I have to say the word "Alexa" in a very specific way. 
You can try changing the word to "Max" in the configuration as someone pointed out it works better since it's less syllables or something.

mohaggag, 2016-11-07:
thank you at all
but i did every thing and i don't know what's wrong

mohaggag, 2016-11-07:
i give up

mohaggag, 2016-11-07:
btw i have the same issue with snowboy on the original avs  but with sensory it works like a charm 
and also i don't know why !!!1

mohaggag, 2016-11-08:
Last thing now it works but i must put the mic in my lips and speak loudly i used two sound cards and 3 mics .... Same results

illperipherals, 2016-11-08:
Have you verified your levels and auto-gain in alsamixer? This does not sound like a software issue.

renekliment, 2016-11-08:
Aha! What about the _Capture tab_ (F4) in `alsamixer`?

mohaggag, 2016-11-09:
i did every thing i don't know but i face same problem with snowboy but sensory works very well

mohaggag, 2016-11-09:
is this will help now it's not listening at all 

sudo journalctl -u AlexaPi.service
-- Logs begin at Wed 2016-11-09 19:33:35 UTC, end at Wed 2016-11-09 19:49:14 UTC
Nov 09 19:33:38 raspberrypi systemd[1]: Starting Alexa client for all your devic
Nov 09 19:33:38 raspberrypi systemd[1]: Started Alexa client for all your device
Nov 09 19:34:09 raspberrypi python[522]: Home directory not accessible: Permissi
Nov 09 19:34:09 raspberrypi python[522]: W: [pulseaudio] core-util.c: Failed to
Nov 09 19:34:09 raspberrypi python[522]: W: [pulseaudio] daemon-conf.c: Failed t
Nov 09 19:34:09 raspberrypi python[522]: Home directory not accessible: Permissi
Nov 09 19:34:09 raspberrypi python[522]: W: [pulseaudio] core-util.c: Failed to
Nov 09 19:34:09 raspberrypi python[522]: W: [pulseaudio] daemon-conf.c: Failed t
Nov 09 19:34:09 raspberrypi python[522]: Home directory not accessible: Permissi
Nov 09 19:34:09 raspberrypi python[522]: W: [pulseaudio] core-util.c: Failed to
Nov 09 19:34:09 raspberrypi python[522]: W: [pulseaudio] daemon-conf.c: Failed t
Nov 09 19:34:42 raspberrypi systemd[1]: Stopping Alexa client for all your devic
Nov 09 19:34:42 raspberrypi python[522]: Checking Internet Connection...
Nov 09 19:34:42 raspberrypi python[522]: Connection OK
Nov 09 19:34:42 raspberrypi systemd[1]: Stopped Alexa client for all your device
lines 1-16/16 (END)
-- Logs begin at Wed 2016-11-09 19:33:35 UTC, end at Wed 2016-11-09 19:49:14 UTC. --
Nov 09 19:33:38 raspberrypi systemd[1]: Starting Alexa client for all your devices...
Nov 09 19:33:38 raspberrypi systemd[1]: Started Alexa client for all your devices.
Nov 09 19:34:09 raspberrypi python[522]: Home directory not accessible: Permission denie
Nov 09 19:34:09 raspberrypi python[522]: W: [pulseaudio] core-util.c: Failed to open con
Nov 09 19:34:09 raspberrypi python[522]: W: [pulseaudio] daemon-conf.c: Failed to open c
Nov 09 19:34:09 raspberrypi python[522]: Home directory not accessible: Permission denie
Nov 09 19:34:09 raspberrypi python[522]: W: [pulseaudio] core-util.c: Failed to open con
Nov 09 19:34:09 raspberrypi python[522]: W: [pulseaudio] daemon-conf.c: Failed to open c
Nov 09 19:34:09 raspberrypi python[522]: Home directory not accessible: Permission denie
Nov 09 19:34:09 raspberrypi python[522]: W: [pulseaudio] core-util.c: Failed to open con
Nov 09 19:34:09 raspberrypi python[522]: W: [pulseaudio] daemon-conf.c: Failed to open c
Nov 09 19:34:42 raspberrypi systemd[1]: Stopping Alexa client for all your devices...
Nov 09 19:34:42 raspberrypi python[522]: Checking Internet Connection...
Nov 09 19:34:42 raspberrypi python[522]: Connection OK
Nov 09 19:34:42 raspberrypi systemd[1]: Stopped Alexa client for all your devices.
~
~
~
~
~
~
~
lines 1-16/16 (END)
-- Logs begin at Wed 2016-11-09 19:33:35 UTC, end at Wed 2016-11-09 19:49:14 UTC. --
Nov 09 19:33:38 raspberrypi systemd[1]: Starting Alexa client for all your devices...
Nov 09 19:33:38 raspberrypi systemd[1]: Started Alexa client for all your devices.
Nov 09 19:34:09 raspberrypi python[522]: Home directory not accessible: Permission denied
Nov 09 19:34:09 raspberrypi python[522]: W: [pulseaudio] core-util.c: Failed to open configuratio
Nov 09 19:34:09 raspberrypi python[522]: W: [pulseaudio] daemon-conf.c: Failed to open configurat
Nov 09 19:34:09 raspberrypi python[522]: Home directory not accessible: Permission denied
Nov 09 19:34:09 raspberrypi python[522]: W: [pulseaudio] core-util.c: Failed to open configuratio
Nov 09 19:34:09 raspberrypi python[522]: W: [pulseaudio] daemon-conf.c: Failed to open configurat
Nov 09 19:34:09 raspberrypi python[522]: Home directory not accessible: Permission denied
Nov 09 19:34:09 raspberrypi python[522]: W: [pulseaudio] core-util.c: Failed to open configuratio
Nov 09 19:34:09 raspberrypi python[522]: W: [pulseaudio] daemon-conf.c: Failed to open configurat
Nov 09 19:34:42 raspberrypi systemd[1]: Stopping Alexa client for all your devices...
Nov 09 19:34:42 raspberrypi python[522]: Checking Internet Connection...
Nov 09 19:34:42 raspberrypi python[522]: Connection OK
Nov 09 19:34:42 raspberrypi systemd[1]: Stopped Alexa client for all your devices.
~
~
~
~
~
~
~
lines 1-16/16 (END)
-- Logs begin at Wed 2016-11-09 19:33:35 UTC, end at Wed 2016-11-09 19:49:14 UTC. --
Nov 09 19:33:38 raspberrypi systemd[1]: Starting Alexa client for all your devices...
[... 65 lines cut ...]
~
~
~
~
 ESCOD
-- Logs begin at Wed 2016-11-09 19:33:35 UTC, end at Wed 2016-11-09 19:49:14 UTC. --
Nov 09 19:33:38 raspberrypi systemd[1]: Starting Alexa client for all your devices...
Nov 09 19:33:38 raspberrypi systemd[1]: Started Alexa client for all your devices.
Nov 09 19:34:09 raspberrypi python[522]: Home directory not accessible: Permission denied
Nov 09 19:34:09 raspberrypi python[522]: W: [pulseaudio] core-util.c: Failed to open configuration file '/home/alexapi/.config/pul
Nov 09 19:34:09 raspberrypi python[522]: W: [pulseaudio] daemon-conf.c: Failed to open configuration file: Permission denied
Nov 09 19:34:09 raspberrypi python[522]: Home directory not accessible: Permission denied
Nov 09 19:34:09 raspberrypi python[522]: W: [pulseaudio] core-util.c: Failed to open configuration file '/home/alexapi/.config/pul
Nov 09 19:34:09 raspberrypi python[522]: W: [pulseaudio] daemon-conf.c: Failed to open configuration file: Permission denied
Nov 09 19:34:09 raspberrypi python[522]: Home directory not accessible: Permission denied
Nov 09 19:34:09 raspberrypi python[522]: W: [pulseaudio] core-util.c: Failed to open configuration file '/home/alexapi/.config/pul
Nov 09 19:34:09 raspberrypi python[522]: W: [pulseaudio] daemon-conf.c: Failed to open configuration file: Permission denied
Nov 09 19:34:42 raspberrypi systemd[1]: Stopping Alexa client for all your devices...
Nov 09 19:34:42 raspberrypi python[522]: Checking Internet Connection...
Nov 09 19:34:42 raspberrypi python[522]: Connection OK
Nov 09 19:34:42 raspberrypi systemd[1]: Stopped Alexa client for all your devices.
~
~
~
~

maso27, 2016-11-09:
@mohaggag 
I'm really sorry you're having so much trouble.  It's got to be frustrating.
One thing we keep having trouble with is when pulseaudio is installed on the system.  Which is something that is automatic on the full desktop versions of raspbian.

If you aren't planning on using the desktop on your pi at all, can I suggest that you try using raspbian lite instead of pixel?

The plan is to be able to support other audio engines down the road, which will hopefully have less trouble with pulseaudio.  But for now this seems to be the most foolproof solution.

mohaggag, 2016-11-09:
i am starting a new project now

mohaggag, 2016-11-09:
new every thing with rasbian lite same issue alexa not listening thanks at all you can close this issue now

illperipherals, 2016-11-09:
While I agree that pulse is causing problems, this is the pathway to doing multi-room audio, and we need to keep that on the roadmap.

renekliment, 2016-11-20:
Closing this, since it was a matter of configuration and later the pronunciation.

closed 2016-11-20 by renekliment

gilo-agilo, 2017-04-24:
I found `ERROR: Device or resource busy [plughw:CARD=Device,DEV=0]` error when Raspberry Pi reboots.
If I restart service manually after reboot (`sudo systemctl -u AlexaPi.service`), then AlexaPi works fine.
It looks like USB mic (in my case: C-Media Electronics Inc) isn't get ready fast enough and AlexaPi fails.
I was wondering if I should tune service dependencies (like adding extra target) or just play with pocketsphinxtrigger.py - to allow 3 attempts with 10 seconds between them before failing (as it used to).

**UPDATE:**
Simple [hack](https://github.com/gilo-agilo/AlexaPi/commit/eb672084305460293037f6a351e88933e22cb05a) with pocketsphinxtrigger.py worked for me. Now after boot/reboot AlexaPi stay operational.

renekliment, 2017-04-24:
@gilo-agilo No. Set up PulseAudio properly.

dikkedimi, 2017-10-03:
I'm trying to run AlexaPi on a pi I previously installed google assistant on. I also have the resource busy error now. Could it be that I need to configure GAssistant to also use pulse? Anyone have any experience manually installing both? I've tried this package and also the PiAssistant package (which is a repack of this), but both have this problem.

I'm using a Logitech C920 webcam as audio capture until my AIY HAT arrives.