alexa-pi/AlexaPi #79: Pi3 running Archlinux setup instructions.

jrlambs, 2016-11-09:
I just got done starting from a fresh ArchLinuxArm install and documenting my steps.

Thanks to @maso27 for the information I used and to @illperipherals for helping me through some of the issues. 

Followed instructions to create a file system for arch from: [https://archlinuxarm.org/platforms/armv8/broadcom/raspberry-pi-3](https://archlinuxarm.org/platforms/armv8/broadcom/raspberry-pi-3)  (Installation Tab)
From initial bootup:

```
pacman -S sudo
nano /etc/sudoers
```

Add the following lines
alarm   ALL=(ALL) ALL
Defaults:alarm      !authenticate

```
pacman -S python2
pacman -S python2-pip
pacman -S python2-virtualenv
pacman -S git
pacman -S vox
pacman -S gcc

cd /usr/bin
```

# AS ROOT
```
ln -sf python2 python
ln -sf pip2 pip
ln -sf virtualenv2 virtualenv
```

#Follow maso27's instructions

#"apt-get update" => "pacman -Sy"
#"apt-get install xxxx -y" => "pacman -S xxxx --noconfirm"
#"python-dev" => "python2"
#"libasound2-dev" => "alsa-lib" and "alsa-utils"
#"libpulse-dev" => "libpulse"
#"python-yaml" => "python2-yaml" 
#remove "python-alsaaudio"
#add "RPi.GPIO>=0.6.2" to requirements.txt [link to updated file](https://drive.google.com/open?id=0B6sTL4s0D8Mdbk5uSlhucEZPcWM)
#add "dtparam=audio=on" to /boot/config.txt
#shairport-sync is already a package in archlinux repos, so just "pacman -S shairport-sync --noconfirm" to replace most of the build steps there.

Update requirements.txt file  here's a link

[.. I'll attach the new script.](https://drive.google.com/open?id=0B6sTL4s0D8MddE5BTE5sLTV0TU0)

At this point I had an issue with the line below... I installed each item on its own.
1) When I got to VLC I was asked 2 questions.  I selected option "1", the default option, for both.
2) obviously when I got to python-yaml I needed to change it to python2-yaml.. that might have been the whole issue.. now that all of the things are installed I can run that pacman command no problem... so it may or may not be an issue
pacman -S swig alsa-lib alsa-utils memcached vlc libpulse gcc python-yaml --noconfirm

#So at this point I the script finished successfully
#Follow maso27's other set of instructions from running as user/group alexapi
```
groupadd gpio -r
usermod -a -G gpio alexapi
nano /etc/udev/rules.d/99-gpio.rules
```

#Insert the following (3 lines):
```
SUBSYSTEM=="bcm2835-gpiomem", KERNEL=="gpiomem", GROUP="gpio", MODE="0660"
SUBSYSTEM=="gpio", KERNEL=="gpiochip*", ACTION=="add", PROGRAM="/bin/sh -c 'chown root:gpio /sys/class/gpio/export /sys/class/gpio/unexport ; chmod 220 /sys/class/gpio/export /sys/class/gpio/unexport'"
SUBSYSTEM=="gpio", KERNEL=="gpio*", ACTION=="add", PROGRAM="/bin/sh -c 'chown root:gpio /sys%p/active_low /sys%p/direction /sys%p/edge /sys%p/value ; chmod 660 /sys%p/active_low /sys%p/direction /sys%p/edge /sys%p/value'"
```

Reboot.... it's working.. all audio is coming through the onboard sound card instead of the USB one.. will work on that later.

renekliment, 2016-11-09:
We'll add this to this setup script once things are calm elsewhere :)

What do you need the sudo(ers) stuff for? You don't need that for AlexaPi.

You can set your output device / card in config.yaml.

jrlambs, 2016-11-09:
Yeah, you don't.  I went back to retrace my steps after I had everything working and for some reason decided to put that in my writeup.. don't know why.  def don't need the sudo stuff for the purposes of AlexaPi

renekliment, 2016-11-09:
Also, please paste any small changes here in a code block and bigger changes / files as [gists](https://gist.github.com/) so people don't have to download stuff just to see and they can comment, edit, share, whatever.

jrlambs, 2016-11-09:
Ah, no problem.. I've updated the links to point to gists, wasn't sure the correct way to do that before... still not positive I did it right.. but it's better than it was

renekliment, 2016-11-09:
Great!

renekliment, 2016-11-12:
#82 should include all needed steps. Could you test?

commit 6f64ae9d (2016-11-27): Setup script refactor (#82), fixes #60, fixes #79

* modularized and generalized, cleanup

* config: renamed ProductID to Device_Type_ID

* support for Arch Linux (instructions by @maso27)

* support for currently supported boards in AlexaPI (GPIO libs install, etc)

* added unit overrides for various things

* set up GPIO permissions (udev)

* use pip instead of system packaging where possible

* disabled pip cache due to pypa/pip#2984

* change default device_platform to dummy

* set device-specific config on install
closed 2016-11-27 by renekliment