alexa-pi/AlexaPi #80: TuneIn

nathankent22, 2016-11-09:
Some radios on TuneIn return the following error -

************************************************
**                                            **
**  No css library available. See             **
**  /usr/share/doc/libdvdread4/README.css     **
**  for more information.                     **
**                                            **
************************************************
libdvdread: Couldn't find device name.
libdvdread:DVDOpenFilePath:findDVDFile /VIDEO_TS/VIDEO_TS.IFO failed
libdvdread:DVDOpenFilePath:findDVDFile /VIDEO_TS/VIDEO_TS.BUP failed
libdvdread: Can't open file VIDEO_TS.IFO.
libdvdnav: vm: failed to read VIDEO_TS.IFO
[69801fb8] core demux error: option sub-original-fps does not exist
[69802120] core input error: no suitable demux module for `file/subtitle:///opt/AlexaPi/src//requirements.txt'

renekliment, 2016-11-09:
Please paste the whole output.
Stop AlexaPi service `systemctl stop AlexaPi.service` and run it interactively in debug mode `/opt/AlexaPi/src/main.py -d`

renekliment, 2016-12-05:
@nathankent22 Any feedback on this? We can't help otherwise I'm afraid.

renekliment, 2016-12-10:
Closing due to incompleteness.

closed 2016-12-10 by renekliment