alexa-pi/AlexaPi #81: Pi Not starting to Listen

Karansheth, 2016-11-09:
I did the setup as mentioned and have used usb webcam's microphone , i tested it using "arecord -D plughw:1 test.wav" command and the microphone works.

When the Pi boots up it Say's Hello but after that no response to any word so i tried debugging as mentioned in wiki here are the logs:

"systemctl status AlexaPi.service"

● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
  Drop-In: /etc/systemd/system/AlexaPi.service.d
           └─restart.conf
   Active: inactive (dead) since Wed 2016-11-09 15:13:02 UTC; 2s ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
  Process: 576 ExecStart=/usr/bin/python /opt/AlexaPi/src/main.py (code=exited, status=0/SUCCESS)
 Main PID: 576 (code=exited, status=0/SUCCESS)

"/opt/AlexaPi/src/main.py -d"

Checking Internet Connection...
Connection OK
Play_Audio Request for: /opt/AlexaPi/src/resources/hello.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended

"sudo journalctl -u AlexaPi.service"

-- Logs begin at Wed 2016-11-09 18:02:26 UTC, end at Wed 2016-11-09 18:23:15 UTC. --
Nov 09 18:02:31 raspberrypi systemd[1]: Starting Alexa client for all your devices...
Nov 09 18:02:31 raspberrypi systemd[1]: Started Alexa client for all your devices.
Nov 09 18:03:05 raspberrypi python[552]: Home directory not accessible: Permission denied
Nov 09 18:03:05 raspberrypi python[552]: W: [pulseaudio] core-util.c: Failed to open configuration file '/home/alexapi/.config/pulse//daemon.conf': Permission denied
Nov 09 18:03:05 raspberrypi python[552]: W: [pulseaudio] daemon-conf.c: Failed to open configuration file: Permission denied
Nov 09 18:03:05 raspberrypi python[552]: Home directory not accessible: Permission denied
Nov 09 18:03:05 raspberrypi python[552]: W: [pulseaudio] core-util.c: Failed to open configuration file '/home/alexapi/.config/pulse//daemon.conf': Permission denied
Nov 09 18:03:05 raspberrypi python[552]: W: [pulseaudio] daemon-conf.c: Failed to open configuration file: Permission denied
Nov 09 18:03:05 raspberrypi python[552]: Home directory not accessible: Permission denied
Nov 09 18:03:05 raspberrypi python[552]: W: [pulseaudio] core-util.c: Failed to open configuration file '/home/alexapi/.config/pulse//daemon.conf': Permission denied
Nov 09 18:03:05 raspberrypi python[552]: W: [pulseaudio] daemon-conf.c: Failed to open configuration file: Permission denied
Nov 09 18:09:47 raspberrypi systemd[1]: Stopping Alexa client for all your devices...
Nov 09 18:09:47 raspberrypi python[552]: Checking Internet Connection...
Nov 09 18:09:47 raspberrypi python[552]: Connection OK
Nov 09 18:09:47 raspberrypi systemd[1]: Stopped Alexa client for all your devices.
Nov 09 18:11:22 raspberrypi systemd[1]: Starting Alexa client for all your devices...
Nov 09 18:11:23 raspberrypi systemd[1]: Started Alexa client for all your devices.
Nov 09 18:11:33 raspberrypi python[1163]: Home directory not accessible: Permission denied
Nov 09 18:11:33 raspberrypi python[1163]: W: [pulseaudio] core-util.c: Failed to open configuration file '/home/alexapi/.config/pulse//daemon.conf': Permission denied
Nov 09 18:11:33 raspberrypi python[1163]: W: [pulseaudio] daemon-conf.c: Failed to open configuration file: Permission denied
Nov 09 18:11:33 raspberrypi python[1163]: Home directory not accessible: Permission denied
Nov 09 18:11:33 raspberrypi python[1163]: W: [pulseaudio] core-util.c: Failed to open configuration file '/home/alexapi/.config/pulse//daemon.conf': Permission denied
Nov 09 18:11:33 raspberrypi python[1163]: W: [pulseaudio] daemon-conf.c: Failed to open configuration file: Permission denied
    
It would be very nice if anyone can help me

Thanks in Advance

renekliment, 2016-11-09:
Unfortunately, we currently don't support PulseAudio. If you don't need it, please uninstall it.

Karansheth, 2016-11-10:
I uninstalled it tried but the logs and the problem remain same

renekliment, 2016-11-10:
Can you guys try https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#disabling-pa ?
This should at least get rid of the silly messages. Please post the log then.

Does it work / crash in the debug mode https://github.com/alexa-pi/AlexaPi/wiki/Debugging#debug-mode ?
If it doesn't crash, have you tried adjusting mic sensitivity?

renekliment, 2016-11-10:
I'm gonna be pretentious and close this as a duplicate of #38. Please post any further comments over there. Thanks!

closed 2016-11-10 by renekliment

illperipherals, 2016-11-11:
I think it may help out in debugging and tracking down audio issues if we use this as a standard form of data collection:
Enter these two commands in the terminal:
cd ~/
wget http://www.alsa-project.org/alsa-info.sh -O alsa-info.sh && bash alsa-info.sh