alexa-pi/AlexaPi #87: Various DOCS update for v1.3

renekliment, 2016-11-11:
- update CHANGELOG.md
- CONTRIBUTING.md: remove the  _spaces for alignment_
- .github/ISSUE_TEMPLATE: add _is:issue_ to the issue link

renekliment, 2016-12-07:
So it looks like we're there already! All features for version 1.3 have been merged in `master`.

I've set the release date to the **21th** in the changelog, which is exactly 2 weeks from now. I'd like to have this time window so people can upgrade and test, before we tag the release. If there should be any issues in the mean time, it should be fairly easy to fix them and after that period I'll merge #101 (feel free to check it out), which updates the docs and I'll tag the release.

It's been crazy 3 months I have to say. A lot of work has been done and a lot of experience has been gained (on my side at least). **I'd like to thank all the people who have contributed to this project.** It's come a long way since the [decision](https://github.com/sammachin/AlexaPi/pull/125) to do a one collaborative effort and I think it has been a _very right_ decision. We now support all the devices for which there were forks and once we support snowboy (and probably do a sox playback_handler), we will eliminate the need for all the forks and everyone can focus on this awesome version. We've also made some drastic changes to the code and it is now much more friendly to the developers, which only helps to catch new people's interest and prevents us from going crazy.

I'd like to draft my vision for the future, mainly _v1.4_. The main objective there is going to be refactoring the triggering code and the AVS code and to support sox. This is going to have some implications:
- With the triggering code refactored, the voice activation is going to be optional (not mandatory as it is now). This is the one feature that Sam's original repo has that we don't. So after this is possible, I'd like to change the note in the repo to recommend **all users** to switch to our repo. Also, some more aggressive notes / links can be added.
- With the triggering code refactored (this includes modularization), we'll add snowboy support, which _is the shit_.
- With the AVS refactor, the code should be much cleaner and it should be possible to start integrating the new API in our repo.
- With _sox_ we should fix audio issues on Orange Pis with legacy kernel, CHIPs and other boards where vlc doesn't work properly for some reason. Also, this should open the door to OpenWRT, which is gonna be awesome!

And to get an idea what's coming, I'd like to ask:
- @lechat What's the status of your `logging` branch?
- @dgonano What's the status of your `magicmirror` branch? Would you like to see it merged upstream?
- @davidm How are you doing on the `avs_v2` branch? When the AVS refactor is done, we could try integrating it as mentioned above if you are interested. Obviously we could discuss the details only after we get there and start the work on it to make sure the interface is there. This will require resolving some merge conflicts since there has been quite a few commits here, but it's something we can work together on and it's definitely worth it. Also, I see you're doing it quite modular, which is a good thing!

So this is it from me. What are **your thoughts**? What are **your visions** for AlexaPi?

dgonano, 2016-12-08:
@renekliment I have this latest version master up and running. It's working but I'm having an issue with music/radio playback. 

I asked it to "play triple j radio" and it did. While playing I invoked alexa again but it stopped working:
```
INFO: cmn_live.c(88): Update from < 36.42 13.75 -11.92 -6.39  5.08  0.94 -4.97 -1.15 -1.66 -0.59  0.26  0.61  0.94 >
INFO: cmn_live.c(105): Update to   < 37.69 12.51 -11.97 -4.01  5.57 -1.23 -5.96 -1.26  0.28 -4.09 -0.75  1.11 -0.90 >
INFO: cmn_live.c(88): Update from < 37.69 12.51 -11.97 -4.01  5.57 -1.23 -5.96 -1.26  0.28 -4.09 -0.75  1.11 -0.90 >
INFO: cmn_live.c(105): Update to   < 38.30 11.69 -10.22 -1.03  6.00 -1.43 -5.92 -1.02  1.42 -2.27  0.24  2.13  1.86 >
Player State: State.Stopped
Play_Audio Request for: /opt/AlexaPi/src/resources/alexayes.mp3
```
The music stopped but it didn't say "yes" and won't accept anymore input. have to crash the program and start again. 

I also tried playing a different station, Nova 1069. It says "playing nova 1069 on tunein" but no audio played
```
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
TUNE IN URL = http://opml.radiotime.com/Tune.ashx?id=s87120&formats=aac,mp3&partnerId=4JqugguZ&serial=AEOK4VMXAH7UA7B3KRR7GPYG63IA&latlon=40.75,-73.98333
```
It also stopped responding to any further commands but managed to quit with ctrl-c

Magic Mirror platform:
I'm just working on getting it in a mergable state now. Everything is working but I just want to tidy it up and add some documentation. Is there a good place I can add details about the platform or just at the start of the file for now? 
Also, I haven't done anything for the install script so a user can select the new platform. I'm just setting it in the config file at the moment.

renekliment, 2016-12-08:
@dgonano Put docs in our wiki.

Thanks for catching that bug. That's exactly something we have to fix before the release.

renekliment, 2016-12-10:
Alright. So the bug found by @dgonano should be fixed by #103. Please test.

By the way, @lechat, you were so right. I see now that the `if debug` is the worst _f******_ idea ever. We should definitely switch to sane logging first thing after the release. I don't know, up until now I just ignored it, but it takes up twice the number of rows, sometimes even more because for it to be readable it requires space around and it's useless for "profiling", because it does't  print time ... and I haven't needed filtering yet, so I'm not talking about that ... I guess I've just seen the light :smile:

dgonano, 2016-12-10:
Testing. 

Yeah, logging handlers are great. Allows multiple levels of logging easily in addition to your stuff

lechat, 2016-12-11:
Logging w/o timestamps is of course bad idea. I may find some time this week to finish logging branch.

renekliment, 2016-12-17:
Also, I wanted to do something like Codacy for AlexaPi, but all those apps have this:

> This application will be able to read and write all public repository data. This includes the following:
> Code
> Issues
> Pull requests
> Wikis
> Settings
> Webhooks and services
> Deploy keys

and I am not gonna authorize that ... 

Do you guys know an app that doesn't need this? If I remember correctly, it's a GitHub issue and it has been criticized for it, but that doesn't change the end result.

PR #101 (2016-12-21): v1.3 docs update, fix #87
Fixes #87.

Mainly updates the `CHANGELOG.md`, which we should update with every commit (if it's relevant) from now on.

Tiny update to other documentation files.
commit 24b4990e (2016-12-21): v1.3 docs update, fix #87 (#101)

* ChangeLog update for v1.3

* small docs update

* simpler PR checkout as pointed out by @dgonano
closed 2016-12-21 by renekliment