alexa-pi/AlexaPi #90: Requirements for Amazon Music?

SneakyDave, 2016-11-13:
OK, so I've applied for Amazon Music, but my application is still "PENDING", but I wanted to verify what others have used for options on applying. 

I have all these options checked:
My device supports both HTTP Live Streaming (HLS) and progressive download as means to render music playback. 
My device will not allow offline playback of Amazon Music content. 
My device will not incorporate Amazon Music into a game experience. 
My device will not enable synchronization of Amazon Music with video, photos or other visual media, e.g. a slideshow synchronized with Amazon Music. 

And then for "Please briefly describe your device’s buffering/caching mechanisms. ", I have "AlexaPi Setup"

And then for "Please briefly describe your device’s music playback capabilities, e.g., speaker quality, audio output, Bluetooth support, etc.", I used "Single, low powered speaker for playback via 3.5 inch jack".

Is there an average length of time on how long applications take?

renekliment, 2016-11-13:
Same here. I hope they'll come through, because it's hard to develop something without testing it.

saurabhreigns, 2016-11-30:
Same here. It has been more than 24 hours and I am not able to listen to even sample music. Can I use some other music service to serve the purpose?. I did try using alexa dahboard (intentended for echo,echo dot and tap users ) on amazon.uk. It lets me login and show me my device's name but it never recognises that my device is online. In setup options, it doesnt have setup options for hackers like us. Please share some knowledge. 
PS: I am from India and I desperately want to listen to local weather response, local news and local radio stations (apart from music ).

renekliment, 2016-12-10:
I got a response after I asked about the status:

> Hello René, thank you for your inquiry.  Currently all Amazon Music approvals are in a pending state. Unfortunately, there is no further information we can share with you at this time. We appreciate your patience while we work towards a resolution.

dgonano, 2016-12-11:
Well that explains why mine weren't getting through either then.

renekliment, 2017-03-27:
More info on this is on the wiki: https://github.com/alexa-pi/AlexaPi/wiki/Q&A-(FAQ)#does-alexapi-support-amazon-music

Closing as there is nothing we can do currently.

closed 2017-03-27 by renekliment

maso27, 2017-03-27:
It's a real bummer that they aren't allowing that anymore.
Can you play stations from iheartradio?

jeffmcnabb, 2019-11-16:
Was there ever an update on Amazon closing this? Do we know if the we could use AVS SDK to create an Echo type device that can have music cast to it?

illperipherals, 2019-11-16:
The RAK WisCore boards allow for music services. I have done some dev work on them, but they seem to be out of production. (they use OpenWRT)