alexa-pi/AlexaPi #91: Missing requirements

ibaranov-cp, 2016-11-13:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

Hey all,
On my install on a Pi3 with Ubuntu Mate 16.04, the following are required during install and use, but had to be done manually for some reason:

cherrypy (pip)
webrtcvad (pip)
python-memcache (apt-get)
pocketsphinx (pip)

Just thought to note it for people

ibaranov-cp, 2016-11-14:
Yes, via the setup script.

Currently away from the device, will do asap.

renekliment, 2016-11-19:
So I did an install yesterday on Ubuntu (Armbian actually) and the package `python-setuptools` wasn't installed by default, so the _pip_ commands failed. So now I don't think it's caused by a different default py/pip version (set to 3), but by just that. 
I suppose you got some error during the setup, right?

I'm confident it's fixed in #82 and since we don't officially support installation on anything else than Raspbian now (will be solved by the #82), I'm closing this now.

Thanks for the report! It didn't occur to me that some packages were not installed there by default.

closed 2016-11-19 by renekliment