alexa-pi/AlexaPi #92: python mmap () failed: Cannot allocate memory

revere521, 2016-11-19:
Here is a section of the log from restart to crash with the mmap memory allocation error: 
```
Nov 18 17:02:18 MagicMirror01 systemd[1]: Starting Alexa client for all your devices...
Nov 18 17:02:18 MagicMirror01 systemd[1]: Started Alexa client for all your devices.
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
Nov 18 18:16:49 MagicMirror01 python[5914]: mmap() failed: Cannot allocate memory
[... 197 lines cut ...]
```

**Steps to reproduce:**

1. unknown
2. 
3. 

**Traceback (if applicable):**
```
Traceback (most recent call last):
Nov 18 18:16:51 MagicMirror01 python[5914]: File "/opt/AlexaPi/src/main.py", line 598, in <module>
Nov 18 18:16:51 MagicMirror01 python[5914]: loop()
Nov 18 18:16:51 MagicMirror01 python[5914]: File "/opt/AlexaPi/src/main.py", line 509, in loop
Nov 18 18:16:51 MagicMirror01 python[5914]: inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, config['
Nov 18 18:16:51 MagicMirror01 python[5914]: alsaaudio.ALSAAudioError: Input/output error [pulse]
Nov 18 18:16:51 MagicMirror01 systemd[1]: AlexaPi.service: main process exited, code=exited, status=1/FAILURE
Nov 18 18:16:51 MagicMirror01 systemd[1]: Unit AlexaPi.service entered failed state.
Nov 18 18:16:52 MagicMirror01 systemd[1]: AlexaPi.service holdoff time over, scheduling restart.
```

**Additional info:**
```

```

revere521, 2016-11-30:
i have been trying to find the nature of the error, and unfortunately i cant get systemtap to run on my pi (more accurately to see the correct kernel headers) - but from what I've been able to read this has to do with memory over commit.  would it be beneficial to change the inp.setperiodsize?  i'm not sure if the context would be to lower it (if that would 'clear' it or reset the listening frame more often or at a smaller size)  or setting it larger so it does not reach it limit.  Do you have any suggestions for what may be casuing this?  since i'm runnign other applications simultaneously that are using memory - is there a better way to share with overcommit?

renekliment, 2016-12-11:
I have no idea what might cause this. Not that there has been a change that should have resolved this, but are you still experiencing the bug with up-to-date `master`?

revere521, 2016-12-11:
@renekliment - i set the pi to reboot every morning in the crontab - that may have mitigated the issue.  This is all thats in the log for today (running the current master so i can have the MMM_Alexa plugin - sweet btw):  `Dec 11 07:00:25 MagicMirror01 systemd[1]: Starting Alexa client for all your devices...
Dec 11 07:00:25 MagicMirror01 systemd[1]: Started Alexa client for all your devices.
`
You can go ahead and close this one - thanks!