alexa-pi/AlexaPi #93: No voice from AVS

lion6230i, 2016-11-23:
Here is my debug information 
```
[root@localhost src]# python main.py -d
INFO: pocketsphinx.c(152): Parsed model-specific feature parameters from /usr/local/lib/python2.7/site-packages/pocketsphinx/model/en-us/feat.params
Current configuration:
[NAME]			[DEFLT]		[VALUE]
-agc			none		none
-agcthresh		2.0		2.000000e+00
-allphone				
-allphone_ci		no		no
-alpha			0.97		9.700000e-01
-ascale			20.0		2.000000e+01
-aw			1		1
-backtrace		no		no
-beam			1e-48		1.000000e-48
-bestpath		yes		yes
-bestpathlw		9.5		9.500000e+00
-ceplen			13		13
-cmn			live		batch
-cmninit		40,3,-1		41.00,-5.29,-0.12,5.09,2.48,-4.07,-1.37,-1.78,-5.08,-2.05,-6.45,-1.42,1.17
-compallsen		no		no
-debug					0
-dict					/usr/local/lib/python2.7/site-packages/pocketsphinx/model/cmudict-en-us.dict
-dictcase		no		no
-dither			no		no
-doublebw		no		no
-ds			1		1
-fdict					
-feat			1s_c_d_dd	1s_c_d_dd
-featparams				
-fillprob		1e-8		1.000000e-08
-frate			100		100
-fsg					
-fsgusealtpron		yes		yes
-fsgusefiller		yes		yes
-fwdflat		yes		yes
-fwdflatbeam		1e-64		1.000000e-64
-fwdflatefwid		4		4
-fwdflatlw		8.5		8.500000e+00
-fwdflatsfwin		25		25
-fwdflatwbeam		7e-29		7.000000e-29
-fwdtree		yes		yes
-hmm					/usr/local/lib/python2.7/site-packages/pocketsphinx/model/en-us
-input_endian		little		little
-jsgf					
-keyphrase				alexa
-kws					
-kws_delay		10		10
-kws_plp		1e-1		1.000000e-01
-kws_threshold		1		1.000000e-05
-latsize		5000		5000
-lda					
-ldadim			0		0
-lifter			0		22
-lm					
-lmctl					
-lmname					
-logbase		1.0001		1.000100e+00
-logfn					
-logspec		no		no
-lowerf			133.33334	1.300000e+02
-lpbeam			1e-40		1.000000e-40
-lponlybeam		7e-29		7.000000e-29
-lw			6.5		6.500000e+00
-maxhmmpf		30000		30000
-maxwpf			-1		-1
-mdef					
-mean					
-mfclogdir				
-min_endfr		0		0
-mixw					
-mixwfloor		0.0000001	1.000000e-07
-mllr					
-mmap			yes		yes
-ncep			13		13
-nfft			512		512
-nfilt			40		25
-nwpen			1.0		1.000000e+00
-pbeam			1e-48		1.000000e-48
-pip			1.0		1.000000e+00
-pl_beam		1e-10		1.000000e-10
-pl_pbeam		1e-10		1.000000e-10
-pl_pip			1.0		1.000000e+00
-pl_weight		3.0		3.000000e+00
-pl_window		5		5
-rawlogdir				
-remove_dc		no		no
-remove_noise		yes		yes
-remove_silence		yes		yes
-round_filters		yes		yes
-samprate		16000		1.600000e+04
-seed			-1		-1
-sendump				
-senlogdir				
-senmgau				
-silprob		0.005		5.000000e-03
-smoothspec		no		no
-svspec					0-12/13-25/26-38
-tmat					
-tmatfloor		0.0001		1.000000e-04
-topn			4		4
-topn_beam		0		0
-toprule				
-transform		legacy		dct
-unit_area		yes		yes
-upperf			6855.4976	6.800000e+03
-uw			1.0		1.000000e+00
-vad_postspeech		50		50
-vad_prespeech		20		20
-vad_startspeech	10		10
-vad_threshold		2.0		2.000000e+00
-var					
-varfloor		0.0001		1.000000e-04
-varnorm		no		no
-verbose		no		no
-warp_params				
-warp_type		inverse_linear	inverse_linear
-wbeam			7e-29		7.000000e-29
-wip			0.65		6.500000e-01
-wlen			0.025625	2.562500e-02

INFO: feat.c(715): Initializing feature stream to type: '1s_c_d_dd', ceplen=13, CMN='batch', VARNORM='no', AGC='none'
INFO: acmod.c(166): Using subvector specification 0-12/13-25/26-38
INFO: mdef.c(518): Reading model definition: /usr/local/lib/python2.7/site-packages/pocketsphinx/model/en-us/mdef
INFO: mdef.c(531): Found byte-order mark BMDF, assuming this is a binary mdef file
INFO: bin_mdef.c(336): Reading binary model definition: /usr/local/lib/python2.7/site-packages/pocketsphinx/model/en-us/mdef
INFO: bin_mdef.c(516): 42 CI-phone, 137053 CD-phone, 3 emitstate/phone, 126 CI-sen, 5126 Sen, 29324 Sen-Seq
INFO: tmat.c(149): Reading HMM transition probability matrices: /usr/local/lib/python2.7/site-packages/pocketsphinx/model/en-us/transition_matrices
INFO: acmod.c(117): Attempting to use PTM computation module
INFO: ms_gauden.c(127): Reading mixture gaussian parameter: /usr/local/lib/python2.7/site-packages/pocketsphinx/model/en-us/means
INFO: ms_gauden.c(242): 42 codebook, 3 feature, size: 
INFO: ms_gauden.c(244):  128x13
INFO: ms_gauden.c(244):  128x13
INFO: ms_gauden.c(244):  128x13
INFO: ms_gauden.c(127): Reading mixture gaussian parameter: /usr/local/lib/python2.7/site-packages/pocketsphinx/model/en-us/variances
INFO: ms_gauden.c(242): 42 codebook, 3 feature, size: 
INFO: ms_gauden.c(244):  128x13
INFO: ms_gauden.c(244):  128x13
INFO: ms_gauden.c(244):  128x13
INFO: ms_gauden.c(304): 222 variance values floored
INFO: ptm_mgau.c(476): Loading senones from dump file /usr/local/lib/python2.7/site-packages/pocketsphinx/model/en-us/sendump
INFO: ptm_mgau.c(500): BEGIN FILE FORMAT DESCRIPTION
INFO: ptm_mgau.c(563): Rows: 128, Columns: 5126
INFO: ptm_mgau.c(595): Using memory-mapped I/O for senones
INFO: ptm_mgau.c(838): Maximum top-N: 4
INFO: phone_loop_search.c(114): State beam -225 Phone exit beam -225 Insertion penalty 0
INFO: dict.c(320): Allocating 138824 * 20 bytes (2711 KiB) for word entries
INFO: dict.c(333): Reading main dictionary: /usr/local/lib/python2.7/site-packages/pocketsphinx/model/cmudict-en-us.dict
INFO: dict.c(213): Dictionary size 134723, allocated 1016 KiB for strings, 1679 KiB for phones
INFO: dict.c(336): 134723 words read
INFO: dict.c(358): Reading filler dictionary: /usr/local/lib/python2.7/site-packages/pocketsphinx/model/en-us/noisedict
INFO: dict.c(213): Dictionary size 134728, allocated 0 KiB for strings, 0 KiB for phones
INFO: dict.c(361): 5 words read
INFO: dict2pid.c(396): Building PID tables for dictionary
INFO: dict2pid.c(406): Allocating 42^3 * 2 bytes (144 KiB) for word-initial triphones
INFO: dict2pid.c(132): Allocated 21336 bytes (20 KiB) for word-final triphones
INFO: dict2pid.c(196): Allocated 21336 bytes (20 KiB) for single-phone word triphones
INFO: kws_search.c(406): KWS(beam: -1080, plp: -23, default threshold -113, delay 10)
Checking Internet Connection...
Connection OK
MemCached: MemCache: inet:127.0.0.1:11211: connect: [Errno 111] Connection refused.  Marking dead.
setup_complete
Play_Audio Request for: /opt/AlexaPi/src/resources/hello.mp3
indicate_playback True
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
indicate_playback False
Type "a" to trigger Alexa: a
Type "a" to trigger Alexa: Play_Audio Request for: /opt/AlexaPi/src/resources/alexayes.mp3
indicate_playback True
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
indicate_playback False
detected the edge, setting up audio
Starting to listen...
indicate_recording_on True
Debug: End recording
indicate_recording_on False
Debug: Sending audio to be processed
Sending Speech Request...
indicate_processing True
indicate_processing False
Processing Request Response...
Request Response is null (This is OKAY!)
INFO: cmn_live.c(120): Update from < 41.00 -5.29 -0.12  5.09  2.48 -4.07 -1.37 -1.78 -5.08 -2.05 -6.45 -1.42  1.17 >
INFO: cmn_live.c(138): Update to   < 50.53 -17.16 -4.21 -2.66 -2.38 -3.27 -2.26 -1.83 -1.29 -0.87 -0.67 -0.95 -0.45 >
INFO: kws_search.c(656): kws 3.40 CPU 4.416 xRT
INFO: kws_search.c(658): kws 13.76 wall 17.874 xRT
```
**Your OS (including version) where you are running AlexaPi:**
CentOS 6.7

**Your hardware platform and model you are running on:**
```
Intel(R) Core(TM)2 Quad CPU    Q8200  @ 2.33GHz
```
**Python release (`python2 --version`):**
```
python 2.7.6
```

**Description of problem:**
```
Can heard hello , but no response from AVS server.
```

**Expected:**
```
Hear AVS response.
```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. setting config.yaml
2. run python auth_web.py
3. run python main.py

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

lion6230i, 2016-11-23:
I found that is my audio setting problem. Sorry!

closed 2016-11-23 by lion6230i

renekliment, 2016-11-23:
Glad you figured it out!

palaniraja, 2017-05-14:
@lion6230i what was the issue and how did you fix?