alexa-pi/AlexaPi #94: Implement configurable locale

caveman99, 2016-11-24:
Since AVS is expanding to Europe this fall, current locales are en-US, en-GB and de-DE. This is hardcoded in the AlexaPI source atm. Can this be moved to a configuration setting so i can develop skills for different language sets? Or is there more to it than changing the locale in communikation with AVS?

renekliment, 2016-11-24:
Hey. You mean https://github.com/alexa-pi/AlexaPi/blob/master/src/main.py#L165, right? According to the [API v1 docs](https://developer.amazon.com/public/solutions/alexa/alexa-voice-service/rest/speechrecognizer-requests), only the `en-us` is possible. If you find relevant info, we'll be happy to change it if it's possible.

caveman99, 2016-11-24:
I just found one reference to multilingual versions of Alexa, regarding skills development ( https://developer.amazon.com/public/community/post/Tx1JIHCCFK3SDBI/Alexa-Skills-Kit-SDK-and-Sample-Skill-Templates-Add-Support-for-Multiple-Languag ) but no reference on either the V1 or V2 AVS API docs. There are actual echo devices working in German now, so this must be an extension to at least the V2 API. Or maybe this is controlled via the Echo App? Don't have an original device (yet) to counter-check. Gonna try it out tonight with the current AlexaPI Code, don't have my development PI with me ;-)

renekliment, 2016-11-24:
I think this is up to the Alexa app - http://alexa.amazon.com/ and there is not much we can do.

renekliment, 2016-11-27:
As I'm convinced this is not up to us, I'm closing this. If you find anything that we could use, feel free to tell us and reopen.

closed 2016-11-27 by renekliment