alexa-pi/AlexaPi #95: New config values needed by api v2 @davidmroth

amahlaka, 2016-11-24:
As i cant create a isues on davidmroth's fork, i have to create it here:
On the avs_v2 branch there seems to be 3 new config values: AuthUrl, BaseUrl and API_Version
as there is no up-to-date documentation of what should be filled in those values, i dont know what to fill in those values 
@davidmroth

renekliment, 2016-11-24:
Last I've heard, David said that the branch is far from complete.

davidmroth, 2016-11-24:
@renekliment Yes, you are correct. However, I have a major update coming soon. Spent way too much time on the new logging class, but I learned a lot :-)

@amahlaka Go ahead and open the issue on my branch, that we don't confuse any of @renekliment 's followers.

closed 2016-11-24 by amahlaka