alexa-pi/AlexaPi #96: stereo to 5.1 upmix

AZ-Cam-Op, 2016-11-26:
I am running your program on a raspberry pi 3 with the most up to date version of raspbian. I have found a way to upmix stereo files to 5.1 sound with some setting in the asoundrc file. It works great if I am playing an mp3 with promoe. Is there a way to get it working with shairport-sync?

The following link will take you to the asoundrc file settings.

https://de.mcbf.net/david/dotasoundrc/

please let me know if this is something that could be made possible. thanks.

revere521, 2016-11-26:
whats the output of aplay -L ?

is there a device that alsa sees as the 5.1 upmix?

AZ-Cam-Op, 2016-11-26:
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: Device_1 [USB Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

card 2 is the 5.1/7.1 usb sound card that I have plugged in. I can only hear music from my iphone using shairport and alexa responses on the stereo channels. If I plug the speakers into any of the other ports (rear, center/bass, sides), I don't get anything at all. However as already stated if I play an mp3 with promoe or vlc I can change my speakers to the different outputs and still hear a left and a right channel. I can also open up chromium and stream something and do the same thing. I can also say that the only audio I now get comes from the usb sound card as it is set as my default card. If I plug the speakers into the pi at the 3.5mm output, I hear nothing at all. I can't test the HDMI right now as this monitor doesn't allow for sound but as long as the default is set to the USB sound card that is the only thing that should be working.

AZ-Cam-Op, 2016-11-26:
I should also add that I edited line 29 of the asoundrc file to say card 2 instead of card 0.

revere521, 2016-11-26:
try the aplay -L  with a capital L, that should give you the list of devices with their hardware names (i.e. plughw:1

One of those should represent the upmix, and then you should be able to just set that as the output in config.yaml

AZ-Cam-Op, 2016-11-26:
ok so stupid question will get the obvious answer probably. Where is the config file located? I checked the /opt folder and couldn't find it. I also did a file search for it and the only config.yaml that comes up is completely empty. I know the Alexa service wouldn't be working if there wasn't a working config file. also here is the output with -L.

pi@raspberrypi:~ $ aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
pulse
    PulseAudio Sound Server
dmix8
ch51dup
dmix6
microphone
2chplug
a2chplug
alt2ch
mono_playback
duplex
aduplex
monoduplex
mainvol
default
sysdefault:CARD=ALSA
    bcm2835 ALSA, bcm2835 ALSA
    Default Audio Device
dmix:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample mixing device
dmix:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample mixing device
dsnoop:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct sample snooping device
dsnoop:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct sample snooping device
hw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Direct hardware device without any conversions
hw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Direct hardware device without any conversions
plughw:CARD=ALSA,DEV=0
    bcm2835 ALSA, bcm2835 ALSA
    Hardware device with all software conversions
plughw:CARD=ALSA,DEV=1
    bcm2835 ALSA, bcm2835 IEC958/HDMI
    Hardware device with all software conversions
sysdefault:CARD=Device_1
    USB Sound Device, USB Audio
    Default Audio Device
front:CARD=Device_1,DEV=0
    USB Sound Device, USB Audio
    Front speakers
surround21:CARD=Device_1,DEV=0
    USB Sound Device, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=Device_1,DEV=0
    USB Sound Device, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=Device_1,DEV=0
    USB Sound Device, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=Device_1,DEV=0
    USB Sound Device, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=Device_1,DEV=0
    USB Sound Device, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=Device_1,DEV=0
    USB Sound Device, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=Device_1,DEV=0
    USB Sound Device, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=Device_1,DEV=0
    USB Sound Device, USB Audio
    Direct sample mixing device
dsnoop:CARD=Device_1,DEV=0
    USB Sound Device, USB Audio
    Direct sample snooping device
hw:CARD=Device_1,DEV=0
    USB Sound Device, USB Audio
    Direct hardware device without any conversions
plughw:CARD=Device_1,DEV=0
    USB Sound Device, USB Audio
    Hardware device with all software conversions

I assume I can just put in plughw:1

Thanks.

revere521, 2016-11-26:
ok, cool - first the file is in /etc/opt/AlexaPi/config.yaml. and there is a good audio device debugging section in the wiki. 

Second, set the output device in the config to "surround51:CARD=Device_1,DEV=0"

revere521, 2016-11-26:
if that doesn't work you may be able to use "ch51dup" - i passed right over that the first time

AZ-Cam-Op, 2016-11-26:
Sadly none of the option I have tried have worked and for some reason, I can't get the sound outputting from the other channels at all anymore. I have no idea how the config.yaml would have screwed with that. it doesn't even make sense to me.

revere521, 2016-11-26:
do you have pulseaudio running?  if so, you may need to run it as a service in system-wide mode and set the config to pulse (rather than alsa).  see the section in the wiki for pulseaudio on the pi.  the config.yaml is purely for AlexaPi sound input/output so it shouldn't stop audio otherwise, there's likely another problem there too.

maso27, 2016-11-26:
I see that your original question is about shairplay-sync.
That is unrelated to AlexaPi.
You might be able to add something to _/etc/shairport-sync.conf_, but my recommendation would be to ask at the source, [here](https://github.com/mikebrady/shairport-sync).

AZ-Cam-Op, 2016-11-26:
I will do that. Still trying to get Alexa to work on all the speaker channels. I reverted back to a saved disk image I had before I made a bunch of changes. Haven't had a chance to test it yet.

AZ-Cam-Op, 2016-11-28:
Ok so now I think this is relevant. I got the PI3 back up and running. I have disabled the onboard sound card and only have my usb sound card. The usb sound card works just fine with promoe and vlc and creates the 5.1upmix. The alexapi service isn't liking something about the sound card setup and is quitting because it can't play any sound. Below are the three different ways of seeing the debugging info. Perhaps you can point me in the right direction.  

pi@raspberrypi:~ $ sudo journalctl -u AlexaPi.service
-- Logs begin at Sun 2016-11-27 17:35:33 MST, end at Sun 2016-11-27 17:40:56 MST. --
Nov 27 17:35:36 raspberrypi systemd[1]: Starting Alexa client for all your devices...
Nov 27 17:35:36 raspberrypi systemd[1]: Started Alexa client for all your devices.
Nov 27 17:36:05 raspberrypi python[585]: Checking Internet Connection...
Nov 27 17:36:05 raspberrypi python[585]: Connection OK
Nov 27 17:36:08 raspberrypi pulseaudio[1014]: [pulseaudio] alsa-mixer.c: Volume element
Nov 27 17:36:08 raspberrypi pulseaudio[1014]: [pulseaudio] alsa-mixer.c: Volume element
Nov 27 17:36:08 raspberrypi pulseaudio[1014]: [pulseaudio] alsa-mixer.c: Volume element
Nov 27 17:36:08 raspberrypi pulseaudio[1014]: [pulseaudio] alsa-mixer.c: Volume element
Nov 27 17:36:08 raspberrypi pulseaudio[1014]: [pulseaudio] alsa-mixer.c: Volume element
Nov 27 17:36:09 raspberrypi pulseaudio[1014]: [pulseaudio] server-lookup.c: Unable to c
Nov 27 17:36:09 raspberrypi pulseaudio[1014]: [pulseaudio] main.c: Unable to contact D-
Nov 27 17:36:09 raspberrypi pulseaudio[1014]: [pulseaudio] bluez4-util.c: org.bluez.Man
Nov 27 17:36:09 raspberrypi python[585]: ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Un
Nov 27 17:36:09 raspberrypi python[585]: [02dd0f18] alsa audio output error: cannot ope
Nov 27 17:36:09 raspberrypi python[585]: [02dd0f18] core audio output error: Audio outp
Nov 27 17:36:09 raspberrypi python[585]: [02dd0f18] core audio output error: The audio 
Nov 27 17:36:09 raspberrypi python[585]: No such file or directory.
Nov 27 17:36:09 raspberrypi python[585]: [02dd0f18] core audio output error: module not
Nov 27 17:36:09 raspberrypi python[585]: [73410fe0] core decoder error: failed to creat
Nov 27 17:36:11 raspberrypi python[585]: ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Un
Nov 27 17:36:11 raspberrypi python[585]: Traceback (most recent call last):
Nov 27 17:36:11 raspberrypi python[585]: File "/opt/AlexaPi/src/main.py", line 598, in 
Nov 27 17:36:11 raspberrypi python[585]: loop()
Nov 27 17:36:11 raspberrypi python[585]: File "/opt/AlexaPi/src/main.py", line 509, in 
Nov 27 17:36:11 raspberrypi python[585]: inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, als
Nov 27 17:36:11 raspberrypi python[585]: alsaaudio.ALSAAudioError: No such file or dire
Nov 27 17:36:12 raspberrypi systemd[1]: AlexaPi.service: main process exited, code=exit
Nov 27 17:36:12 raspberrypi systemd[1]: Unit AlexaPi.service entered failed state.
lines 1-29/29 (END)

/opt/AlexaPi/src/main.py -d
INFO: pocketsphinx.c(152): Parsed model-specific feature parameters from /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/feat.params
Current configuration:
[NAME]			[DEFLT]		[VALUE]
-agc			none		none
-agcthresh		2.0		2.000000e+00
-allphone				
-allphone_ci		no		no
-alpha			0.97		9.700000e-01
-ascale			20.0		2.000000e+01
-aw			1		1
-backtrace		no		no
-beam			1e-48		1.000000e-48
-bestpath		yes		yes
-bestpathlw		9.5		9.500000e+00
-ceplen			13		13
-cmn			live		batch
-cmninit		40,3,-1		41.00,-5.29,-0.12,5.09,2.48,-4.07,-1.37,-1.78,-5.08,-2.05,-6.45,-1.42,1.17
-compallsen		no		no
-debug					0
-dict					/usr/local/lib/python2.7/dist-packages/pocketsphinx/model/cmudict-en-us.dict
-dictcase		no		no
-dither			no		no
-doublebw		no		no
-ds			1		1
-fdict					
-feat			1s_c_d_dd	1s_c_d_dd
-featparams				
-fillprob		1e-8		1.000000e-08
-frate			100		100
-fsg					
-fsgusealtpron		yes		yes
-fsgusefiller		yes		yes
-fwdflat		yes		yes
-fwdflatbeam		1e-64		1.000000e-64
-fwdflatefwid		4		4
[... 945 lines cut ...]
snd_rawmidi
brcmfmac
snd_seq_device
brcmutil
snd_pcm
snd_timer
snd
cfg80211
rfkill
bcm2835_gpiomem
i2c_bcm2708
bcm2835_wdt
evdev
uio_pdrv_genirq
uio
i2c_dev
fuse
ipv6

!!ALSA/HDA dmesg
!!--------------

[    4.191882] usb 1-1.5: [2] FU [PCM Capture Volume] ch = 2, val = -4096/2832/1
[    4.193710] usbcore: registered new interface driver snd-usb-audio
[    4.231662] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: May 27 2016 00:13:38 version 7.45.41.26 (r640327) FWID 01-df77e4a7

closed 2016-11-29 by AZ-Cam-Op