alexa-pi/AlexaPi #97: Setup issue: 500 Error

ViennaMike, 2016-11-27:
Running (or rather, trying to install) on Raspberry Pi3. 

**Your OS (including version) where you are running AlexaPi:**
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"

**Your hardware platform and model you are running on:**
Raspberry Pi3

**Python release (`python2 --version`):**
2.7

**Description of problem:**
Trying to install. I get through the install script, enter http://localhost:5050/ in the Pi browser, log into Amazon, then get "500 Internal Server Error The server encountered an unexpected condition which prevented it from fulfilling the request." back from localhost:5050/code. Same thing if I use the actual IP address. Also, if it's relevant, the install script instructions say to that URL, then when done, close out the script with ctrl-C, but I'm getting the shell prompt back before I get a chance to go to that URL in the browser.

I tried re-entering all the parameters from my Amazon device registration, so I don't think there's an error there. It's not the 400 error you get if you haven't whitelisted the URL with amazon (got that one when my URL changed on the pi, fixed it, that error went away).

renekliment, 2016-11-27:
> but I'm getting the shell prompt back before I get a chance to go to that URL in the browser.

How do you get the 500 then?

ViennaMike, 2016-11-28:
Entered it in the browser window anyway, figured it couldn't hurt.

renekliment, 2016-11-28:
It's also weird that you can log in to Amazon even though the script has ended.
I can't reproduce it. I'll try a fresh install later.

davidmroth, 2016-11-28:
@ViennaMike @renekliment I recently ran into the same issue with my code base. I have update auth_web.py with some additional logging that may help with determining what your issue is.

@ViennaMike, if you have time, you can try this branch: [auth_web_updates](https://github.com/davidmroth/AlexaPi/tree/auth_web_updates)

lechat, 2016-11-28:
Could it be that port 5050 is already used by something and Cherrypy fails to bind to that port?

Here is the code that allow to select random unused port:

```
            import socket
            sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
            sock.bind(('', 0))
            sock.listen(1)
            port = sock.getsockname()[1]
            sock.close()

            print port
```
I will submit pull request for this.

davidmroth, 2016-11-28:
@lechat Awesome snippet!

Obviously, one would have to whitelist the port in the Amazon Developer Console to match the random port that is automatically selected by your code.

lechat, 2016-11-28:
Well, then it is of no use for random port... Perhaps better solution would be to check if 5050 is used by something and report that before going to start auth_web.

zeryl, 2016-11-30:
Doesn't look to be related to the port being in use.  It looks like Amazon is throwing the 500, but not ever doing the callback internally.  Is it possible Amazon changed the ability to use an internal IP for the callback?  I used the altered auth web, but never see it re-hit, when amazon throws the 500.

davidmroth, 2016-11-30:
500 error is an internal script error most likely caused by some issue with the config file or the auth_web.py script.  Can someone try using the update I made? https://github.com/davidmroth/AlexaPi/tree/auth_web_updates

zeryl, 2016-11-30:
@davidmroth The issue I was seeing, was not within the python code.  It was on the Amazon site.  It was never redirected back to the Pi.  As my edit shows, the issue was the Device_Type_Id field.  Capitals matter.

davidmroth, 2016-11-30:
@zeryl Interesting... When you say capitals matter... What are you referring to exactly? The field name is derived from the config.yam..  The default field name that is parsed is: 'Device_Type_ID'. Are you saying this was the cause of your issue?

If so, not sure why no one else is having the same issue.

Secondly, after doing a quick code review, we never actually send the string 'Device_Type_ID' to Amazon. Instead, we send 'productID'. 

If possible, I would really like to understand in more detail what you found. Whatever the issue, feels like it may be different than what @ViennaMike is experiencing.

zeryl, 2016-11-30:
@davidmroth That is correct.  in Config.yaml, I had "RasPI" as the Device_Type_ID.  This was throwing a 500 after clicking "accept" for the oatuh on Amazon's site.  Once I changed to RasPi, it worked first shot.

davidmroth, 2016-11-30:
@zeryl So, you changed the YAML "Device_Type_ID" value from "RasPi" to "RasPi" in the config.yaml? 

I feel like I'm misreading your reply.

zeryl, 2016-11-30:
@davidmroth RasPI (upper case I) -> RasPi (lower case i).  As soon as I did this, re-visited http://{ip}:5050, hit accept on Amazon's oAuth page, it refreshed as expected, and redirected back to the Pi with the token displayed.

davidmroth, 2016-12-01:
@zeryl Thank you for the clarification. 

Yes, you are correct in that capitalization does matter. Like I said previously, the issue can be caused by incorrect _"config.yaml"_ settings. They have to match the setting in the Amazon Developer's console exactly...to include capitalization.

At this point, unless @ViennaMike disagrees (still haven't heard back from him), I believe we can consider this matter closed.

renekliment, 2016-12-07:
@ViennaMike Do you still have issues with this? Have you verified your config?

@lechat Do you want to rewrite your _random_port_ branch into something that checks if the port 5050 is used already then? If not, could you please file an issue in the tracker? Thank you.

renekliment, 2016-12-11:
Closing as invalid / no response. Feel free to reopen if needed.

closed 2016-12-11 by renekliment