alexa-pi/AlexaPi #98: No Audio

AZ-Cam-Op, 2016-11-29:
All other audio works fine. The Alexa service won't start because it can't work with Alsa. The log files are below.

pi@raspberrypi:~ $ sudo journalctl -u AlexaPi.service
-- Logs begin at Sun 2016-11-27 17:35:33 MST, end at Sun 2016-11-27 17:40:56 MST. --
Nov 27 17:35:36 raspberrypi systemd[1]: Starting Alexa client for all your devices...
Nov 27 17:35:36 raspberrypi systemd[1]: Started Alexa client for all your devices.
Nov 27 17:36:05 raspberrypi python[585]: Checking Internet Connection...
Nov 27 17:36:05 raspberrypi python[585]: Connection OK
Nov 27 17:36:08 raspberrypi pulseaudio[1014]: [pulseaudio] alsa-mixer.c: Volume element
Nov 27 17:36:08 raspberrypi pulseaudio[1014]: [pulseaudio] alsa-mixer.c: Volume element
Nov 27 17:36:08 raspberrypi pulseaudio[1014]: [pulseaudio] alsa-mixer.c: Volume element
Nov 27 17:36:08 raspberrypi pulseaudio[1014]: [pulseaudio] alsa-mixer.c: Volume element
Nov 27 17:36:08 raspberrypi pulseaudio[1014]: [pulseaudio] alsa-mixer.c: Volume element
Nov 27 17:36:09 raspberrypi pulseaudio[1014]: [pulseaudio] server-lookup.c: Unable to c
Nov 27 17:36:09 raspberrypi pulseaudio[1014]: [pulseaudio] main.c: Unable to contact D-
Nov 27 17:36:09 raspberrypi pulseaudio[1014]: [pulseaudio] bluez4-util.c: org.bluez.Man
Nov 27 17:36:09 raspberrypi python[585]: ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Un
Nov 27 17:36:09 raspberrypi python[585]: [02dd0f18] alsa audio output error: cannot ope
Nov 27 17:36:09 raspberrypi python[585]: [02dd0f18] core audio output error: Audio outp
Nov 27 17:36:09 raspberrypi python[585]: [02dd0f18] core audio output error: The audio 
Nov 27 17:36:09 raspberrypi python[585]: No such file or directory.
Nov 27 17:36:09 raspberrypi python[585]: [02dd0f18] core audio output error: module not
Nov 27 17:36:09 raspberrypi python[585]: [73410fe0] core decoder error: failed to creat
Nov 27 17:36:11 raspberrypi python[585]: ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Un
Nov 27 17:36:11 raspberrypi python[585]: Traceback (most recent call last):
Nov 27 17:36:11 raspberrypi python[585]: File "/opt/AlexaPi/src/main.py", line 598, in 
Nov 27 17:36:11 raspberrypi python[585]: loop()
Nov 27 17:36:11 raspberrypi python[585]: File "/opt/AlexaPi/src/main.py", line 509, in 
Nov 27 17:36:11 raspberrypi python[585]: inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, als
Nov 27 17:36:11 raspberrypi python[585]: alsaaudio.ALSAAudioError: No such file or dire
Nov 27 17:36:12 raspberrypi systemd[1]: AlexaPi.service: main process exited, code=exit
Nov 27 17:36:12 raspberrypi systemd[1]: Unit AlexaPi.service entered failed state.
lines 1-29/29 (END)
/opt/AlexaPi/src/main.py -d
INFO: pocketsphinx.c(152): Parsed model-specific feature parameters from /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/feat.params
Current configuration:
[NAME]			[DEFLT]		[VALUE]
-agc			none		none
-agcthresh		2.0		2.000000e+00
-allphone				
-allphone_ci		no		no
-alpha			0.97		9.700000e-01
-ascale			20.0		2.000000e+01
-aw			1		1
-backtrace		no		no
-beam			1e-48		1.000000e-48
-bestpath		yes		yes
-bestpathlw		9.5		9.500000e+00
-ceplen			13		13
-cmn			live		batch
-cmninit		40,3,-1		41.00,-5.29,-0.12,5.09,2.48,-4.07,-1.37,-1.78,-5.08,-2.05,-6.45,-1.42,1.17
-compallsen		no		no
-debug					0
-dict					/usr/local/lib/python2.7/dist-packages/pocketsphinx/model/cmudict-en-us.dict
-dictcase		no		no
-dither			no		no
-doublebw		no		no
-ds			1		1
-fdict					
-feat			1s_c_d_dd	1s_c_d_dd
-featparams				
-fillprob		1e-8		1.000000e-08
-frate			100		100
-fsg					
-fsgusealtpron		yes		yes
-fsgusefiller		yes		yes
-fwdflat		yes		yes
-fwdflatbeam		1e-64		1.000000e-64
-fwdflatefwid		4		4
-fwdflatlw		8.5		8.500000e+00
[... 943 lines cut ...]
snd_rawmidi
brcmfmac
snd_seq_device
brcmutil
snd_pcm
snd_timer
snd
cfg80211
rfkill
bcm2835_gpiomem
i2c_bcm2708
bcm2835_wdt
evdev
uio_pdrv_genirq
uio
i2c_dev
fuse
ipv6

!!ALSA/HDA dmesg
!!--------------

[    4.191882] usb 1-1.5: [2] FU [PCM Capture Volume] ch = 2, val = -4096/2832/1
[    4.193710] usbcore: registered new interface driver snd-usb-audio
[    4.231662] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: May 27 2016 00:13:38 version 7.45.41.26 (r640327) FWID 01-df77e4a7

revere521, 2016-11-30:
this does not seem right: 
```
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM raspberrypi
[025b3280] alsa audio output error: cannot open ALSA device "**raspberrypi**": No such file or directory
[025b3280] core audio output error: Audio output failed
[025b3280] core audio output error: The audio device "**raspberrypi**" could not be used:
No such file or directory.
```
Did that get put in the config.yaml in the wrong place?

AZ-Cam-Op, 2016-11-30:
This is what my config.yaml looks like. I followed the setup instructions like every other time and this is how it came out. I haven't touched it.

# Rename this file to config.yaml and fill in the fields

# See https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging
# to get a better idea of how to set things in here
sound:
  # Name of your ALSA microphone/soundcard in arecord -L
  # you can also try setting it to 'default'
  input_device: "raspberrypi"

  # you can go here with alsa, pulse or others
  # keep blank for the best audio output selected by the underlying code
  output: "alsa"
  # output device: "raspberrypi"
  # currently supports ALSA only
  output_device: "raspberrypi"

# Amazon Alexa settings
alexa:
  Client_ID: 
  Client_Secret: 
  Device_Type_ID: "alexapi"
  Security_Profile_Description: "alexapi"
  Security_Profile_ID:
  refresh_token: 

sphinx:
  trigger_phrase: "alexa"

platform:
  # Name of your platform, e.g. raspberrypi, orangepi, desktop
  device: "raspberrypi"

platforms:
  # common config for all platforms (if they support the options)
  # can be overriden in the the respective platform sections
  common:
    # whether you want to hear Alexa's "yes" after you press a button
    # "true" or "false"
    should_confirm_trigger: false

    long_press:
      # system command to run after pressing the button for _duration_ number of seconds
      # empty to disable
      command: ""

      # duration in seconds
      duration: 10

      # CURRENTLY DISABLED
      # optional audio file to play before executing the command
      # you might wanna use full path to make sure AlexaPi finds it
      # you can use {resources_path} for the path to the AlexaPi "resources" directory
      audio_file: ""
      
  chip:
    # GPIO Pin with button connected
    button: XIO-P1
    # GPIO Pin for the playback/activity light
    plb_light: XIO-P3
    # GPIO Pin for the recording light
    rec_light: XIO-P5

  raspberrypi:
    # GPIO Pin with button connected
    button: 18
    # GPIO Pin for the playback/activity light
    plb_light: 24
    # GPIO Pin for the recording light
    rec_light: 25

  orangepi:
    # GPIO Pin with button connected
    button: 20
    # GPIO Pin for the playback/activity light
    plb_light: 9
    # GPIO Pin for the recording light
    rec_light: 8

  desktop:
    should_confirm_trigger: true

    min_seconds_to_record: 3

  dummy:
    # whether to print messages when methods are called
    verbose: false

AZ-Cam-Op, 2016-11-30:
I tried changing the output_device of config.yaml to default. When raspbian started I heard the Alexa "Hello" but after that it stopped working. Here is some more debugging stuff. If I need to run a different debug routine, let me know.

pi@raspberrypi:~ $ /opt/AlexaPi/src/main.py -dINFO: pocketsphinx.c(152): Parsed model-specific feature parameters from /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/feat.params
Current configuration:
[NAME]			[DEFLT]		[VALUE]
-agc			none		none
-agcthresh		2.0		2.000000e+00
-allphone				
-allphone_ci		no		no
-alpha			0.97		9.700000e-01
-ascale			20.0		2.000000e+01
-aw			1		1
-backtrace		no		no
-beam			1e-48		1.000000e-48
-bestpath		yes		yes
-bestpathlw		9.5		9.500000e+00
-ceplen			13		13
-cmn			live		batch
-cmninit		40,3,-1		41.00,-5.29,-0.12,5.09,2.48,-4.07,-1.37,-1.78,-5.08,-2.05,-6.45,-1.42,1.17
-compallsen		no		no
-debug					0
-dict					/usr/local/lib/python2.7/dist-packages/pocketsphinx/model/cmudict-en-us.dict
-dictcase		no		no
-dither			no		no
-doublebw		no		no
-ds			1		1
-fdict					
-feat			1s_c_d_dd	1s_c_d_dd
-featparams				
-fillprob		1e-8		1.000000e-08
-frate			100		100
-fsg					
-fsgusealtpron		yes		yes
-fsgusefiller		yes		yes
-fwdflat		yes		yes
-fwdflatbeam		1e-64		1.000000e-64
-fwdflatefwid		4		4
-fwdflatlw		8.5		8.500000e+00
-fwdflatsfwin		25		25
-fwdflatwbeam		7e-29		7.000000e-29
-fwdtree		yes		yes
-hmm					/usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us
-input_endian		little		little
-jsgf					
-keyphrase				alexa
-kws					
-kws_delay		10		10
-kws_plp		1e-1		1.000000e-01
-kws_threshold		1		1.000000e-05
-latsize		5000		5000
-lda					
-ldadim			0		0
-lifter			0		22
-lm					
-lmctl					
-lmname					
-logbase		1.0001		1.000100e+00
-logfn					
-logspec		no		no
-lowerf			133.33334	1.300000e+02
-lpbeam			1e-40		1.000000e-40
-lponlybeam		7e-29		7.000000e-29
-lw			6.5		6.500000e+00
-maxhmmpf		30000		30000
-maxwpf			-1		-1
-mdef					
-mean					
-mfclogdir				
-min_endfr		0		0
-mixw					
[... 77 lines cut ...]
INFO: dict.c(213): Dictionary size 134723, allocated 1016 KiB for strings, 1679 KiB for phones
INFO: dict.c(336): 134723 words read
INFO: dict.c(358): Reading filler dictionary: /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/noisedict
INFO: dict.c(213): Dictionary size 134728, allocated 0 KiB for strings, 0 KiB for phones
INFO: dict.c(361): 5 words read
INFO: dict2pid.c(396): Building PID tables for dictionary
INFO: dict2pid.c(406): Allocating 42^3 * 2 bytes (144 KiB) for word-initial triphones
INFO: dict2pid.c(132): Allocated 21336 bytes (20 KiB) for word-final triphones
INFO: dict2pid.c(196): Allocated 21336 bytes (20 KiB) for single-phone word triphones
INFO: kws_search.c(406): KWS(beam: -1080, plp: -23, default threshold -113, delay 10)
Checking Internet Connection...
Connection OK
Play_Audio Request for: /opt/AlexaPi/src/resources/hello.mp3
Player State: State.Opening
Player State: State.Playing
Player State: State.Ended
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM raspberrypi
Traceback (most recent call last):
  File "/opt/AlexaPi/src/main.py", line 598, in <module>
    loop()
  File "/opt/AlexaPi/src/main.py", line 509, in loop
    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, config['sound']['input_device'])
alsaaudio.ALSAAudioError: No such file or directory [raspberrypi]
INFO: kws_search.c(448): TOTAL kws 0.00 CPU nan xRT
INFO: kws_search.c(451): TOTAL kws 0.00 wall nan xRT

renekliment, 2016-11-30:
Hey! Sorry for this, it's a bug we missed.
Set `output_device` to `default` and `input_device` to either `plughw:0` or `plughw:1` depending on which USB sound input you want to use.

renekliment, 2016-11-30:
Now, all the `*_device` variables accidentally get set to `raspberry`. Could you please test / confirm if #99 fixes this for you?

AZ-Cam-Op, 2016-12-01:
I am not sure what the hot fix in #99 is asking me to do.

I have the output to default and the input to plughw:0. when I initialize alexapi manually with the debug script running, it works just fine. I am still having issues on boot though. I hear the initial hello and then when i check what is running with the top command alexapi is no longer running. I would provide a debug file but like I said when I run it that way, it works just fine.

renekliment, 2016-12-01:
#99 sets up the default audio devices correctly in the config.

Regarding running AlexaPI as service: https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#pulseaudio

AZ-Cam-Op, 2016-12-02:
Ok, I did the debugging stuff with Pulse Audio and got that running. Alexa worked great. However if I open up chromium and go to youtube and try and watch a video it locks up the system. The video plays but the keyboard doesn't work any longer and there is no audio. Also, if I open up promoe instead of chromium and try and play music, all I get is static. I'm sure this has something to do with me trying to get this crazy stereo to 5.1 upmix stuff working. I have pretty much given up hope of being able to have the 5.1 upmix and alexa at the same time. While it would be cool to have Alexa in my truck on the go, it isn't necessary. If you have any thoughts on how I can get all of it working together that would be great, if not, I will be moving on and I appreciate all of the help everyone has given me thus far.

closed 2016-12-07 by renekliment

renekliment, 2016-12-07:
@AZ-Cam-Op I'm not sure what the issue is - whether it's a PA bug, wrong config, or something else.

Just to make sure, please post again the output of `journalctl -u AlexaPi.service` and output when you run it interactively in debug mode, both with the fixed config.

commit 253823f4 (2016-12-07): add space before key name to prevent matching other words (#99), fix #98