alexa-pi/AlexaPi #100: Fresh install on RaspberryPi has sound issues (PA system-wide mode)

idimitrakopoulos, 2016-12-04:
Hi all,

what a great project which I can't wait to get my hands on! :) I still havent managed to make it work on a Raspberry PI Model B+.

1. Fresh install of Jessie
2. Went straight to clone the latest repo  and installed it (that happened yesterday). Today I checked again to see I am on the latest changes

`pi@raspberrypi:/opt/AlexaPi $ git status`

```
On branch master
Your branch is up-to-date with 'origin/master'.
nothing to commit, working directory clean
```

Yup!

So I installed PA in system-wide as some other posts suggested. What happens now is that I get:

`alsa audio output error: cannot estimate delay: Input/output error`

when querying the AlexaPi service. 

`pi@raspberrypi:/opt/AlexaPi $ sudo systemctl status AlexaPi.service -l` 

```
● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
  Drop-In: /etc/systemd/system/AlexaPi.service.d
           └─restart.conf
   Active: active (running) since Sun 2016-12-04 20:44:34 GMT+2; 4min 11s ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
 Main PID: 1733 (python)
   CGroup: /system.slice/AlexaPi.service
           └─1733 /usr/bin/python /opt/AlexaPi/src/main.py

Dec 04 20:44:34 raspberrypi systemd[1]: Started Alexa client for all your devices.
Dec 04 20:44:41 raspberrypi python[1733]: [02097cf8] alsa audio output error: cannot estimate delay: Input/output error

```

and I also get the following when querying the PA service

`pi@raspberrypi:~ $ sudo systemctl status pulseaudio.service -l`

```
● pulseaudio.service - PulseAudio Daemon
   Loaded: loaded (/etc/systemd/system/pulseaudio.service; enabled)
   Active: active (running) since Sun 2016-12-04 20:54:45 GMT+2; 1min 8s ago
 Main PID: 432 (pulseaudio)
   CGroup: /system.slice/pulseaudio.service
           └─432 /usr/bin/pulseaudio --system --realtime --disallow-exit --no-cpu-limit

Dec 04 20:54:45 raspberrypi pulseaudio[432]: W: [pulseaudio] main.c: Running in system mode, but --disallow-module-loading not set!
Dec 04 20:54:45 raspberrypi pulseaudio[432]: N: [pulseaudio] main.c: Running in system mode, forcibly disabling SHM mode!
Dec 04 20:54:45 raspberrypi pulseaudio[432]: N: [pulseaudio] main.c: Running in system mode, forcibly disabling exit idle time!
Dec 04 20:54:45 raspberrypi pulseaudio[432]: W: [pulseaudio] main.c: OK, so you are running PA in system mode. Please note that you most likely shouldn't be doing that.
Dec 04 20:54:45 raspberrypi pulseaudio[432]: W: [pulseaudio] main.c: If you do it nonetheless then it's your own fault if things don't work as expected.
Dec 04 20:54:45 raspberrypi pulseaudio[432]: W: [pulseaudio] main.c: Please read http://pulseaudio.org/wiki/WhatIsWrongWithSystemMode for an explanation why system mode is usually a bad idea.
Dec 04 20:54:46 raspberrypi pulseaudio[432]: W: [pulseaudio] authkey.c: Failed to open cookie file '/var/run/pulse/.config/pulse/cookie': No such file or directory
Dec 04 20:54:46 raspberrypi pulseaudio[432]: W: [pulseaudio] authkey.c: Failed to load authorization key '/var/run/pulse/.config/pulse/cookie': No such file or directory
Dec 04 20:54:46 raspberrypi pulseaudio[432]: W: [pulseaudio] authkey.c: Failed to open cookie file '/var/run/pulse/.pulse-cookie': No such file or directory
Dec 04 20:54:46 raspberrypi pulseaudio[432]: W: [pulseaudio] authkey.c: Failed to load authorization key '/var/run/pulse/.pulse-cookie': No such file or directory
```

Alexa doesnt respond to my calling its name. I have searched lots of similar tickets in this project but to no avail. Any ideas?

p.s. not sure if this helps but it also stopped saying Hello on launch

renekliment, 2016-12-04:
Hey!

> `alsa audio output error: cannot estimate delay: Input/output error`

This is normal, just ignore it.

Check your `configuration.yaml` to make sure you have correct audio devices set up. There is currently a bug in `master` that sets it up incorrectly. See https://github.com/alexa-pi/AlexaPi/issues/98#issuecomment-263794025

idimitrakopoulos, 2016-12-04:
Thanks @renekliment but I think I have already changed my yaml as per tutorial...

See this part below

```
sound:

   input_device: "pulse"
   output: "pulse"
   output_device: ""

  # Name of your ALSA microphone/soundcard in arecord -L
  # you can also try setting it to 'default'
  #input_device: "plughw:0"

  # you can go here with alsa, pulse or others
  # keep blank for the best audio output selected by the underlying code
  #output: "alsa"
  # output device: "raspberrypi"
  # currently supports ALSA only
  #output_device: "default"
 
```

My setup is Raspberry Pi B+ and USB microphone from ebay. 

`pi@raspberrypi:~ $ arecord -L`
```
null
    Discard all samples (playback) or generate zero samples (capture)
pulse
    PulseAudio Sound Server
default
    Playback/recording through the PulseAudio sound server
sysdefault:CARD=Device
    USB PnP Sound Device, USB Audio
    Default Audio Device
front:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    Front speakers
surround21:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    Direct sample mixing device
dsnoop:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    Direct sample snooping device
hw:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    Direct hardware device without any conversions
plughw:CARD=Device,DEV=0
    USB PnP Sound Device, USB Audio
    Hardware device with all software conversions
```
And ...
```
pi@raspberrypi:~ $ arecord -l
```
```
**** List of CAPTURE Hardware Devices ****
card 1: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
```
Any ideas? :) :)

idimitrakopoulos, 2016-12-05:
Hi again @renekliment,

I see that in another comment https://github.com/alexa-pi/AlexaPi/issues/38#issuecomment-255590140 you have stated that PA must be disabled so I'm kinda confused on whether i should gi ahead and completely disable it since the System wide option didnt fix my problem. 

Can you tell me what else i should look for? any idea? anything...

renekliment, 2016-12-05:
Hey. You can either disable PA if you're not using other audio apps or use it in system mode. The documentation (https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#pulseaudio) is up-to-date and you can choose whichever option you want to.

I'm not entirely sure. You can check your audio output levels ... this is more PA setting than AlexaPi. You can come and ask at our Giiter chatroom (link in README).

idimitrakopoulos, 2016-12-05:
Hmmm after debugging a bit more with ./main.py -d I get the following output

```
Checking Internet Connection...
Connection OK
Play_Audio Request for: /opt/AlexaPi/src/resources/hello.mp3
Player State: State.Opening
Player State: State.Playing
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Access denied

[01bdae80] alsa audio output error: cannot open ALSA device "default": Connection refused
[01bdae80] core audio output error: Audio output failed
[01bdae80] core audio output error: The audio device "default" could not be used:
Connection refused.
[01bdae80] core audio output error: module not functional
[74511220] core decoder error: failed to create audio output
Player State: State.Ended
ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Access denied

Traceback (most recent call last):
  File "./main.py", line 598, in <module>
    loop()
  File "./main.py", line 509, in loop
    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, config['sound']['input_device'])
alsaaudio.ALSAAudioError: Connection refused [pulse]
INFO: kws_search.c(448): TOTAL kws 0.00 CPU nan xRT
INFO: kws_search.c(451): TOTAL kws 0.00 wall nan xRT
```
pulseaudio reports

```
pi@raspberrypi:/opt/AlexaPi/src $ sudo systemctl status pulseaudio.service -l
● pulseaudio.service - PulseAudio Daemon
   Loaded: loaded (/etc/systemd/system/pulseaudio.service; enabled)
   Active: active (running) since Mon 2016-12-05 15:31:10 GMT+2; 1min 14s ago
 Main PID: 1243 (pulseaudio)
   CGroup: /system.slice/pulseaudio.service
           └─1243 /usr/bin/pulseaudio --system --realtime --disallow-exit --no-cpu-limit

Dec 05 15:31:10 raspberrypi systemd[1]: Started PulseAudio Daemon.
Dec 05 15:31:10 raspberrypi pulseaudio[1243]: W: [pulseaudio] main.c: Running in system mode, but --disallow-module-loading not set!
Dec 05 15:31:10 raspberrypi pulseaudio[1243]: N: [pulseaudio] main.c: Running in system mode, forcibly disabling SHM mode!
Dec 05 15:31:10 raspberrypi pulseaudio[1243]: N: [pulseaudio] main.c: Running in system mode, forcibly disabling exit idle time!
Dec 05 15:31:10 raspberrypi pulseaudio[1243]: W: [pulseaudio] main.c: OK, so you are running PA in system mode. Please note that you most likely shouldn't be doing that.
Dec 05 15:31:10 raspberrypi pulseaudio[1243]: W: [pulseaudio] main.c: If you do it nonetheless then it's your own fault if things don't work as expected.
Dec 05 15:31:10 raspberrypi pulseaudio[1243]: W: [pulseaudio] main.c: Please read http://pulseaudio.org/wiki/WhatIsWrongWithSystemMode for an explanation why system mode is usually a bad idea.
Dec 05 15:32:14 raspberrypi pulseaudio[1243]: W: [pulseaudio] protocol-native.c: Denied access to client with invalid authorization data.
Dec 05 15:32:16 raspberrypi pulseaudio[1243]: W: [pulseaudio] protocol-native.c: Denied access to client with invalid authorization data.
```
Something is probably very wrong here....

spo0n, 2016-12-06:
i have a similar problem.

I did the tutorial for PA system-wide mode and i'm getting this when i run sudo journalctl -u AlexaPi.service

`pi@raspberrypi:~ $ sudo journalctl -u AlexaPi.service
-- Logs begin at Tue 2016-12-06 03:55:26 UTC, end at Tue 2016-12-06 04:00:59 UTC. --
Dec 06 03:55:41 raspberrypi systemd[1]: Starting Alexa client for all your devices...
Dec 06 03:55:41 raspberrypi systemd[1]: Started Alexa client for all your devices.
Dec 06 03:56:15 raspberrypi python[734]: Failed to create secure directory (/var/lib/AlexaPi/.config/pulse): Operation not permitted
Dec 06 03:56:15 raspberrypi python[734]: Failed to create secure directory (/var/lib/AlexaPi/.config/pulse): Operation not permitted
Dec 06 03:56:15 raspberrypi python[734]: Failed to create secure directory (/var/lib/AlexaPi/.config/pulse): Operation not permitted
Dec 06 03:56:15 raspberrypi python[734]: Failed to create secure directory (/var/lib/AlexaPi/.config/pulse): Operation not permitted
Dec 06 03:56:17 raspberrypi python[734]: [025c59b8] alsa audio output error: cannot estimate delay: Input/output error
Dec 06 03:56:18 raspberrypi python[734]: Failed to create secure directory (/var/lib/AlexaPi/.config/pulse): Operation not permitted
Dec 06 03:56:37 raspberrypi python[734]: Failed to create secure directory (/var/lib/AlexaPi/.config/pulse): Operation not permitted
Dec 06 03:56:37 raspberrypi python[734]: Failed to create secure directory (/var/lib/AlexaPi/.config/pulse): Operation not permitted
Dec 06 03:56:37 raspberrypi python[734]: Failed to create secure directory (/var/lib/AlexaPi/.config/pulse): Operation not permitted
Dec 06 03:56:38 raspberrypi python[734]: Failed to create secure directory (/var/lib/AlexaPi/.config/pulse): Operation not permitted
Dec 06 03:56:49 raspberrypi python[734]: Failed to create secure directory (/var/lib/AlexaPi/.config/pulse): Operation not permitted
Dec 06 04:00:09 raspberrypi systemd[1]: Stopping Alexa client for all your devices...
Dec 06 04:00:09 raspberrypi python[734]: Checking Internet Connection...
Dec 06 04:00:09 raspberrypi python[734]: Connection OK
Dec 06 04:00:09 raspberrypi systemd[1]: Stopped Alexa client for all your devices.
Dec 06 04:00:52 raspberrypi systemd[1]: Starting Alexa client for all your devices...
Dec 06 04:00:52 raspberrypi systemd[1]: Started Alexa client for all your devices.
Dec 06 04:00:57 raspberrypi python[1262]: Failed to create secure directory (/var/lib/AlexaPi/.config/pulse): Operation not permitted
Dec 06 04:00:57 raspberrypi python[1262]: Failed to create secure directory (/var/lib/AlexaPi/.config/pulse): Operation not permitted
Dec 06 04:00:57 raspberrypi python[1262]: Failed to create secure directory (/var/lib/AlexaPi/.config/pulse): Operation not permitted
Dec 06 04:00:57 raspberrypi python[1262]: Failed to create secure directory (/var/lib/AlexaPi/.config/pulse): Operation not permitted`

can you help me?

renekliment, 2016-12-06:
Alright, this is weird. Let's try a few things:

```
grep "pulse\|alexa" /etc/group
ps uax | grep "pulse\|alexa"
systemctl status AlexaPi
systemctl status pulseaudio
```

and paste the output here.

idimitrakopoulos, 2016-12-06:
By the way I had mine working by disabling PA completely. However let me give some pointers to others struggling to make it work as I discovered them along the way.

1. Make sure you don't have a shitty mic. I have to hold it really close to me for Alexa to respond.
2. Ensure your speaker sound and microphone gain are set high enough. You can do this using `alsamixer`
3. Make sure your raspberry is getting enough power (5.1V and 2.5A recommended although it works with lower) . I saw differences in Alexa responses not being cut-off when using a proper USB charger rather than my PC port

@renekliment You can close this ticket if you want

spo0n, 2016-12-06:
@renekliment 

here we goes the output you asked : 

> pi@raspberrypi:~ $ grep "pulse\|alexa" /etc/group
audio : x : 29:pi,pulse,alexapi,shairport-sync
gpio : x :997:pi,alexapi
pulse : x :116:
pulse-access : x :117:pi,alexapi
alexapi : x :996:

> pi@raspberrypi:~$ ps uax | grep "pulse\|alexa"
pulse      427  2.6  0.8 100924  7728 ?        S<sl 12:02   0:05 /usr/bin/pulseaudio --system --realtime --disallow-exit --no-cpu-limit
alexapi    723  6.9  5.1 174684 48492 ?        Ssl  12:02   0:13 /usr/bin/python /opt/AlexaPi/src/main.py
pi        1111  0.0  0.0   1912    96 ?        S    12:03   0:00 /bin/sh /usr/bin/start-pulseaudio-x11
pi        1293  0.0  0.2   4280  1916 pts/0    S+   12:06   0:00 grep --color=auto pulse\|alexa

> pi@raspberrypi:~$ systemctl status AlexaPi
● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
  Drop-In: /etc/systemd/system/AlexaPi.service.d
           └─restart.conf
   Active: active (running) since Tue 2016-12-06 04:45:04 UTC; 7h ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
 Main PID: 723 (python)
   CGroup: /system.slice/AlexaPi.service
           └─723 /usr/bin/python /opt/AlexaPi/src/main.py

> pi@raspberrypi:~ $ systemctl status pulseaudio
● pulseaudio.service - PulseAudio Daemon
   Loaded: loaded (/etc/systemd/system/pulseaudio.service; enabled)
   Active: active (running) since Tue 2016-12-06 04:44:51 UTC; 7h ago
 Main PID: 427 (pulseaudio)
   CGroup: /system.slice/pulseaudio.service
           └─427 /usr/bin/pulseaudio --system --realtime --disallow-exit --no-cpu-limit

I hope it helps.

revere521, 2016-12-06:
@spo0n - make sure you did this part:  
`mkdir -p /var/lib/AlexaPi/.config/pulse
sudo cp /etc/pulse/client.conf /var/lib/AlexaPi/.config/pulse/`

It looks like its complaining that pulse can't access that directory where the pulse configs should be copied

@idimitrakopoulos - I ended up having to play with the audio output device to get it to work, i am running AlexaPi along with MagicMirror2 - so i'm using my HDMI audio to a monitor's builtin speakers.  That said, i had to change `output_device: ""` to `output_device: "pulse"`  so, all 3 said pulse.   You may also need to run 'aplay -L' and see what your hw device is called adn set it to that like 'output_device: "plughw:CARD=ALSA,DEV=0"'  or whatever the correct device lists for you.

 That said the only reason I bothered to setup PulseAudio in the first place is that i have a PS3 Eye as my mic (its an array mic and can hear me across the room now)  - and that does not play nice with alsa at all.  If your audio devices are  good with alsa, and you don't need pulse for some other sound source/application - then disable it and stick with alsa -- way fewer headaches. 

@renekliment - the wiki may need to be updated so all three options for audio are set to 'pulse"  with a direction to try play -L for the best device if any issues are encountered.  what do you think?

spo0n, 2016-12-06:
@revere521 - when i try to create the directory, give me this error : 

>pi@raspberrypi:~ $ mkdir -p /var/lib/AlexaPi/.config/pulse
mkdir: cannot create directory ‘/var/lib’: Permission denied

idimitrakopoulos, 2016-12-06:
@revere521 thanks for your help! i actually didnt do that and it could have been the issue! well now I'll try it and let everyone know.

Additionally I was in the market for omni USB micrphones but after your msg I checked the PS3 Eye and it's cheap enough! i might go with that if you recommend it.

spo0n, 2016-12-06:
@revere521 
i think something wrong because i can't create the dir, but under sudo i can but it keeps getting the error of permissions. i dont understand what happen

revere521, 2016-12-07:
@spo0n you will need to elevate permissions with sudo to create a directory outside the home dir (/home/pi) - so that command should be `sudo mkdir -p /var/lib/AlexaPi/.config/pulse` then run `sudo cp /etc/pulse/client.conf /var/lib/AlexaPi/.config/pulse/` that copies the pulseaudio config files to that new directory.   
 then you may need to set the permissions of the AlexaPi as a home folder if its still complaining - like this: 
`chown -R alexapi:alexapi /var/lib/AlexaPi/
usermod --home /var/lib/AlexaPi alexapi`

@idimitrakopoulos the PS3 Eye seems to work really well as an array mic - and its cheap enough that if its too much of a hassle to get it working - no big deal.  You may have to do some additional playing around with it - but I have had it running for a few weeks now and AlexaPi has no issues hearing me from pretty good distances.

renekliment, 2016-12-07:
If you get _access denied_ errors in pulse / AlexaPi log, make sure the user (_pi_, or even _root_) is in the `pulse-access` group: https://github.com/alexa-pi/AlexaPi/issues/38#issuecomment-265480333

spo0n, 2016-12-07:
@renekliment  @revere521  it works, i'm not getting any errors now apart this one : 

> alsa audio output error: cannot estimate delay: Input/output error

but it's normal, thank you guys.

@renekliment 

Another question, i'm using a distro based and 100% compatible with raspian, do you think it's gonna work or not??

renekliment, 2016-12-07:
Great to hear that.

> 100% compatible with raspian, do you think it's gonna work or not??

Well, that question doesn't make much sense, does it :smile: Anything Deiban based will work.

spo0n, 2016-12-07:
@renekliment 

I thought so, the worst thing can happen it's some changes of the distro may cause incompatibility no??

renekliment, 2016-12-11:
The wiki has been updated for the root override and a few other things. I see this as resolved, therefore closing. Feel free to reopen if needed.

closed 2016-12-11 by renekliment

rafael-azevedo, 2016-12-24:
When trying to run with with pulse configured 

config.conf moved to /var/lib/AlexaPi/.config/pulse/ with line autospawn=no

```
pi@raspberrypi:/opt/AlexaPi/src $ grep "pulse\|alexa" /etc/group
audio:x:29:pi,pulse,alexapi,shairport-sync
gpio:x:997:pi,alexapi
pulse:x:116:
pulse-access:x:117:pi,alexapi,root
alexapi:x:996:
```

I get the following error 

```
Connection OK
[749004d8] alsa audio output error: cannot estimate delay: Input/output error
ALSA lib conf.c:4705:(snd_config_expand) Unknown parameters CARD=pulse
ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM default:CARD=pulse
Traceback (most recent call last):
  File "./main.py", line 587, in <module>
    loop()
  File "./main.py", line 505, in loop
    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, config['sound']['input_device'])
alsaaudio.ALSAAudioError: Invalid argument
```

shreyasubale, 2016-12-24:
I was getting choppy audio with PA  ( initially i thought it wasn't working at all, but when i played some music i found out that the audio is actualy very choppy ) until i installed vlc manually with` apt-get install vlc`

maso27, 2017-01-02:
@shreyasubale 
Thanks for sharing!  I just tried this on my CHIP (still on external sound card), and it brought the PA-based output from being so choppy it was unusable to being perfect.

Adding it to the wiki.

maso27, 2017-01-02:
@shreyasubale 
Thanks for the tip!
I just tried this on my CHIP (still an external sound card), and it did help things.
I've added it to the wiki.